classicrocker883/MRiscoCProUI #1: Is it possible to support Aquila X2 H32 Chip?

audynamo, 2022-09-15:
I tried to compile Marlin-H32 (https://github.com/alexqzd/Marlin-H32) project myself, it is working, but it is based on not up-to-date Marlin, so some features might be missing (compared with this repo). I am not sure how to use this project to compile (merge with) Marlin-H32 as it has a lot of h32chip related code.
Do you have any idea what is difference between GD32F103RE/RC N32F103RE which use 'pins_Creality_V3.h' header to define pins, and the H32 maybe is called as 'HC32F46X'. Could we reuse those pins definition to drive H32, or we still need to compile H32's own driver library?

classicrocker883, 2022-09-15:
so you were able to compile H32 with the Eclipse IDE, right?

the difference between the chips- GD32, N32, STM32.... is that compiling code is easy and straightforward using MS Visual Studio Code and PlatformIO. if you wanted a more up to date Marlin in the H32 project, then what you will have to do is copy the new Marlin folder into the H32 Marlin folder. It should overwrite all the other files. However, I would make a backup copy before doing so. you really want the `src` folder and everything that is in it and the configuration.h and _adv.h files. you may have to tweak the newer config files to be like the H32 ones.

but if you get compiling errors, then try fixing whatever errors you get.
if all else fails, the other thing to do might be to copy one folder or file at a time. I mean that is if you know what youre doing.

and it should work eventually. i might be giving this a try one day

audynamo, 2022-09-16:
Thank you for your good suggestion to replace one folder/file at a time.
I am trying to do the opposite, adding related hc32 code into MriscocProUi's Marlin subfolder hc32, delete framework/cores/* as it seems all files in there are related to arduinostm32 library that platformio will pick up to compile. Tweaking with include path, I managed to compile all code include hc32's, however, when linking all .o files, I got the error:
```
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RE_creality/src/src/hc32f46x/drivers/src/hc32f46x_interrupts.c.o: in function `SysTick_Handler':
/Users/xinweix/Work/AquilaX2/MriscocProUI/Marlin/src/hc32f46x/drivers/src/hc32f46x_interrupts.c:455: multiple definition of `SysTick_Handler'; .pio/build/STM32F103RE_creality/SrcWrapper/src/stm32/clock.c.o:/Users/xinweix/.platformio/packages/framework-arduinoststm32/libraries/SrcWrapper/src/stm32/clock.c:87: first defined here
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RE_creality/src/src/hc32f46x/main/hdsc32core/common/system_hc32f46x.c.o: in function `SystemCoreClockUpdate':
/Users/xinweix/Work/AquilaX2/MriscocProUI/Marlin/src/hc32f46x/main/hdsc32core/common/system_hc32f46x.c:88: multiple definition of `SystemCoreClockUpdate'; .pio/build/STM32F103RE_creality/SrcWrapper/src/stm32/system_stm32yyxx.c.o:/Users/xinweix/.platformio/packages/framework-arduinoststm32/system/STM32F1xx/system_stm32f1xx.c:253: first defined here
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RE_creality/src/src/hc32f46x/main/hdsc32core/common/system_hc32f46x.c.o: in function `SystemInit':
/Users/xinweix/Work/AquilaX2/MriscocProUI/Marlin/src/hc32f46x/main/hdsc32core/common/system_hc32f46x.c:84: multiple definition of `SystemInit'; .pio/build/STM32F103RE_creality/SrcWrapper/src/stm32/system_stm32yyxx.c.o:/Users/xinweix/.platformio/packages/framework-arduinoststm32/system/STM32F1xx/system_stm32f1xx.c:154: first defined here
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RE_creality/src/src/hc32f46x/main/hdsc32core/common/system_hc32f46x.c.o:(.data.SystemCoreClock+0x0): multiple definition of `SystemCoreClock'; .pio/build/STM32F103RE_creality/SrcWrapper/src/stm32/system_stm32yyxx.c.o:(.data.SystemCoreClock+0x0): first defined here
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RE_creality/src/src/hc32f46x/main/main.cpp.o: in function `main':
/Users/xinweix/Work/AquilaX2/MriscocProUI/Marlin/src/hc32f46x/main/main.cpp:44: multiple definition of `main'; .pio/build/STM32F103RE_creality/FrameworkArduino/main.cpp.o:/Users/xinweix/.platformio/packages/framework-arduinoststm32/cores/arduino/main.cpp:49: first defined here
```
The last main error, I suppose we can delete the main.cpp in hc32 and reuse `.pio/build/STM32F103RE_creality/FrameworkArduino/main.cpp`, however, the other errors, seem to be duplicate definition of `SysTick_Handler, SystemInit, SystemCoreClock`, they are defined again in hc32's code, but stm32 library has alternative implementation.
I am stuck on this, not sure how to proceed. Could you give me some suggestion? Thanks

classicrocker883, 2022-09-16:
@audynamo by your wording u seem much more informed than me lol. I did try to do what I said by replacing the files from the H32 folder using Eclipse. I tried replacing every single like file, but the issue I ran into was the included headers aren't all the same so it makes it difficult. so then I tried doing a few at a time but for something like this u can't cut corners. 

I think it may be totally doable going file by file, line by line lol... just takes time trial and error.

but that's through the Eclipse environment, which is so hard to use.

the way you're going about it is much better, by bringing in the hc32 files and replacing. what may help is that in the pc system files in a .platformio folder, there are libraries like the `libmaple`. so it may be calling for those to be implemented. and be sure to look out for any hidden files too.

and you're doing all this in VS Code correct? 
so far the only errors are the ones saying "first defined here"? I do recall that when platformio first opens the folder, it generates some files.

try removing the folder from the workspace and putting it back in. I do also recall there is a folder that lists all the files it links to for the .o and hidden  files can't be searched using vscode I think. if u can there is a good program that compares files and folders called Meld it that may help

classicrocker883, 2022-09-16:
@audynamo also u may have to remake the make file.

do you know how to do that? in the terminal u do

>sudo make

there may be other commands but that's the basics

audynamo, 2022-09-16:
Hi @classicrocker883, I use VSCode's platformio to build. The thing I notice is while building `MriscocProUI`, it did download the corresponding library, and from the error message, the duplication comes from `SrcWrapper` library's `clock.c` and `system_stm32yyxx.c` files. I think this library is detected by platformio itself and I have no idea how to exclude some/all of its files.
I think maybe alternatively we have to modify `hc32f46x_interrupts.c` to point to SrcWrapper's implementation?

classicrocker883, 2022-09-17:
yea see if that will work.
I took a look through some of the folders and buildroot/shared/scripts/proui.py has

```
stm32f1 = _GetMarlinEnv(MarlinEnv, 'MCU_STM32F1')
stm32f4 = _GetMarlinEnv(MarlinEnv, 'MCU_STM32F4')
stm32g0 = _GetMarlinEnv(MarlinEnv, 'MCU_STM32G0')
gd32f10 = _GetMarlinEnv(MarlinEnv, 'MAPLE_STM32F1') 
```

what the author did was separate the Aquila G32 chip from the standard stm32f1, when I went through Alex's code I couldn't find a whole lot of difference, that it used the same .ini with one or two build flags added, like ' -DTEMP_TIMER_CHAN=4'. there must be some other differences because I tried using all kinds of ways to get this ProUI to work with the G32, it will compile, and boot, but that's as far as it can go. I'm not sure what it's missing. I could try taking his main folder and put the ProUI in it and see if that works, as long as that compiles.

but for the H32, that may have to do with the clock.c.

check the .ini files like stm32f1-common and platform.ini and see if u can add those files directly

classicrocker883, 2022-09-17:
@audynamo and I wanted to ask what source code are u using? because if it compiles fine but doesn't boot after flashing, it might work better with Alex's main code, because I haven't been able to get it to work after flashing for the Aquila G32

audynamo, 2022-09-17:
@classicrocker883 I am using all folders inside `https://github.com/alexqzd/Marlin-H32/tree/main/firmware/Sources` except `Marlin`, and exclude `drivers/board` as it seems all files there look similar with files in stm32 library. Maybe I am wrong. And I move `board_tim0.h` file inside `drivers/board` to `drivers/inc`
I am not familiar with all STM32 chip stuff, this is just a dirty mod, maybe you can have a thorough review

classicrocker883, 2022-09-19:
i think that the hc32 stuff replaces all the STM32 files needed. or most. but sure i can take a look through

classicrocker883, 2022-09-21:
in src/HAL/STM32/HAL.cpp

```
// HAL initialization task
void MarlinHAL::init() {
  // Ensure F_CPU is a constant expression.
  // If the compiler breaks here, it means that delay code that should compute at compile time will not work.
  // So better safe than sorry here.
  constexpr int cpuFreq = F_CPU;
```
### F_CPU come up with an error , does this give any insight?

>#define F_CPU SystemCoreClock
Expands to:

SystemCoreClock
expression must have a constant valueC/C++(28)
HAL.cpp(70, 27): the value of variable "SystemCoreClock" (declared at line 51 of "/home/andrew/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/system_stm32f1xx.h") cannot be used as a constant

audynamo, 2022-09-21:
From the compiling flag of `MriscocProUi`
```arm-none-eabi-g++ -o .pio/build/STM32F103RE_creality/src/src/module/tool_change.cpp.o -c -Wno-register -std=gnu++14 -std=gnu++14 -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-use-cxa-atexit -g3 -fmax-errors=5 -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=60104 -DSTM32F103xE -DSTM32F1 -D__MARLIN_FIRMWARE__ -DNDEBUG -DTIM_IRQ_PRIO=13 -DADC_RESOLUTION=12 -DHAL_SD_MODULE_ENABLED -DSS_TIMER=4 -DTIMER_SERVO=TIM5 -DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8 -DMCU_STM32F103RE -DBOARD_F_CPU=72000000L -DSERIAL_RX_BUFFER_SIZE=128 -DSERIAL_TX_BUFFER_SIZE=64 -DUSART_RX_BUF_SIZE=128 -DUSART_TX_BUF_SIZE=64 -DSTM32F1xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_GENERICSTM32F103RE -DBOARD_NAME=\"GENERICSTM32F103RE\" -DHAL_UART_MODULE_ENABLED -DHAL_PCD_MODULE_ENABLED```
and this post https://qgit.de/melch/Marlin/commit/56462cf082e8bf2bae6cb288c5daa4b77289cfb0?style=split
I think we can add the flag '-DBOARD_F_CPU=72000000L' and revisit there to add more flags if applicable.

classicrocker883, 2022-09-21:
[image]
[image]

you may have to add the paths to the core files here

audynamo, 2022-09-21:
Hi, are you doing `adding h32 related code from Marlin-H32 into MriscosProUI`? I am trying to build `Marlin-H32` with `HAS_TMC_UART` enabled, which in turn needs `TMCStepper` library. However it is really painful as it won't compile and I think another library `SPI` is needed.
If I understand correctly, when using platformio to build `MriscosProUI` it does scan the dependency and both `SPI` and `TMCStepper` libraries will be compiled properly. I would prefer to add h32 code into MriscocProUI project.
The only thing I am concerned is that as maple/stm32 library is compiled by platformio, how can it be co-exsist with h32 related code?

classicrocker883, 2022-09-22:
I've been working on just getting it to work with the G32. If you'd like to make a pull request I can see about picking up where you are. I've tried every which way and it will compile fine, it flashes, but it stops loading after the bootscreen.

as for the H32 it can 'co exist' by skipping over the files. just remove the stm32 build flags and add hc32 ones

audynamo, 2022-09-22:
could you please create a branch of your changes that I can do a pull to have a look? I failed to make it integrating working in MriscocProUI repo. Thanks.

audynamo, 2022-09-24:
How to create a pull request in your forked repo? I downloaded from here https://github.com/classicrocker883/MriscocProUI.git, create a local branch `AquilaX2-H32` and made some changes, how can I push this branch to github to be visible to you? I tried and got
```
git push --set-upstream origin AquilaX2-H32
Username for 'https://github.com': xenway@gmail.com
Password for 'https://xenway@gmail.com@github.com':
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/classicrocker883/MriscocProUI.git/'
```

classicrocker883, 2022-09-24:
I think what u have to do is on your github, create a fork repository. then you can request a pull

classicrocker883, 2022-09-24:
oh for VS code, you have to do some things to enable authentication. I forgot exactly what to do but something about a Token in the advanced settings, and you need to enter your information into the vscode terminal. I'll edit to see if I can find it

classicrocker883, 2022-09-24:
I have a bunch of different folders / branches of what I tried getting it to work.

I wouldnt know which one I'd branch off to it's a big mess I might have to start over because none of it worked. it compiles fine but still won't load past boot. all I really did was copy some of Alex's source into it, I even tried putting the newest Marlin into Alex's main folder which didn't work either.

if u want to see whatever I did ill post one that should technically have worked without issue.

audynamo, 2022-09-24:
I got some questions:
1. there are some redefinitions such as `PIN_MAP, boardADCPins` which are declared as `extern`, especially for `boardADCPins` I couldn't find any usage, is it been used in maple? In `Marlin-H32` project, this variable is just declared, not used.
2. Do you include hc32f46x_flash.ld file for the final linking stage?
3. I can post my diff file here, and the error I got:
```
 *  Executing task in folder MriscocProUI-Github: platformio run 

Processing STM32F103RET6_voxelab_aquila_G32 (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Processing Professional Firmware requirements
GD32F1 Architecture detected
Unified Mesh Bed Leveling detected
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple @ 3.10000.201129 (1.0.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SoftwareSerialM @ 1.0.0
|-- TMCStepper @ 0.7.3
|   |-- SoftwareSerialM @ 1.0.0
|-- STM32ADC @ 1.0
|-- EEPROM
|-- USBComposite for STM32F1 @ 0.99
|-- Wire @ 1.0
|-- Servo(STM32F1) @ 1.1.2
Building in release mode
Compiling .pio/build/STM32F103RET6_voxelab_aquila_G32/src/src/inc/Warnings.cpp.o
Compiling .pio/build/STM32F103RET6_voxelab_aquila_G32/src/src/module/settings.cpp.o
Linking .pio/build/STM32F103RET6_voxelab_aquila_G32/firmware-20220925-092312.elf
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .rodata by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .rodata by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RET6_voxelab_aquila_G32/firmware-20220925-092312.elf section `.text' will not fit in region `FLASH'
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .rodata by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .rodata by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .rodata by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/Users/xinweix/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 35904 bytes
.pio/build/STM32F103RET6_voxelab_aquila_G32/FrameworkArduinoVariant/wirish/syscalls.c.o: In function `_sbrk':
[... cut ...]
--------------------------------  --------  ------------
STM32F103RET6_voxelab_aquila_G32  FAILED    00:00:23.260
=================================================================== 1 failed, 0 succeeded in 00:00:23.260 ===================================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```
[diff.txt](https://github.com/classicrocker883/MriscocProUI/files/9639738/diff.txt)

classicrocker883, 2022-09-24:
I'm actually unsure what the differences are with Maple and regular STM32. I thought that some things are not included, but I find that Maple .bin compiles are larger than without Maple. for some reason when I compile mine I think 27 libraries are loaded instead of 28 which is what u have. and when I do that for regular STM32, there are I think 10 or 12 libraries. so I'm not sure what is different or why I read Maple is slimmed down.

for the ld file I used the standard creality.

even though these chips can handle 512k and 64k flash, check the ld file to make sure it includes that, because I found it was changed to half those values.

classicrocker883, 2022-09-24:
[if you still need help with linking Github ](https://youtu.be/f7xZivqLZxc)

audynamo, 2022-09-25:
Sorry I forgot to mention:
1. I didn't include the `framework` folder as I think they are copied from libmaple library. That is why I tried the maple build, do you need to include them when building STM32?
2. I still don't understand how Marlin-H32 works regarding these redefinition varaibles found such as `boardADCPin` and `PIN_MAP` as I think they are all H32 related essential variables and I can't find the usage of `boardADCPin` across the whole repo. Is there some hidden file being compiled to use it?
3. If we try to reuse STM32 library where there are these variables re-defined, I don't know how to get rid of such redefinition during linking stage.
4. I think `startup._hc32f46x.S` is essential to boot up the hc32 chip, and in Marlin-H32 project, it is using another `main.cpp` as the main function, instead of main function inside Marlin folder.

classicrocker883, 2022-09-27:
are you saying `startup._hc32f46x.S` is using another `main.cpp` in the H32 project in the Eclipse IDE? in the file that `#includes` or calls to a file or parameter, right click it and go to definition. that  should bring you to where it is.

classicrocker883, 2022-09-27:
[image]

audynamo, 2022-09-27:
Do you mean you use a vanilla Marlin 2.0.x fix repo and add hc32 related code into that repo, such that it can compile successfully?
From the screenshot, it shows the board is `megaatmega2560`. Shouldn't it be `stm32`?

Inside `startup_hc32f46x.S` line 367 it is calling `bl main`, I assume this main is the file in `Source/main/main.cpp` in Marlin-H32 project. However, there is no main function called in Marlin 2.0.x fix repo. Maybe I misunderstand it.

classicrocker883, 2022-10-03:
so this is the vanilla marlin bugfix, correct. this is the standard include tree as it is at the bottom of `platform.ini`. and if u can add the hc32 files into a vanilla environment, it may just work, but what I was getting at was you can manually change the main.cpp to what you need, with build flags you can just add or remove all the libraries and files, like the core of the hc32.

sorry for the delay in reply, there is a new update for Mriscoc ProUI and ive been getting it to work for G32 and N32, 

if you like, clone the repo which is the main fork for this MriscocProUI, and add put all the H32 files u need in there.

there is a debug check u can do which may help in finding errors, but like it may show errors that really arent there or a problem as i experienced. if u click the PlatformIO tab on vscode, then under Project Tasksm under the default env, go to advanced, then do check, or any of the other verbose ones.

johnnykang, 2022-10-27:
what's the chance is getting a H32 release so far?

classicrocker883, 2022-10-29:
@audynamo hey you may be better off using this repo here https://github.com/SlevinSkrreeek/Marlin

it used the current Marlin with Alex's configuration. u may have to do some file changes or editing but it may be easier implementing the hc32 files into that, because I haven't been able to get the hc32 to work in platformio, so at least we can have the ProUI (without some of the extra features) but with the updated Marlin like linear advance and input shaping. is this the way we should take you think? or have u been able to get hc32 to work in vscode? if u can post or link your findings and files i can look and see what may need to be done.

classicrocker883, 2022-10-29:
@johnnykang so far the most feasible thing is getting the hc32 with the updated Marlin. it may not have all the extra features in the ProUI, but it will have some and things like linear advance and input shaping (*it should work with those)

right now i was working on g32 and n32 and they seem to be good, so i can start with the h32 and i should have something in the next week or two.

classicrocker883, 2023-03-31:
https://github.com/shadow578/Marlin-H32

please check this out. i was able to successfully compile a  .bin file, however I do not have an HC32 chip to test. ill be trying to see if i can get the proui to work.

@audynamo

closed 2023-05-27 by classicrocker883