classicrocker883/MRiscoCProUI #2: Will not install

schuh8, 2022-09-20:
Attempted to install " G32_maple-Pro UI-newtest.bin" on my Aquila C2 today. Have tried (on properly formatted SD) the bin file under the root and also under a folder "firmware".

When I power up I get either the Voxlab logo on a red banner or a white screen. The white screen has "Voxlab" on it and appears to be loading the firmware (i.e. acts similar to loading other firmware) but continues loading indefinitely with no result.

I then attempted to reinstall Alex's firmware with no problem.

 **To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

**Additional context**
Add any other context about the problem here.

classicrocker883, 2022-09-21:
Yes this is an issue im working hard to figure out.

its the oddest thing, where it will compile fine and work for every other board out there.

unfortunately this is as far as everyone else gets and has been a priority.

same thing for the newest Marlin and Mriscoc's ProUI, even though there are source files for the Aquila, it just doesnt want to work.

if you want to check periodically for more test files. otherwise it works great for the Creality 4.2.7 and 4.2.2

314ru, 2022-10-15:
what is the difference between the versions: RC, RC_maple, RE?

whitefiretiger, 2022-10-20:
> Yes this is an issue im working hard to figure out.
>
> its the oddest thing, where it will compile fine and work for every other board out there.
>
> unfortunately this is as far as everyone else gets [...]

I'm in the same boat, none of the G32 versions will properly install. I had actually tried to do this project myself before finding you were already working on it before me, I had the same issue that no matter what I tried it just wouldn't install. I know on the Alex  in the platformio.ini it was on default_envs = STM32F103RET6 but that doesn't seem to exist on the newer Marlin and that's the only thing I could confirm as different. ill defiantly be checking back to see if you get it figured out, I'm trying to avoid buying a V4.2.7 mb for better firmware while my G32 works.

classicrocker883, 2022-11-01:
@whitefiretiger hows it going? so far Ive been able to get it all to work with the G32 and N32. that is UBL works, but without some extra features, and Manual Mesh works all the way. So something with the BL touches may have issues with the full capabilities of this ProUI firmware. its something Mriscoc may have to figure out in his next release. but ill be trying to pinpoint the issue in the meantime.

classicrocker883, 2023-04-03:
cant use **RE** type chips with 512k. must fit within 256k **RC** chips for Aquila Boards specifically in `default_envs`. actually *'firmware'.bin* files must be below 89% of 256k for chip memory. the **aquila.id** sets the maximum memory allocated for boot loading and flashing.

closed 2023-04-03 by classicrocker883