classicrocker883/MRiscoCProUI #3: BL Touch

schuh8, 2022-09-25:
BL touch blinks constantly, whether in use or not. Actual function is perfect just constant blinking as long as machine (Aquila w/ 4.2.7) is turned on.

classicrocker883, 2022-10-18:
@schuh8 There is an updated firmware that is posted in the releases. Has this been fixed or is it still ongoing? Also, I do notice it blinks when first flashed, and the light is red when first started, then does a self-test and stays lit while idle. it has red+blue LEDs which make a purple (for CR touch) while in normal mode. is there anything else to make note of or everything else works as it should, its just the blinking while idle?

schuh8, 2022-10-19:
Haven't had a chance to install the update yet, will do soon and report back. BTW the the GD32 board was posted today. It should arrive in 3 or 4 days. Cheers.

classicrocker883, 2022-11-01:
hey im just wondering how its going. so far I have noticed a similarity of what wants to work and what doesnt. i think it revolves around the BLtouch, because I'm able to get it to work sometimes and sometimes not. like if i have UBL enabled, it wont work with the extra ProUIex features. but it will when its disabled. then when i change it to Bilinear from UBL, the ProUIex features do work, however the BL or my CR touch wont deploy. that is, it works when first turned on, it passes the self test. but when I go tot the menu option to deploy, it just raises the Z and thats it. i havent tested any more than that so far.

so it all may have something to do with Mriscoc's library. Im not sure how to access it, such as its archived into a file and I dont know how to 'unzip' it. or if its actually meant to be or not. i think he made it so no one can like steal it. before releasing his source code, he was reluctant, so I guess we can just hope for him to find and make the fixes in each of his new updates.

closed 2023-05-27 by classicrocker883