classicrocker883/MRiscoCProUI #6: Extruder not responding

bl-krupa, 2022-10-24:
**Describe the bug**
Aquila X2 with 4.2.7 board. Flashed most recent bin file for this board posted on 10/23. Extruder motor not responding to commands. Went to Prepare and Move axis. Ensured nozzle was heated, attempted to have e motor push filament. Nothing. Attempted to have it retract filament. Nothing. Attempted to disable steppers, nothing.

classicrocker883, 2022-10-28:
thank you for pointing this out, i have found the issue, it seemed something was changed in the update and now its been fixed and the extruder should be working again. look out in the next day or 2 ill post the updated files.

classicrocker883, 2022-10-30:
@bl-krupa hey i uploaded the newer version, its fixed

bl-krupa, 2022-10-30:
Thanks! I'll try it out soon.

closed 2022-11-01 by classicrocker883

bl-krupa, 2022-11-01:
I flashed the new version and couldn't test the issue because I couldn't even auto home. Got a homing failed message. Tried it all again, same. Flashed back to previous version.