classicrocker883/MRiscoCProUI #7: Have you successfully set up the backlight timeout for DWIN displays?

adam3654, 2022-10-24:
I am struggling with setting up a backlight timeout  with the professional firmware and a stock e3v2 DWIN display; and a Google search led me to your discussion on the marlin branch, and I found your fork here. If you got it working could you point me to the changes you made?

Thanks,
Adam

classicrocker883, 2022-10-27:
hi sorry for the delay. I think I have what was changed on the issues tab of Mriscoc. I'll look it up. but MriscocI-Fall is my newest branch with the update to Marlin, feel free to look through the code, search for `LCD_BACKLIGHT_TIMEOUT_MINS`

`ui.refresh_backlight_timeout`

classicrocker883, 2022-10-27:
in file e3v2/proui/dwin.cpp

>  #if LCD_BACKLIGHT_TIMEOUT_MINS
  void SetTimer()   { SetPIntOnClick(ui.backlight_timeout_min, ui.backlight_timeout_max);
   }
  #endif

in the tune_menu AND advanced/control_menu

> #if LCD_BACKLIGHT_TIMEOUT_MINS
      EDIT_ITEM(ICON_Box, MSG_SCREEN_TIMEOUT, onDrawPIntMenu, SetTimer, &ui.backlight_timeout_minutes);
    #endif

under `EachMomentUpdate()`

>#if LCD_BACKLIGHT_TIMEOUT_MINS
    if (ui.backlight_off_ms && ELAPSED(ms, ui.backlight_off_ms)) {
      TurnOffBacklight(); ; // Backlight off
      ui.backlight_off_ms = 0;
    }
  #endif

in plot.cpp under `PlotClass::Update`
and in src/feature/bedlevel/ubl/ubl_G29.cpp
under `unified_bed_leveling::probe_entire_mesh`
>  #if LCD_BACKLIGHT_TIMEOUT_MINS
    ui.refresh_backlight_timeout();
  #endif

adam3654, 2022-10-27:
Thank you so much for your detailed reply! I had it working in the JyersUI flavor of the Bugfix Marlin Branch but couldn't translate it over to the ProUI. I will go through your outline as soon as I can. I have SKR motherboards on my printers so it is really nice to be able to turn the screen off when it isn't in use, because the fans turn off also and it can be in a silent stand-by state without actually being turned off.

adam3654, 2022-10-29:
I took a stab at adding the changes you listed, and after massaging the conditionals to pass the sanity check, I think I ended up applying too much of the common marlinui code for cr10 style lcd displays and am getting errors. When I did it in the JyersUI code I remember pulling blocks of code out of the common lcd code and bringing it into the DWIN.cpp file. I'm not sure if that's the right way to do it, but it worked. Next chance I get I'm going to try to squash all the commits I made to get it to work, and see if I can come up with a similar approach for the ProUI. I'll report back results.

classicrocker883, 2022-11-01:
so it worked for you? I havent had it enabled for the CR-10 style, which im aware that is what skr mini e3 may use. so have u been able to get it to work with that? did u have to change a lot of code? because as I understand some boards have PINS for enabling and disabling. im unsure if the skr mini uses such pins for example
>#if PIN_EXISTS(LCD_BACKLIGHT)
WRITE(LCD_BACKLIGHT_PIN, HIGH);
#endif

did u have to deal with pins?

adam3654, 2022-11-01:
I haven't been able to get it to work yet. I am using the SKR mini board with the stock Creality E3V2 DWIN LCD display (with modified cable) for this experiment. I may have been wrong with my use of the term "CR-10 style" as I don't think they even have a `BACKLIGHT_PIN`. I was trying to describe the LCDs that use the backlight pin and fit the description of what the original marlin code was intended to control. 

I have started working through what I did in the Jyers Fork code, and am trying to apply it to the ProUI code. The Jyers fork seems to redirect `ui` code from the marlinUI code into the DWIN code by redefining some functionality (i.e. `void MarlinUI::update() { CrealityDWIN.Update(); }`).  I was able to pull in some code from the `marlinUI.cpp` file and massage it into the DWIN code with a relatively small amount of work.   It doesn't seem like it is going to be as easy to apply the same approach to the ProUI code.  The code isn't as similar as I would have hoped, and I am not very good with C++ to begin with.

 I didn't get much time over the weekend to play, but hopefully in the next few days I'll get another chance to sit down at the computer for a couple hours.  Next chance I get I will look through your post on the ProUI Issues discussion again and try to better understand it.  Thanks again for the help.

closed 2023-05-27 by classicrocker883