classicrocker883/MRiscoCProUI #8: How to build for N32?

KimmoHop, 2022-10-31:
When building for Aquila N32, which environment and board should be selected in platformio.io?
N32F103RE_voxelab_maple gives
`Error: Build environment 'N32F103RE_voxelab_maple' is incompatible with BOARD_CREALITY_V427. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple`

Using BOARD_VOXELAB_AQUILA gives
`Error: Build environment 'N32F103RE_voxelab_maple' is incompatible with BOARD_VOXELAB_AQUILA. Use one of these: GD32F103RE_aquila, GD32F103RC_aquila, GD32F103RE_aquila_maple, GD32F103RC_aquila_maple, GD32F103RE_voxelab, GD32F103RC_voxelab, GD32F103RE_voxelab_maple, GD32F103RC_voxelab_maple`

What is the difference between Mriscoc-fall and MriscocProUI?

This is very confusing, when there is not enough description...

classicrocker883, 2022-11-01:
Mriscoc-fall is the one most updated. I wrote that in the readme in the other branch. MriscocProUI was what i was working on before the latest fall update by mriscoc. So i branched off that and worked mainly on that. I will probably delete the proui branch and when there is another update merge the current Fall one with that so theres no confusion.

right now the Fall branch works when having `N32F103RE_voxelab_maple` for the default_env in platformio.ini with `MB BOARD_VOXELAB_AQUILA `

if that doesnt work what you can also do is add `-DVOXELAB_N32` to `env: GD32F103RE_voxelab_maple` or `STM32F103RE_creality_maple` (or with _RC_) build_flags after `-DTEMP_TIMER_CHAN=4`  in file --> **ini/stm32f1-maple.ini**
and use BOARD_CREALITY_V4 if you use _creality_maple_

KimmoHop, 2022-11-01:
Marlin way of defining compatible environments is, well, awkward ;D

Thanks a lot, now MM build without ProUIex goes through (after commenting out line with onDrawUBLSlot()). I might even try it ;)

closed 2022-11-01 by KimmoHop

KimmoHop, 2022-11-02:
@classicrocker883 ProUIex (at least toolbar, not sure what else there should be) seems to work fine with N32 👍