classicrocker883/MRiscoCProUI #10: The y-axis is disabled.

314ru, 2022-11-05:
I use version 2.1.3 gd32 MM. After about 6 minutes of printing, the axis stops moving. At the same time, the X-axis works. What could be the matter? The delay of turning off the screen was set to 10 and 99 minutes. The same result. 
[image]
[image]
[image]

314ru, 2022-11-06:
returned to Alex's firmware. It was possible to print the part. But LA doesn't work. I like the new firmware - a lot of useful features, but for some reason there is a bug...

classicrocker883, 2022-11-06:
could u clarify about the axis not moving? it's just the Y that will not move? and is the lcd timeout related?

There is another timeout function like disable steppers which is set to 6 minutes, but this shouldn't be related and happen while printing. but 6 minutes is what it's set at, so that does seem to relate. 

Question, how are your stepper motors? are they more noisy when printing? because I've got some comments saying this. and is there anything else to add?

314ru, 2022-11-07:
After ~6 minutes of printing, the Y axis turns off. it can be moved by hand. After canceling printing, the printer turns on the axis and goes home. I did not find a direct connection with the operation of the LCD.

Is there a limit of 6 minutes in the Alex firmware?

I didn't notice that the engines became noisier. Subjectively, everything is the same as before.

classicrocker883, 2022-11-08:
there really isnt a limit that should effect while printing. if it is exactly 6 minutes then that is related. otherwise I think it could be related to something else.

I since then changed some things in the files.

i uploaded newer updated firmware. the problem should be fixed, the motors should not be noisier. if they are or if the Y axis turns off after any time let me know.

314ru, 2022-11-10:
> @314ru I have recreated the issue. after a certain amount of time after homing, the screen gets artifacts and the Y axis stops taking commands. ill be sure to find a fix as quick as possible.

1. The day before yesterday I installed an updated firmware. I started the printer yesterday. The error of disabling the Y axis remained.
2. The engines are running quietly - there are no problems

classicrocker883, 2022-11-13:
so i realized that Input_shaping has been causing issues. I left it disabled now and this problem seems to have gone away.

@314ru Ill upload some more recent firmware files without input_shaping, as it isn't quite compatible with the ProUI since its so new.

314ru, 2022-11-17:
> so i realized that Input_shaping has been causing issues. I left it disabled now and this problem seems to have gone away.
> 
> @314ru Ill upload some more recent firmware files without input_shaping, as it isn't quite compatible with the ProUI since its so new.

is it possible to turn off Input_shaping in the printer menu? So that the crash would disappear

classicrocker883, 2022-11-18:
@314ru i dont believe so. i mean you can try changing the settings to the lowest. but I will upload an updated file, so by the time you read this it should be there or soon after.

314ru, 2022-11-18:
> @314ru i dont believe so. i mean you can try changing the settings to the lowest. but I will upload an updated file, so by the time you read this it should be there or soon after.

the bug didn't show up anymore. I continue to test the firmware, actively printing details. I will definitely test LA a little later. 

**Thanks for the firmware!**

closed 2023-01-25 by classicrocker883