classicrocker883/MRiscoCProUI #19: Question on compiling for GD32 Voxelab Aquila.

bstork0807, 2023-02-12:
Not sure this is the correct place for this but I believe I have tried all options when compiling. I have a Voxelab Aquila G32 and I can take you pre-compiled which works great. I just wanted to tweak some stuff and set it in the firmware so I wouldnt have to fix all the settings after reflashing.. What is your recommendation to do this.. I have VScode setup ready to go..

classicrocker883, 2023-02-15:
hey @bstork0807, yes you can use VS code to compile if you know how to do all that you should be good to go.

However some things to take note. The Configuration.h and Configuration_adv.h files directly in the Marlin folder should be replaced with the ones under Configurations. you'll see MM and UBL or default.

Then you may have to change the default_env in the platformio.ini file to your correct chip. just be aware that some tweaks (enabling options) cause the file to be too big and you won't be able to flash it properly.

I'm curious as to what sort of things you'd like changed?

bstork0807, 2023-02-15:
I have most of what you said... Copied the Configuration.h and Configuration_adv.h files for ubl..  When it comes to the platformio file I have the GD32 chip and have used your GD32-UBL_ProUI_BLT.bin and it works great.. Any info on how you set the platfirmio.ini and what you pick when you autobuild would be greatly appreciated.   As far as what i want to change, at this point want to make a bin file that will configure all my settings so I dont have to go in after the fact. For example extruder esteps..

classicrocker883, 2023-02-17:
`STM32F103RC_voxelab_maple_GD32` this is what i put after the `default_envs=` in platformio.ini
personally i dont use the Marlin Auto Build extension. if you want to change the settings with your own just go into the Configuration.h file do ctrl-f and type whatever you'd like to change. my config files are saved in the configurations folder under Andrew427. however that is specifically for the 427 board which if you want the UBL with ProUI, i suggest going into those specific Config files, copy and paste into the main marlin folder, the ones youre looking for have _Pro_ at the end, so copy those but dont forget to remove "_Pro_". because for the Aquila Boards, they can only hold upto 256k memory, not 512k like the 427. that is why some options or features aren't enabled for the _UBL_ProUI_.

so if youre looking to reenable some things, you would have to disable some other things, or have some features removed from the UI, if that makes any sense. are u good? anything else u need?

bstork0807, 2023-02-17:
OK seems to be getting closer and thank you for all your help so far.. For the board I have put in BOARD_VOXELAB_AQUILA and in the platformio I have tried both STM32F103RC_voxelab_maple_GD32 and GD32F103RC_voxelab which resulted in the following.. thinking maybe a config with in vscode.. any ideas would be appreciated.
[image]

bstork0807, 2023-02-17:
Ignore the previous message.. it turns out the aquila256k.ld script is named aquila_256k.ld in the directory for some reason.. but now that I got past that point I noticed the following error in the output section. I am thinking it has to do with removing the PRO.. how do I do that?
[image]

I am now getting this at the end of the compile with the maple

[image]

classicrocker883, 2023-02-18:
i had disabled several things, not exactly sure which ones but like the nozzle clean, if you `Select to Compare` the files, the same configuration or configuration_adv files, you should be able to see the differences. the thing about the aquila is yeah it wont hold more than 89% of the 256k. i remember seeing that same or similar error, but im not remembering what the cause or fix is.

can you attach your configuration.h and configuration_adv.h files so i can look? or was there any other file you changed (*besides the platformio.ini*)

classicrocker883, 2023-02-18:
ok i recreated the issue by commenting out `#define ProUIex 1` in configuration.h line 3191.

that should be enabled. that is the ProUI, if it is disabled, you will have to also disable `HAS_GCODE_PREVIEW`

bstork0807, 2023-02-19:
Attached are the configurations and platformio
[configs.zip](https://github.com/classicrocker883/MriscocProUI/files/10775485/configs.zip)

Lord-Memester, 2023-02-19:
I was so close to being done. Then I got this:
[image]
Some sort of ROM overflow. I truly don't know what to do about that.

Lord-Memester, 2023-02-19:
I was successfully able to build it! I cloned the repository, so I'll upload my works as a fork.
[image]

classicrocker883, 2023-02-19:
the overflow means it is too big, there are somethings you would have to disable, start with the least used like the options under where `ProUIex` is: the lockscreen, esdiag, ect...

there are a couple other ways to save space like commenting out some things in the *dwin.cpp* file under **src/lcd/e3v2/proui** i know that is pretty vague if you dont know much about code, but basically how the UI user interface looks and how the menus are constructed are coded in that specific file. such as being able to view the PID plot graph while printing, that can be commented out to save space, also it has two different ways to view the Mesh grid, you can save space by having just one.

since you have almost 86% meaning u can squeeze 3% more space, there are some things you can go back to enable, but honestly its more trial and error figuring out how much byte size youre adding or taking away. 8600 bytes over can be one or two things, or several small things. so its good to start with disabling the larger ones first, because you can keep the smaller type options, you know the more options youre able to keep the better, so you at least _have_ them.

Lord-Memester, 2023-02-19:
Sorry, but what lines would those be in `dwin.cpp`? Having some trouble finding them.

bstork0807, 2023-02-20:
Good evening. Just curious if you were able to compare your files to mine. If you could share your config files that you used to build the gd32 proui bin files I would have no issue building it myself and comparing. Thank you again.

Lord-Memester, 2023-02-21:
> Good evening. Just curious if you were able to compare your files to mine. If you could share your config files that you used to build the gd32 proui bin files I would have no issue building it myself and comparing. Thank you again.

Do you mean me or @classicrocker883 ?

bstork0807, 2023-02-21:
I was asking classicrocker883 but looking for help on what was commented out to save the space.

classicrocker883, 2023-02-21:
> Sorry, but what lines would those be in `dwin.cpp`? Having some trouble finding them.
> 
> Edit: Is it lines `267-348`?

I was thinking more like lines `1626-1664`
*//Temperature (PID Tuning Graph) Plot During Printing*

you'd have to comment out the two menu options under `draw_tune_menu`
**PID Plot bed+nozzle**

bstork0807, 2023-02-21:
Andrew. When you compiled the gd32 version of the bin files did you reduce to 256k or 512k?  I heard even though the gd32rct version has 256k it can handle the 512k software

classicrocker883, 2023-02-22:
@bstork0807 yes it had to be reduced to 256k, its true these chips **should be able** to handle 512k as the specs say, however that is not the case when flashing the firmware. for whatever reason, voxelab has probably set a limit on these chips and they cant be compiled to be more than 89% of 256k. you can see for yourself and  in platformio.ini set `default_envs=` it to `STM32F103RE_voxelab_maple_GD32` just by changing the C to an E. the reason why I had put GD32 or N32 at the end was so its easier when compiling using the WORKFLOWS actions.

that was initially a problem for me - getting the firmware to flash properly. I didn't realize it had to do with size of the file until after lots of trial and error.

but if youre able to somehow get more memory to work with these chips that would be great. it may have something to do with the bootloader, I think it was something voxelab hard coded in their chips because if the firmware file is too big it could brick the chip, so their thinking is its a safeguard. for that, u may need special STM32 CubeMX software or similar.

anyway, ill take a look at the files if youre still having trouble compiling.

Lord-Memester, 2023-02-22:
> @bstork0807 yes it had to be reduced to 256k, its true these chips **should be able** to handle 512k as the specs say, however that is not the case when flashing the firmware. for whatever reason, [...]

Should we attempt to contact Voxelab support about this? At least in the past, they've been responsive enough that they might be able to offer an explanation for this.

bstork0807, 2023-02-22:
I havent tested as of yet but was able to compile using the non-maple library because it takes up less space.. what is the reason the maple library is used?? Below is the successful compile.
[image]

classicrocker883, 2023-02-23:
> I havent tested as of yet but was able to compile using the non-maple library because it takes up less space.. what is the reason the maple library is used?? Below is the successful compile. [image]

yeah it does use less, because it doesn't use as many libraries. I'm honestly not exactly sure the differences, all I know is the Aquila boards won't flash with non-Maple firmware.

did you try flashing it? and how did it go?

because in my tests I haven't been able to get it to flash properly. so I stuck with the Maple for `default_envs`

classicrocker883, 2023-02-23:
> 
> Should we attempt to contact Voxelab support about this? At least in the past, they've been responsive enough that they might be able to offer an explanation for this.

you can try and ask them, you might get an answer. but I doubt it will be one to unlock the chips potential. even so, it may not be worth the hassle. sure go ahead and ask why they limited the chips memory or ram or rom whatever it was. or it could be something with the chips manufacturer. while you're at it, ask why can only Maple environments work for the chip, and if there is a way to use non-Maple. if that makes sense.

I think a good source of knowledge would be Mr Alex alexqzd himself. he was in contact with Voxelab for a while and was able to get all the chips working. I'm sure he would know the answers, but I haven't heard or seen a lot of him around lately...

bstork0807, 2023-02-23:
Well figured it out.. looks like it was my mistake the entire time.. The files have not been put in the released folder and that has been where I have been pulling the directories from.. I was able to compile successfully using the Pro files below. I just need to load them on the printer and had a print job running this morning. I will let everyone know how it works. Looks like the following had to be disabled: 
//#define G26_MESH_VALIDATION
//#define EEPROM_CHITCHAT 
//#define NOZZLE_CLEAN_FEATURE
//#define HAS_ESDIAG 1
//#define SET_PROGRESS_MANUALLY 
//#define POWER_LOSS_RECOVERY
[image]

Lord-Memester, 2023-02-23:
I actually haven't had much luck compiling with ProUI. If that's what
you're going with, please let me know how it goes.

bstork0807, 2023-02-23:
So what issues did you have compiling.. just so I know what I am looking for..

Lord-Memester, 2023-02-23:
[image: image.png]
Size overflow.

Lord-Memester, 2023-02-23:
Size overflow.
[image]

bstork0807, 2023-02-23:
did you copy the configuration pro.h and configuration_adv pro.h and rename them to configuration.h and configuration_adv.h?

Lord-Memester, 2023-02-23:
No, I wanted them to be custom, so I'm just messing about with the normal config files.

bstork0807, 2023-02-23:
yes.. i was trying to customize as well but if you use those files as starting points it should work for you

Lord-Memester, 2023-02-26:
@classicrocker883 I just compiled the firmware with a non-maple build and the output was a file less than 201 KB! I haven't tested it yet, but the file size at least seems promising. I'll edit this comment to let you know how it goes.

bstork0807, 2023-02-26:
I was able to do the same just haven't had time to load on the printer as my son is printing parts for school robotics competition.

Lord-Memester, 2023-02-26:
> I was able to do the same just haven't had time to load on the printer as my son is printing parts for school robotics competition.

Wait, that wouldn't happen to be a FIRST Robotics competition, would it?

bstork0807, 2023-02-26:
It is the FRC 2023. Have a practice comp in about a week.

Lord-Memester, 2023-02-26:
Ooh, what team number? Depending on where you live, we might end up facing off!

bstork0807, 2023-02-26:
3171 Westhampton Beach.  Long Island NY.  We are going to SBPLI 1 @Hosfstra and then we will be in Pittsburgh PA

Lord-Memester, 2023-02-26:
Far from there. I'm in the Midwest, and it's unlikely my team will make it far enough along in the competitions to meet. Oh well.

Lord-Memester, 2023-02-26:
> 3171 Westhampton Beach. Long Island NY. We are going to SBPLI 1 @Hosfstra and then we will be in Pittsburgh PA
> 
> We did go to Houston last year.
> 
> Where are you from?

You may want to keep in mind that anyone on the internet can view this information, in case you didn't already know.

bstork0807, 2023-03-04:
Did you try and load the new compile?

Lord-Memester, 2023-03-04:
Depending who you're asking, I've since edited my comment. I haven't tried loading the new version on yet as I've been calibrating input shaping.

bstork0807, 2023-03-04:
I was asking you,sorry didn't see that you modified. I compiled with maple and it is working great. I am working on tweaking linear advance.

Lord-Memester, 2023-03-04:
> I am working on tweaking linear advance.

Oh, do let me know how it goes! I'm wary of going to more extreme measures to figure out the issue, so if it works for you on your (mostly stock) setup, I'd love to know.

classicrocker883, 2023-03-04:
> I was asking you,sorry didn't see that you modified. I compiled with maple and it is working great. I am working on tweaking linear advance.

for linear advance, I'd like you to know that in **src/module/planner.h**, you will see line ~200
`typedef struct block_t`

however in the MarlinFirmware files, it is `typedef struct PlannerBlock`.

it was at one time changed from block_t to PlannerBlock, but would not compile so I changed it back to block_t. now I don't know if it compiles correctly using the MarlinFirmware source code files, if it is deliberate or a mistake.

that is one of the only changes I had made regarding linear advance, so maybe this is something to look into.

Lord-Memester, 2023-03-04:
Interesting. This is the new marlin you recently combined with the previous, right?

classicrocker883, 2023-03-04:
I combined with the new Marlin under the branch `2023-Feb`, haven't tested but it compiled fine. (it has the struct block_t - not PlannerBlock)

Lord-Memester, 2023-03-04:
> I combined with the new Marlin under the branch `2023-Feb`, haven't tested but it compiled fine. (it has the struct block_t - not PlannerBlock)

Let me know when you've tested it, please!

classicrocker883, 2023-03-07:
@Lord-Memester are you able to compile your own firmware? because I saw in the newest update to Marlin there were specific changes to linear advance and to the extruder in the code. I just (on Mar-06) updated the 2023-Feb branch yet again and it compiles fine. 

I'm sure this would fix something! I'll post some .bin files probably sometime today. care to share what you use so I'll have that ready first? like the chip/board and UBL, manual mesh, 5x5, 7x7, ect...

Lord-Memester, 2023-03-07:
I can, but I haven't gotten around to it. I've been swamped with work, and I won't have much of a break until next week. For now, you can look through my fork, though I haven't incorporated the latest changes into my build.

Lord-Memester, 2023-03-07:
[image]
Recent compilation error. I don't want to run ProUI, and this seems to be causing issues.

classicrocker883, 2023-03-08:
@Lord-Memester I since fixed this, it should be working fine. Also i posted a couple quick pre releases for testing if you want to check it out.

Lord-Memester, 2023-03-08:
> @Lord-Memester I since fixed this, it should be working fine. Also i posted a couple quick pre releases for testing if you want to check it out.

I'll see if I have time to try the prereleases. If you're willing, would you specify what files you changed?

SrgWhoopsie, 2023-03-08:
i know im using a different board but i noticed there was not an answer for this above from a couple weeks ago, and i cannot get get the build env do its thing lol
[image]
what am i doing wrong?

Lord-Memester, 2023-03-08:
> i know im using a different board but i noticed there was not an answer for this above from a couple weeks ago, and i cannot get get the build env do its thing lol [image] what am i doing wrong?

Well, I have to say, if you're using the wrong board, you might want to start there. Try enabling verbose mode and then paste the output in here, or try making an issue in the main Marlin firmware repository.

SrgWhoopsie, 2023-03-08:
For what its worth, im not using the wrong board..... i ditched the stock board for the e3 mini, hence why its changed. 
Considering this is in the platformio.ini file i assumed there would be no issues. [image]

this issue happens as soon as i open the folder in vscode, doesnt matter if i change board or leave it on the STM32F103RC_creality_maple. 
This is why im confused, as i dont get this issue when opening any other source code folder for marlin ive downloaded from git only the proui stuff.

Mriscoc reference [image]

Classicrocker reference 
[image]

Or have i understood this wrong and mriscoc or classicrockers firmware connot be complied for anything other that a STM32F series chipsets?

Also sorry for the shotty photo, cannot screen grab ATM as im at work.

Lord-Memester, 2023-03-09:
I'm realizing I may not be the best person to answer, and we should probably wait for @classicrocker883 to chime in. However, I would like to suggest that you manually type the command to build the firmware in the terminal. I don't remember it off the top of my head, but you can specify which board you want it to build for even if the platformio.ini says something else. Maybe give that a try?

Lord-Memester, 2023-03-10:
> @Lord-Memester I since fixed this, it should be working fine. Also I posted a couple quick pre releases for testing if you want to check it out.

@classicrocker883 I have since compiled it with Linear Advance under the latest branch, and encountered the same issue. With my very limited time recently, it might be best that you test them yourself, but I can compile it with the debug flag enabled for LA and let you know what I see (if anything).

bstork0807, 2023-03-10:
I will check this weekend and post. Been crazy busy family and work life.

classicrocker883, 2023-03-11:
@SrgWhoopsie have you changed the Author's name? that error can come up if you change it from `STRING_CONFIG_H_AUTHOR "Miguel A. Risco-Castillo (MRiscoC)" `

also I haven't yet updated the BTT SKR Mini E3 config files if you are using those. ill get to it and probably by the time you see this youll see an update (but I am primarily working on the `Mriscoc-Feb` branch so youll see those changes there).

you must have that board under the `MOTHERBOARD` in Configuration.h, there are differences between each board and it has to match to the `default_envs` in platformio.ini, as far as i know there are a couple different types of chips for the SKR mini E3, so be sure that chip matches the `default_envs`.

classicrocker883, 2023-03-11:
@SrgWhoopsie are you trying to compile using `STM32F401RC_btt`?
you must go into **ini/stm32-common.ini** and comment out using # these two lines line that have "proui" in them
[image]

i updated the config files for the SKR mini, so be sure to look them over.

SrgWhoopsie, 2023-03-11:
@classicrocker883 so when i first tried compiling i did change author, but since i have left everything as pulled from git, im just tried opening up your feb branch in vscode via github desktop, (i just cloned it to my pc inside git desktop) so its exact as you branch is, i get the failure as soon as vscode opens, then if i copy the config files over from the config folder, and try to compile i still get the same issue, 

 
[image]
i added a screen grab with verbose 

to clarify i am using the STM32G0B1RE env as i have the v3, tho i cannot remember if the board has a STM32G0B0RET6 or a STM32G0B1RET6
i forgot to take a photo before installing lol defs a G0 .

ive compiled the latest marlin bugfix for the printer and its running that no issues for the time being, just using DWIN_MARLINUI_PORTRAIT tho i forgot to publish my changes i made on my work laptop to my fork lol. tho a quick make of firmware platform io says ive got a good meat of mem left 
[image]

so putting the hash on the proui in the stm32 common fixed it, but does this mean that its not going to load the dependencies for proui when i compile? seems its a dependencies issue with pio from this terminal output  
[image]

if i use the configuration files in the configuration folder for the BTT-SKR_Mini_E3_V3 i get a env_validate error
[image]

[13 more replies]

closed 2023-05-20 by classicrocker883