classicrocker883/MRiscoCProUI #25: Missing MPC menu

kagebarton, 2023-03-02:
I enabled MPC for the hotend and while the hotend PID option was gone, there is no MPC menu. I did verify that the MPC g-codes work through the terminal.

classicrocker883, 2023-03-04:
I believe the mpc menu has to be enabled in the dwin.cpp file which is the main file for how the UI looks. I'll look into this and see what should change.

classicrocker883, 2023-03-04:
in the dwin.cpp file you can add

>#if ENABLED(MPCTEMP)
      MENU_ITEM_F(ICON_MPCNozzle, STR_MPC_AUTOTUNE " Settings", onDrawSubMenu, Draw_HotendMPC_Menu);
    #endif

to `void Draw_Temperature_Menu() ` and change the value 5 to 6 where it says Srt_menu, before preheat_count.

kagebarton, 2023-03-05:
Perfect. Thanks!

I disabled PID when I enabled MPC on the hotend.

closed 2023-05-27 by classicrocker883