classicrocker883/MRiscoCProUI #30: question

Moldytzu, 2023-04-14:
Sorry for posting this in the bug tracker... I have an H32 aquilla and I was wandering if I can compile your firmware for my machine. What could the steps to do so be?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

**Additional context**
Add any other context about the problem here.

classicrocker883, 2023-04-17:
for H32 machines, as of now its possible using VSCode and platformio. I am in the midst of testing everything before I make my own build ready to go.

if youd like to compile your own firmware for H32, its possible by selecting the `hc32_build` branch. if youre familiar with doing this through vscode then it should be easy for u, otherwise there are stepbystep directions and tutorials if you search, like on youtube.

i do have another repository called HC-32 which has the same source code files more or less.

another thing before you start, the HC32 build for platformio currently only supports the color LCD, like on the Aquila X2. if u have a C2 with blue display this wont work just yet, that is an issue being worked on.

so feel free to compile your own for the HC32 board, I have yet to finalize my **configurations** folder, and beaware that usually the printers come with HC32 boards with 256k flash memory. you can confirm this by looking directly at the chip on the mainboard and reading the numbers. 

the thing is HC32 may not fully be able to use ProUI and all its features. many options and such have to be disabled inorder for everything to fit so it can flash correctly.

classicrocker883, 2023-04-22:
did u manage to compile successfully? did u need anything else? and you have an Aquila with a color LCD, like the X2, correct? 

because I am close to having firmware.bin files ready for flashing.

Moldytzu, 2023-04-22:
hi. I didn't try to do anything yet. I will tell you as soon as possible if I have some results.

closed 2023-05-27 by classicrocker883