classicrocker883/MRiscoCProUI #36: Problem with advance

Lord-Memester, 2023-05-20:
### Discussed in https://github.com/classicrocker883/MriscocProUI/discussions/35

<div type='discussions-op-text'>

<sup>Originally posted by **gogsi1** May 20, 2023</sup>
Voxelab Aquila X2. 

At any value of advance k, all motors start to vibrate when printing.</div>

Lord-Memester, 2023-05-20:
@gogsi1 They always vibrate. Do you mean they make an audible vibrating noise?

gogsi1, 2023-05-24:
> 
No, not sound. The movements themselves are twitchy (in waves)

Lord-Memester, 2023-05-26:
> > @gogsi1 They always vibrate. Do you mean they make an audible vibrating noise?
> 
> No, not sound. The movements themselves are twitchy (in waves)

Have you tried enabling or disabling step smoothing? The toggle should be visible at the bottom of either the control menu or the configuration menu (I don't remember which at the moment nor can I guarantee that it will help)

classicrocker883, 2023-05-27:
i believe it is under *Control* --> *Motion*

there has been an update with the code, i will be posting the updated firmware soon. for now if you want I can compile the firmware you use and post it for you to try like an early release see if that fixes anything.

gogsi1, 2023-05-27:
> i believe it is under _Control_ --> _Motion_
> 
> there has been an update with the code, i will be posting the updated firmware soon. for now if you want I can compile the firmware you use and post it for you to try like an early release see if that fixes anything.

Thanks, I'll wait and I'll be able to compile it myself, it's just that I have a head and the Z axis is modified, I need to edit the config.

Lord-Memester, 2023-05-27:
> I have a head
> 

What? I'm confused, do you mean you have a different hotend?

gogsi1, 2023-05-27:
> What? I'm confused, do you mean you have a different hotend?
Yes
My Setup:
voron stealthburner.
Belt Driven Ender 3 - KevinAkaSam's.

classicrocker883, 2023-05-27:
I updated everything most recent under *2023-June* branch, feel free to use it.

gogsi1, 2023-05-28:
> I updated everything most recent under _2023-June_ branch, feel free to use it.

Problem still exists.

classicrocker883, 2023-05-28:
https://github.com/classicrocker883/Marlin/tree/bugfix-2.1.x-N32

try this branch, it is just the latest Marlin, not the ProUI with all the extra things, but you will be able to copy over your config.h files change platform.ini to ` _GD32`

I made it so you'll be able to adjust lin_advance from the menu. let me know how it goes if you still have the issue. because that would mean it is either a greater Marlin issue, or with Mriscoc.

gogsi1, 2023-05-28:
> https://github.com/classicrocker883/Marlin/tree/bugfix-2.1.x-N32
>
> try this branch, it is just the latest Marlin, not the ProUI with all the extra things, but you will be able to copy over your [...]

Failed to compile "STM32F103RC_voxelab_maple_GD32", if you select another one that also fits, the display does not work

`Error: Build environment 'STM32F103RC_voxelab_maple_GD32' is incompatible with BOARD_CREALITY_V4. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple, STM32F103RC_voxelab_maple_N32`

classicrocker883, 2023-05-28:
change to BOARD_VOXEKAB_AQUILA like from the previous repo

gogsi1, 2023-05-29:
> change to BOARD_VOXELAB_AQUILA like from the previous repo

It got even worse, even when parking by a lady, the motors go crazy.

classicrocker883, 2023-05-29:
>parking by a lady, the motors go crazy.

? yea i know mine do lol. are we talking about the same thing?

well send an issue to Marlin Issues and see what they say

gogsi1, 2023-05-29:
> > parking by a lady, the motors go crazy.
> 
> ? yea i know mine do lol. are we talking about the same thing?
> 
> well send an issue to Marlin Issues and see what they say

Thanks for the replies and help. For now, I will use your firmware with turn off advance. P.S. I don't know how to explain in detail what the problem is or describe it in a way that everyone understands. I have a problem with this

classicrocker883, 2023-05-30:
I can post the issue with a reference linked to here. for now do you think this would be beneficial?
```
  //#define LA_DEBUG              // Print debug information to serial during operation. Disable for production use.
  #define ALLOW_LOW_EJERK         // Allow a DEFAULT_EJERK value of <10. Recommended for direct drive hotends.
 ```

you can try to increase the `E Jerk` value to 10 or more, its under the *Control Menu* --> *Motion*
if that doesn't change anything, basically what I can do is enable `LA_DEBUG` in a firmware file you can then flash and you could use that to help find what exactly is happening.

classicrocker883, 2023-06-01:
in Configuration_adv.h, add `#define OLD_ADAPTIVE_MULTISTEPPING 1` to any line really. so that it remains enabled. see if this helps linear advance.