classicrocker883/MRiscoCProUI #38: Prints are cancelled

Arson31, 2023-05-26:
Sliced a model on Cura and sent through Octoprint. The print starts and gets to desired temperature. The print head homes and immediately cancels the print.

classicrocker883, 2023-05-27:
I went ahead and posted an updated version with the most recent updates to this firmware. think of it as a pre-release.[GD32_MM-ProUI-June.bin](https://github.com/classicrocker883/MriscocProUI/releases/download/2.1.3c1/GD32_MM-ProUI-June.bin)

the updates and changes made are said to have fixed this kind of issue. please let me know if you try it out and how it works out for you. thanks

Arson31, 2023-05-28:
@classicrocker883 Does this contain the tramming wizard? I couldn't find it anywhere in GD32-MM-ProUI.bin.

Arson31, 2023-06-05:
@classicrocker883 Nevermind I found it. I did a dry run with the new firmware and the print started like it should. I'll open another ticket if I find anymore problems.

closed 2023-06-05 by Arson31

classicrocker883, 2023-06-07:
ok sounds good. thought I had replied before saying I'll be posting an updated release. but if you're hearing it now, look out for that hopefully this week.