classicrocker883/MRiscoCProUI #41: GD32 BLT-ProUI 5/2023 is not saving mesh

vw72, 2023-06-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Build mesh
2. Save Mesh
3. Power off printer
4. Power on printer
5. Go to Mesh settings and view mesh - Error No Valid Mesh

**Expected behavior**
Expected behavior is that mesh is saved between power cycles

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version 2.1.3c1 and release date May 2023

**Additional context**
Add any other context about the problem here.

classicrocker883, 2023-06-08:
this issue may already be fixed, I uploaded a newer version of an updated pre-release [you can find here](https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3c-test2)

with **BLT**, there should not be the ability to view the mesh because that is more or less reserved for **UBL**,

even though you may not be able to "View Mesh", go to the *Mesh Edit* menu item if you can and it should let you see the value for each point on the grid and edit it.

can you report back with having flashed the newer update?

also, before restarting the printer after a mesh was built, were you able to view the mesh and/or manually edit it?

vw72, 2023-06-09:
This is still not working with GD32_BLT-ProUI.  It correctly builds the
mesh and I can see it with the mesh viewer and I can edit the mesh, but as
soon as i reboot the printer both the mesh viewer and the besh edit say
invalid mesh.

Joe

classicrocker883, 2023-06-09:
uploaded some new files with the fixes let me know how it goes

https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3c-test3

vw72, 2023-06-10:
> did you save the mesh before you restarted?
>
> ill see if I can have the mesh save upon Leveling Done

Yes, I did.  I tried it both with the save button and then also with the
save settings menu choice

vw72, 2023-06-10:
I tried both the GD32 BLT and UBL versions and the mesh is not saved on
either.  I know you say that BLT doesn't have a mesh, but after doing the
automatic mesh it displays a mesh and allows you to edit it.  It simply
does not save it.  Same for the UBL version. Both respond with invalid mesh
after cycling the power.

vw72, 2023-06-10:
Okay, I think I might have found something. I loaded an older version you
made for me to fix the sd card reader called GD32_UBL-ProUI-SDex.bin.  It
is version 2.1.3 with a time stamp of 20230503 - 18:25.

While I thought this one was working correctly, it was not. However, my bed
was really zeroed in so it didn't make a difference.  But more importantly,
here is what I found

1: Build the mesh as normal.
2: View the mesh to make sure it is there
3: restart the printer
4: When viewing the mesh, instead of invalid mesh, the mesh shows all
zeroes, instead of the previous mesh

This is similar to the problem that I reported with the only difference
being #4 on the later version, instead of all zeroes for the mesh, it
simply says invalid mesh.  However, this version has an option to load the
mesh and upon doing so, the originally created mesh now works as expected.
On the current versions you've had me test, that option is not available.
However, it makes me wonder if the problem is not in saving the mesh but
that the system isn't automatically loading loading the mesh properly
(maybe saving to the wrong slot or reading from the wrong slot)?

This may be just a coincidence, too, but I wanted to let you know just in
case it helps.

Joe

classicrocker883, 2023-06-10:
im not finding a way for the BLT - which is the Bilinear Auto Bed Leveling - to save the mesh upon restart. im sure it does save it somewhere in eeprom, but the way the code is either it doesn't simply allow a mesh to be saved like UBL - having to save several slots, or if it does save it somewhere, the code doesn't effectively load it when being turned on.

think of it like volatile memory - RAM. its not meant for long term storage and basically erases when powered down. 

im not actually sure if this is how Bilinear auto leveling has always been, or if its to do with Mriscoc ProUI, or Marlin as a whole. that is if I try JyersUI i wonder if its any different.

as for the Mesh being invalid or all zeroes when first loading, that can be from having it saved in one slot and loaded from another.

I posted a small change with the BLT version, where it should automatically save the mesh after being built. and I believe it also show you the grid, and an option to save (again).

if that doesnt save after restart and load, then its probably the BLT version isn't meant to. or the code to save and load isnt matching. I mean what I can do is arbitrarily have it like UBL, but then whats the point because it becomes literally exactly like UBL, lol..

Mriscoc said hes pushing more for UBL than the bl-touch bilinear version. and i would have to agree, the only downside for UBL is it takes a lot of space. so for anyone who wants some extra features that aren't otherwise enabled with UBL, they have more room to enable everything.

the recent upload is here **https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3c-test4**

classicrocker883, 2023-06-10:
i see that the mesh does save on restart. im not sure what changed that resets it. but i should have that at least back to normal

classicrocker883, 2023-06-10:
update:

I tried out Mriscoc's vanilla and it does the same thing.
however, If I manually Load the mesh, it does infact load correctly.

I recall I seen this before, but I thought maybe I fixed it? that i dont remember. but if you say restarted your printer after saving a mesh, and went to print. it will indeed have no problem, because it loads by the start gcode. so maybe I didn't worry to much about it.

anyway, I could try and get it right, where it loads back upon restarting, but say you save on slot 2, it would load slot 0 every time, or 1 if i made it. like i dont think there is a way to "save" the last selected slot to "load" it.

anyway, its not too much of an issue, because the mesh does save and does load, even if u have to manually do so when viewing.

vw72, 2023-06-10:
The GD32_BLT-ProUI test 4 seems to use the mesh with the M420 S1 in the
start up gcode, but you still can't view the mesh after it is issued, so I
guess it wasn't meant to be (although I thought Alex's firmware did, but I
could be wrong or he was actually using UBL).

I also tried the GD32_UBL_ProUI test 4 and could not build a mesh at all.
It locks up the printer as soon as you hit the confirm button when trying
to build the mesh. Do you want me to open a separate ticket on that?

I agree with you about the UBL method being preferable to the BLT. I was
previously using it until one of the updates caused it not to probe the
right most row of points.  I think you said that has been fixed, but I
couldn't verify because I couldn't create the mesh.

Also, I believe the bed screw inset defaults to 35.0.  I was under the
impression that the Aquila uses 32.5 and I found a post on reddit from you
saying it should be 25.5.  Do you know which is the correct one to use
(I've been using 32.5)?

Joe

vw72, 2023-06-10:
Bad news. I still don't think the BLT version is loading the mesh when it
goes to print.  If I start a print job and have the M420 S1 command in the
gcode, the skirt and first layer is not consistent and the z axis displayed
on the screen sits on .20 for that layer.  However, if I remove the M420 S1
and replace it with G29, then it builds the mesh and then when printing,
the skirt and first layer are very consistent and the z axis display
fluctuates slightly depending on where it is on the bed.  I've also had the
printer reboot itself a view times unexpectedly.

It might be best to back out the test 4 changes in case that is causing the
reboots and just tell people that if they use the blt version it works best
by issuing the G29 instead of the M420 S1 or use the UBL version.

Joe

classicrocker883, 2023-06-10:
>it works best
by issuing the G29 instead of the M420 S1 or use the UBL version.

I agree, I think that the BLT version doesn't save the mesh in a specific location. or whatever address it is in, M420 S1 doesn't enable it because it's not there. I wonder if G29A will load the mesh and bypass the mesh building. it says 'A' will abort leveling. so maybe this will basically do the same as M420 S1? I'm not sure.

>could not build a mesh at all.
It locks up the printer as soon as you hit the confirm button when trying
to build the mesh. 

yes I was just aware of this trying to test it out myself, so no need for a ticket.

> one of the updates caused it not to probe the
right most row of points.

yeah I made a change so the **Mesh Inset** will save upon starting up. before the Mesh X Maximum would equal the max bed size - probe.z offset. so technically, with a smaller mesh Inset, it will be able to probe all the points because it's staying within the bounds of the probe.

the thing is, it basically makes that portion of the bed not usable regarding bed leveling. so what was changed was having the mesh Inset be normal, and if the probe cannot reach those points, with `G29 P3` it automatically calculates accurately what it should be.

so yea u might not probe all the points, but say with 7x7, you get 43/49 and more bed to print with. otherwise yeah you can probe all the possible points, but you lose part of the bed it can build a mesh on.

I'm trying to fix that auto build mesh freeze issue, and also can u test the tramming wizard? for me I'm having an issue with no values showing.

vw72, 2023-06-11:
*I agree, I think that the BLT version doesn't save the mesh in a specific
location. or whatever address it is in, M420 S1 doesn't enable it because
it's not there. I wonder if G29A will load the mesh and bypass the mesh
building. it says 'A' will abort leveling. so maybe this will basically do
the same as M420 S1? I'm not sure.*
I haven't tried that but I will and get back to you. I was thinking G29A
only worked with UBL, though, so I'm not sure.

*yeah I made a change so the Mesh Inset will save upon starting up. before
the Mesh X Maximum would equal the max bed size - probe.z offset. so
technically, with a smaller mesh Inset, it will be able to probe all the
points because it's staying within the bounds of the probe.*

I got around this by going into the physical size settings and setting the
x and y to 230.  It seemed to be just enough so that probing could
complete.  The bed is actually 235x235, so you could probably increase it
to that, but I use a magnetic pei sheet and if somebody was using a bed
with clips, that could be problematic.

*the thing is, it basically makes that portion of the bed not usable
regarding bed leveling. so what was changed was having the mesh Inset be
normal, and if the probe cannot reach those points, with G29 P3 it
automatically calculates accurately what it should be.*

That's probably so, but when I was having the problem, it always showed the
last column mesh values to equal the last column actually probed.

I*'m trying to fix that auto build mesh freeze issue, and also can u test
the tramming wizard? for me I'm having an issue with no values showing.*

I just tried the tramming wizard and it seems to work fine with the BLT
version. I find it quicker, however, to go into the Bed Tramming menu and
instead of running the wizard just go to each of the corners and adjust the
z value that is displayed to match each other (or be close).  But the
wizard does work for me. Are you having problems with the BLT or UBL
version? And if the UBL, I wonder if it is related to the mesh building
freezing?

Anyway, it's really shaping up. Let me know if there is anything else you
want tested.

Joe

DownTownJulioBrown, 2023-06-21:
I hope this helps on
2.1.3d 3fc2d94 GD32_MM-ProUI-MPC.bin
2.1.3d 3fc2d94 GD32_MM-ProUI.bin
and also MM from aca685cb9dc58d6da40e71af87efdb4e9d78ac31

I figured this out on a whim on the current builds
all save mesh but do not load at startup.

While using octprint, issuing the M501 after a reboot loads mesh data.
```
After reboot

Send: M420 S1
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Failed to enable Bed Leveling
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 10.00
Recv: ok

After M501

Send: M420 S1
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 10.00
Recv: ok
```

In the printer I could also

1. Build mesh 
2. Save Mesh 
3. Power off printer 
4. Power on printer 
5.  Go to Control/Advanced Settings/Load Settings
6. Then go to Mesh settings and view mesh 

I auto load the mesh from EEPROM at startup with an M501 command in an auto0.g file on the root of an sdcard. ( https://marlinfw.org/docs/features/autostart.html )

The auto0.g file might help you while the mesh issue exists.

classicrocker883, 2023-06-21:
this might be helpful. that was an issue being able to save or load non-UBL versions. M501 is the restore all settings command, so this is a clue as to why it would not simply load the mesh upon starting. I may have the settings automatically load like this, but without the need for a file on the sd card. thanks for the input.

closed 2023-10-03 by github-actions[bot]