classicrocker883/MRiscoCProUI #42: Cannot change micro-sd card when using an SD card adapter without rebooting the printer

vw72, 2023-06-04:
**Describe the bug**
If one is using a micro-sd to standard sd card adapter the printer doesn't recognize that the card has been changed (or if no card is in the printer when powered on, it doesn't recognize that a card has been added).  One must reboot the printer to see the new card contents.

**To Reproduce**
Steps to reproduce the behavior:
1. Put SD card containing the micro-sd card in the external SD card reader
2. Power on the printer
3. Eject the SD card and replace with another
4. Printer doesn't recognize the change and still displays the old directory

**Expected behavior**
When replacing the SD card, it should update the directory listing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
May release

**Additional context**
You fixed this for me with a separate build of a beta release (on reddit), but the latest release does not have the fix.

classicrocker883, 2023-06-08:
I uploaded [a new pre-release here](https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3c-test2)

if you'd like to try that out since the new Mriscoc update it has addressed this issue.
Also, the **SD card extender** can be enabled in *Advanced* settings under the *Control* Menu. before it used to be only enabled when compiling firmware, now it can be turned on or off anytime.

reply back if you have success or need further assitance.

vw72, 2023-06-09:
On both GD32_BLT-ProUI and GD32_UBL-ProUI, This is working now and I have
the menu option you mentioned.

closed 2023-06-16 by classicrocker883