classicrocker883/MRiscoCProUI #44: H32 builds (previous releases)

whasupjohn, 2023-06-05:
Hello, I am attempting to run any version of the software from the H32 and the only file that loads properly is HC32_Default-NP-ProUI.bin -- all the others seem to load and then the loading bar repeats, over and over.

I know you said to stay tuned for updates -- what I'm looking for is to upgrade my Aquila (armed with a 32-bit HC32 board) and enable the CRtouch I have just installed.  I've read your notes that the *test* versions are to be avoided.  I'm just trying to get this working and was wondering if I could help test ... try different things, any insight, etc.

classicrocker883, 2023-06-08:
apparently there was an issue with BL-touch working with Alex's build. in the other test files there was some up and down success and failure. like the UI will load, but then crash, or the endstops would not register, as in the Z probe would keep going into the bed.

I uploaded some very new versions here, you can try the files ending in **-test** let me know how it goes. there has been much work in the code and everything should be working as it should. otherwise I will keep working at it. im interested as to know what works and what doesnt.

https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3c-hc

whasupjohn, 2023-06-08:
I very much appreciate the effort and will test this weekend.  I'll make sure to comment once I have results.

tlewis99, 2023-06-10:
The code did load this time.  Once I was in I was very impressed with all the features.  I went and did a Mesh and both versions I tested (the 5x5 and the 7x7) both were able to probe all the locations and build the mesh.  The problem occurred on saving the mesh and both builds rebooted the printer, nothing saved.  Not sure what other data points I can give you.  I'm more than interested in the firmware and will help you with whatever feedback you need.  Perhaps this is a common issue with the other mesh saving problems others have reported.

Thanks.
John

classicrocker883, 2023-06-10:
I'm pleased to hear it seems to work like normal. I'm working on the issue with saving, it's supposed to load the mesh on starting up. how about try this: after restart, go to mesh level settings - if you view the mesh and it is all zeroes, go to load mesh. it should load the previous values and all should be there when you view it again.

if so, it does save, it just doesn't load upon starting up.

I think i found a fix for it.

as for another issue that you can look out for is the Tramming Wizard. are you able to use it normally? I went to test it and for me it only loads Zeroes for each of the 4 corners.

also when doing the autolevel - to auto build mesh - there's a popup asking if you want to continue, right? if you press continue does it freeze? or does it continue to probe, and does it show each point show up on the grid?

whasupjohn, 2023-06-12:
Sorry I was out of town the last couple of days, I saw new code set out, did you have loadable firmware to test, if so I can get right on it!  Thanks again.

John

whasupjohn, 2023-06-14:
OK, so after doing the mesh I ask it to save and it reboots.   It does not save and when it reload it says there is no valid mesh.  I am using the HC32_BLT-5x5-ProUI.bin version.  I'll test the 7x7 next.  So, for clairty it doesn't seem like it saves therefore it can't load.

John

whasupjohn, 2023-06-14:
Same behavior with the 7x7 mesh.  Initially I get an EEPROM error (before I load or level).  It asks if I want to load slot zero, I say yes.  Nothing there, so I zero it out, then run the level.  After level is run, prompts to save to EEPROM, I say yes and it reloads.  I check, nothing saved in slot zero.  Not sure if you have other builds for me to try, but I'm more than willing.  This was using the HC32_UBL-7x7-JyersUI.bin file.

John

whasupjohn, 2023-06-14:
Further notes.  Looks like on any of the builds, I can't even save a simple thing like the z offset without a reload.  So, this may just be a generic save issue -shrug-.  Let me know if you have more for me to test and any specific things you need me to test.

whasupjohn, 2023-06-14:
So, this last build you just dropped is still exibiting the oddness that it does the mesh properly, then I go to save it, and the software reloads.  I check to see if there is any valid data and it says no.  I'm running a second levelling now.  I did check to make sure that the mesh was zeroed out and the readings are being taken, it looks fantastic right up until I say save.  Then I get the greet screen and the progress bar.  I then get the welcome box with the revision info.  I wish I had better news =(

whasupjohn, 2023-06-14:
Same results.  I took a video of the last part if you'd like to see it.  It definitely seems like its erroring out on writing to the EEPROM.  Odd, since I've seen and had mesh working (stored and auto loaded on reboot) on this machine, so I know its capable.

classicrocker883, 2023-06-14:
https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-hc

here is a new update for this, let me know how it goes, thank you!

classicrocker883, 2023-06-14:
as for the mesh saving and loading with the BLT version, bl touch, I don't think that it is actually possible because there the code for that is under UBL bed leveling. Bilinear auto bed leveling (which is the BLT version) doesnt seem to link to multiple slots.

I may have explained this before somewhere, but if you're looking for multiple meshes then UBL is the way to go. I'm unsure if BLT or BL touch version is able to save a mesh after restart. I'll have to see if it is for Mriscoc's vanilla repo, or JyersUI is the same. 

I may be able to get UBL and the ProUI to work if it is able to flash onto the Hc32 board. because apparently the limit was 208k, but I am unsure if that still needs to be or if it it's 256 like the Aquila.

Ryloth, 2023-06-15:
> https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-hc
> 
> here is a new update for this, let me know how it goes, thank you!

Regarding those more basic firmwares with some simple observations;
* HC32_UBL-ProUI, stored settings like z-offset even after resetting
* HC32_UBL-ProUI, resets right after calculating the mesh
* HC32_BLT-ProUI, store settings causes reset and does not save settings like z-offset
* HC32_BLT-ProUI, resets the printer after trying to save the mesh but calculates just fine,
OR when starting to build mesh or selecting Continue after mesh is calculated the UI control knob becomes unavailable but UI elements continue to update like current temperature

whasupjohn, 2023-06-15:
I have also loaded HCL32_BLT-ProUI.bin and can confirm that although it did go through the motions on my machine to find all the mesh points, it reloaded at time of save.  I also tried HC32_BLT-ProUI.bin and it would reload after homing and going to grab the first data point.  I wish I had better results =(

whasupjohn, 2023-06-15:
Also of note, when I use the Z offset wizard, it auto homes properly then I use the center nozzle command sometimes it completes, sometime it hangs.  If it does complete and I select Z-offset I can dial it to where I need and press to complete the action and the system reloads.  It does keep the value tho....

classicrocker883, 2023-06-17:
hey guys wanted to update on the progress. I'm working on tackling these issues, a couple are similar to the G32 and N32 builds, like the mesh save and reload or freeze.

I may need some help with the Z offset issue mentioned. so I'll try to recreate any problems on my end, however I don't have the H32 board to test with. but the issues are similar to the other chips.

I'd just like some more details - like with the Z offset - step by step what to do and what happens and what is *supposed* to happen. videos or pictures are welcome.

anything I'm able to find out I'll let you know, anything I'm unable to get around I'll relay to Mr Shadow578 who is working on the HC32 framework who made it possible to compile HC32 boards with VS Code and PlatformIO.

whasupjohn, 2023-06-17:
I appreciate all the work you are doing.  I'm in the middle of considering upgrades and I am probably going to swap my board out.  Do you have a way to use/test and H32 board if I send it to you?

classicrocker883, 2023-06-18:
> I appreciate all the work you are doing. I'm in the middle of considering upgrades and I am probably going to swap my board out. Do you have a way to use/test and H32 board if I send it to you?

@whasupjohn yes I could use the H32 for testing, that would be wonderful!

I'll be happy to receive it whenever you're ready. [if you search here](https://api.github.com/users/classicrocker883/events/public) *find "email" you'll eventually see my user name and @yahoo send me an email and I'll give you me address

Ryloth, 2023-07-01:
> I'd just like some more details - like with the Z offset - step by step what to do and what happens and what is _supposed_ to happen. videos or pictures are welcome.

[Here](https://drive.google.com/file/d/1X1NILEq5LBJjQyBpOxQuNtrYQwJl2cmP/view?usp=sharing) is a video showing the issue with storing the z-offset in more detail, in particular, with the HC32_UBL-ProUI.bin (892130d) where this setting survives an unexpected reset after storing the setting whereas with HC32_BLT-ProUI.bin (892130d) it operates similarly but would always reset to '0.00' after the unexpected reset.

In normal operation, the printer software should not reset after storing this setting and should stay in the menu where you can continue making changes, and, obviously, the setting would persist.

> for the encoder knob, is that an issue at all? someone has mentioned it can get a bit janky or like stutter, and that may be due to Mriscocs recent May update. I had to change the values to something different - just need a second opinion. so the values for encoder speed rates could be changed if need to. I'm able to have the encoder rates change on the fly through the lcd screen for testing if they don't seem right.

This behaviour doesn't seem to be an issue for me or at least I don't notice any unexpected behaviour when using the knob compared to stock Aquila X2 firmware.

classicrocker883, 2023-07-02:
thank you @Ryloth for this, the *_BLT-* version with it not saving after restart seems to be an issue across all other chips, its not particular to this Repo, it seems to stem from the upsteam by Mriscoc or MarlinFirmware. that is being looked into, hopefully will be fixed in coming updates.

as for it printer rebooting unexpectedly after issuing a save, this appears to only be with the HC32 chips. I'll be doing some testing as I look further into the cause. As this may have to do with the framework/toolchain for the *HC32F46x* by Shadow578 - who made all this possible, being able to compile using VS Code and platformio.

ive tried getting in touch with him about it, because he may be able to figure that one out. otherwise i'm not sure where to begin. but id like to just say if you or anyone knows programming, C/C++, MCU, Arduino, or whatever related, to take a look into the code and find a solution.

Ryloth, 2023-07-02:
The good news through all of this is, even without bed-leveling properly working or not being able to save certain settings, both of the firmware I mentioned let me print just fine from start to finish and I've done that a few times now within the past few weeks.

As for coding, I can script a bit but I don't believe I'm much use past that. Hopefully, you find yourself a willing participant in that regard.

classicrocker883, 2023-07-03:
im glad to hear that it works for you. thank you anyway. so I just heard of this site called [app.BountySource.com](app.BountySource.com) which is like a gofundme for coding. not sure exactly how it all works, but hopefully someone can see the issue and solve it.

check back for any updates. I will post any progress regarding this.

closed 2023-10-03 by github-actions[bot]