classicrocker883/MRiscoCProUI #49: HC32 HC32_BLT-ProUI crashing

Ace00018, 2023-06-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'leveling'
2. Click on 'Store Settings'
4. Crash and subsequent reload to main screen

**Expected behavior**
a beep and the settings to save

**Version (please complete the following information):**
HC32_BLT-ProUI, 6/14/2023 (in my local time zone)

**Additional context**
this also occurs when trying to save any other settings or a mesh and results on a loss of the mesh (because it can not be saved)

Ace00018, 2023-06-16:
forgot to add but i think its implied that the bug is HC32 HC32_BLT-ProUI crashing when any stats mentioned occur

classicrocker883, 2023-06-18:
yes this issue was brought up. I think it is because in the BLT version of a mesh cannot be saved like UBL, and in the code it appears to have some overlap which may be causing the error. I'll be working on a fix so hang tight.

Ace00018, 2023-06-19:
great that's good to hear as soon as that issue is solved this firmware will be going on all my printers.

also will you be making a fix for this crash occurring when saving things like e steps and z offset?

classicrocker883, 2023-06-20:
ill be able to test with the GD32 variant, I haven't got an H board to test with but someone said they will kindly donate one. as long as that is all good it should work with the other. sometime this week hopefully the next couple days i should have something ready to go.

classicrocker883, 2023-06-26:
there is an updated release. it should be working fine, however I have reports that the save feature reboots the screen. this can be something with the framework, or some other issue because it's fine for G32 N32.

Ace00018, 2023-08-28:
hey coming back to this issue after a while and i tried the latest release you posted as "working HC32" release has the exact same issue on my printer

Ace00018, 2023-08-28:
> there is an updated release. it should be working fine, however I have reports that the save feature reboots the screen. this can be something with the framework, or some other issue because it's fine for G32 N32.

hey coming back to this issue after a while and i tried the latest release you posted as "working HC32" release has the exact same issue on my printer

Ace00018, 2023-08-28:
hey so sorry i spoke too soon it saves settings the restart was just very jarring to see after saving my settings good work my friend

classicrocker883, 2023-08-28:
> hey so sorry i spoke too soon it saves settings the restart was just very jarring to see after saving my settings good work my friend

will be posting on new version where issues are gone, however it won't be the ProUI with all extra features, it will just be like Alex's except updated of course. this is just until the ProUI save issue is worked out.

Ace00018, 2023-08-28:
Great I guess I never thought about if pro ui features whould be causing
issues

Ace00018, 2023-08-28:
hoping to see a BLT/UBL-NoPro build soon thanks for your help :)