classicrocker883/MRiscoCProUI #51: H32-MM-ProUI 892130d Crashing

whiney1, 2023-06-27:
**Describe the bug**
Updated my Voxelab Aquila X2 H32 with the new firmware, coming from Alex's firmware. Also just updated the screen assets using the links in the install file.
Everything seems happy post-update but there are very consistent crashes, almost always when using the screen control knob. Sometimes during settings menu, sometimes during adjusting mid-print, sometimes clicking into SD card, etc. 
Doesn't always happen - for example I was able to fully probe a 4x4 manual mesh and save it without issues.

Printer freezes, steppers unlock, eventually printer will self reset after a short while and go back to the home screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest firmware on Aquila X2 H32.
2. Conduct regular use.

**Expected behavior**
No crashes in normal operations.

**Version (please complete the following information):**
H32-MM-ProUI 892130d 27 Jun 2023

**Additional context**
Printer has not shown this behaviour on other firmwares.
Screen assets updates as mentioned, no effect on crashes.

Originally it seemed to be occuring from pressing into 'Print' on the home screen so I thought it was the SD card, so I then tried re-formatting SD card, and also trying a new SD card. But no help, and issues occur during prints etc too.

classicrocker883, 2023-07-03:
this is what I suggest for now. as I noticed this type of issue happens on other board/chip variants. so its not just to the  **HC32**. it can have something to do with the recent update by Mriscoc. further investigation and testing is needed, since I did notice some changes to the encoder knob code - particularly to the **`delay()`** function.

clicking the encoder Knob in the SD print menu will cause this restart problem to happen.

so for now I recommend using the knob at a *slower* pace. I did post a newer [update here](https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-hc1) if you haven't use that version already.  
someone has graciously offered to donate a H32 Board for me to do testing on, and I will be happy to receive it to help fix said issues.

if you can let me know if youre using the newest release from the link, and if slowing down your knob usage in the *Print* from SD menu helps.

whiney1, 2023-07-07:
Hi classicrocker

thanks for your work on all this. I was on the new update, and the slow knob miiiight have helped - hard to say as it still crashed at times. It also definitely crashed out side of the "SD Print' menu too.

I've used this as an excuse to getting around to trying Klipper so I'll close the issue here. Cheers

closed 2023-07-07 by whiney1

classicrocker883, 2023-07-07:
just received a board with an H chip. I'll be doing my testing and updates to the code. Hopefully everything works out.