classicrocker883/MRiscoCProUI #55: Using Pro - UBL files and Auto Build. No other customization.

LostFool, 2023-07-04:
platformio run --silent -e GD32F103RC_voxelab_maple
In file included from Marlin\src\MarlinCore.cpp:64:0:
Marlin\src\lcd/marlinui.h: In static member function 'static void MarlinUI::set_status_no_expire(FSTR_P)':
Marlin\src\lcd/marlinui.h:505:84: error: call of overloaded 'set_status(const __FlashStringHelper* const&, int)' is ambiguous
   static void set_status_no_expire(FSTR_P const fstr)       { set_status(fstr, true); }
                                                                                    ^
Marlin\src\lcd/marlinui.h:461:15: note: candidate: static void MarlinUI::set_status(FSTR_P, int8_t)
   static void set_status(FSTR_P const, int8_t); // set_status undefined reference libproui.a PROUI_EX workaround
               ^~~~~~~~~~
Marlin\src\lcd/marlinui.h:465:15: note: candidate: static void MarlinUI::set_status(FSTR_P, bool)
   static void set_status(FSTR_P const fstr, const bool persist=false)       { set_status_P(FTOP(fstr), persist); }
               ^~~~~~~~~~
*** [.pio\build\GD32F103RC_voxelab_maple\src\src\MarlinCore.cpp.o] Error 1
In file included from Marlin\src\feature\bedlevel\ubl\ubl_G29.cpp:33:0:
Marlin\src\feature\bedlevel\ubl\../../../lcd/marlinui.h: In static member function 'static void MarlinUI::set_status_no_expire(FSTR_P)':
Marlin\src\feature\bedlevel\ubl\../../../lcd/marlinui.h:505:84: error: call of overloaded 'set_status(const __FlashStringHelper* const&, int)' is ambiguous
   static void set_status_no_expire(FSTR_P const fstr)       { set_status(fstr, true); }
                                                                                    ^
Marlin\src\feature\bedlevel\ubl\../../../lcd/marlinui.h:461:15: note: candidate: static void MarlinUI::set_status(FSTR_P, int8_t)
   static void set_status(FSTR_P const, int8_t); // set_status undefined reference libproui.a PROUI_EX workaround
               ^~~~~~~~~~
Marlin\src\feature\bedlevel\ubl\../../../lcd/marlinui.h:465:15: note: candidate: static void MarlinUI::set_status(FSTR_P, bool)
   static void set_status(FSTR_P const fstr, const bool persist=false)       { set_status_P(FTOP(fstr), persist); }
               ^~~~~~~~~~
*** [.pio\build\GD32F103RC_voxelab_maple\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o] Error 1

LostFool, 2023-07-05:
To work around it I disabled the line in question. Have not flashed the firmware yet but I was able to compile via platformio. I am not certain if something else is out of wack as this is a fresh install of visual studio code because I recently upgraded to windows 11.

This is not my first fimware flash, nor of MriscocProUI. But not to say I'm an expert. Just not a complete noob.
Will try to report back with status and any issues that arise.

classicrocker883, 2023-07-05:
try not using Auto build. or use the 2023-June branch.

in order for the firmware to work with the more recent Marlin, I had to make a double instance for `set_status` 
as you can see from the comment 
>// set_status undefined reference libproui.a PROUI_EX workaround

until Mriscoc comes out with a new update of his repo, I'm not able to change the libproui.a library inorder to accommodate a real fix. so for now the workaround works normally.

closed 2023-07-05 by classicrocker883

LostFool, 2023-07-06:
I successfully completed one print with no issues. I didn't try every menu item or every function. So a consideration for others is to comment out line 505:84 in Marlin\src\lcd/marlinui.h
error : Marlin\src\lcd/marlinui.h:505:84: error: call of overloaded 'set_status(const __FlashStringHelper* const&, int)'

classicrocker883, 2023-07-07:
> I successfully completed one print with no issues. I didn't try every menu item or every function. So a consideration for others is to comment out line 505:84 in Marlin\src\lcd/marlinui.h error : [...]

yes, I may have made a mistake, in my testing I would get a warning doing *UBL* and I did the above "workaround" so it would compile.

```
Marlin/lib/proui\libproui.a(proui.cpp.o): In function `ProUIClass::LevelingDone()':
proui.cpp:(.text._ZN10ProUIClass12LevelingDoneEv+0x26): warning: undefined reference to `MarlinUI::set_status(__FlashStringHelper const*, signed char)'
```

it is possible to have a successful compile. but for other builds, yes commenting out `void set_status` seems to work when you compile for Manual Mesh.

ill try to get a better fix for now.

classicrocker883, 2023-07-07:
so you can have this commented out but for UBL builds it should be enabled.

all you have to do is change this line in **marlinui.h**

`  static void set_status_no_expire(FSTR_P const fstr)       { set_status(fstr, (bool)true); }`  
at the end, just add _**(bool)**_ before "true"

and everything should be working good.
i updated the repo with this fix.