classicrocker883/MRiscoCProUI #56: Aquila GD32 Bed temp too cold OR Thermal Runaway . shutdown

npaisnel, 2023-07-05:
**Describe the bug**
SOMETIMES during a Bed Tramming Wizard. Homing or Auto bed mesh build, or Homing at start of print, get either Bed Temp Too low or a Thermal runaway bleeper.
Only answer is to power off and restart

This has not happened with the OEM Aquila 1.2.4 no BLT or BLT 3.1.2 GD32 firmware

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Control | Temperature | Bed Temp
2. Set bed temp to 70 ..or other
3. Go to   Bed Levelling and start
4. Or go to Levelling/ Auto Mesh Wizard, I have 9x9 set..but happens on others too

**Version (please complete the following information):**
GD32 UBL-ProUI 
Ver 2,1,3 MRiscoC 20230624 - 10:08

**Additional context**
It never did it with the standard firmware.  and it is not 100 % reproproducable.  After three attempts jsut now I completed a Auto test Creation

npaisnel, 2023-07-05:
Trying to get a good way to reproduce. or fix the Low MinTemp error

A fresh Print seems to start at Hot end 160 and bed of 50.

I can change that in Control Temperature and SAVE..but it does not save across reboots.

If I set the Bed temp and Hot end at something like 220 and 70, then during the Tilt 4 pin measuring phase, it can complete ..If I preheat first as the 4 pin title runs, the bed starts Cooling..but as it was already hot..it does not cool enough before the Print heating starts.

I have attached the G-code
[CE3_Imp 4mm.gcode.zip](https://github.com/classicrocker883/MriscocProUI/files/11955713/CE3_Imp.4mm.gcode.zip)

Still working on the Thermal runaway cause...but mention it in case it is related.

As I say  with OEM Voxelab firmware this never occurs

classicrocker883, 2023-07-07:
im trying to understand recreating the issue. so there is a minimum temperature to print. its called "Extruder Min Temp." in Filament Settings menu in the Prepare menu.

you seen to want to print at a lower temp than it is allowing. try changing that

classicrocker883, 2023-07-07:
OK thanks for finding out.

closed 2023-07-07 by classicrocker883

npaisnel, 2023-07-10:
I am now not sure if this is firmware or hardware.

I can run a complete bed levelling with OEM software AND I can run a complete Auto Bed levelling mesh creation with your firmware..but only with Both bed and nozzle temps un set..so cold.

But look at the reported temperatures in the video.
Every time the head lifts a little  both reported temps drop ...
I have moved and checked thermistor wires constantly ..wiggled them, left them loose, re- fixed them with cable ties.  Result is the same.  Always as the head lifts a little..both temperatures show an indicated low figure.  Yes could still be hardware..but something else...PSU maybe ..or is it firmware... ?

Thoughts appreciated.

https://github.com/classicrocker883/MriscocProUI/assets/41953602/8f91a0bb-ac53-4117-b23c-61ddbf56d08e

classicrocker883, 2023-07-13:
@npaisnel did you say you had this fixed? 

>Both bed and nozzle temps un set..so cold.
> But look at the reported temperatures in the video. Every time the head lifts a little both reported temps drop ... I have moved and checked thermistor wires constantly

this looks like something with the wires. be careful checking, you don't want it on and plugged in if there is a possibility of a short.

classicrocker883, 2023-07-13:
what I think is happening is the thermistor (or another wire probably going to hotend) is rubbing or grounding on something. that's why both temp readings change. if you have a multimeter check for continuity to ground/frame of printer and with each wire. again with everything off and unplugged, wiggle or move the wires until you hear that beep. you might be able to see it touching on the ends, or could be a break in the wire, rubbing on the frame.

npaisnel, 2023-07-14:
I thought it was fixed too.
And yes it does totally point to a wiring issue //but it only seems to appear during th end auto level sequence.  I did three successful 2 hour pints he other night, no issues as long as I do not have a ABL sequence in the start G-code.

Whilst looking at the reported temperature, I can manually move jiggle the wires going to the hot end, the extruder feed stepper, the bed and noting ..the temperature stays rock solid

I can move the XYZ axis either manually with steppers disabled, or I can move them using the Move menu item, the displayed temperature stays rock solid. temp stays solid

So yes although it seems very like a wiring issue i just can't trace it..

As a test I went back again to the OEM Firmware and ran the bed levelling sequence.. and that too fails int eh same way..so yep we can rule out Firmware...

Chucking out a random thought...I am jsut wondering if it is deeper than wiring ..but more a board / PSU problem... I am thinking wild idea now as all the obvious wiring moving and checking has been done.  My thought is that since the processor it working at bit harder and the heated elements are also drawing power, that extra little step of power going to the Z - stepper causes the reporting of the temp to drop out..but it only happens with Z- axis movements not Y or Z...hmm

Well i amp going to park this now and just not do any more Auto bed level Mesh creation with heat on ...

I have  Geeetech I3C to B conversion and leadsecrew swap going on, on the other table ,so need to leave one machine  working, so won't strip this down to check wires again yet...But it is a 'new ' machine .  It was a gift from some one that bought it new, could not get on with it and gave it to me.. so it has nto had so much work that wires would have chaffed away ...and there are no trapped wires behind panels etc either .

One for another day ....

npaisnel, 2023-07-14:
See the photo, an 8 hour print of a new Satsana extruder/fan shroud, Worked perfectly, 
So, this printer CAN print / work perfectly.
Yet the alarm/ MINTEMP error always occurrs when bed is hot and a Mesh creation routine is in progress, and the reported temp of -14 occurs 

So it works for normal print, yet fails under specific circumstances… so you can see why I thought ‘firmware’ 

[image]