classicrocker883/MRiscoCProUI #57: Bed Tramming Wizard

npaisnel, 2023-07-08:
I am hoping this is not another hardware / false alarm, but I shall mention it anyway.
 
**Describe the bug**
Bed Tramming Wizard, with ‘Calculate Averages’ OFF.   
Each time the Wizard is run, WITH NO BED SCREW ADJUSTMENTS being made, each corner reports and incrementally bigger measured reading  on each ‘round’ of the bed.

 
**To Reproduce**
Run Bed Tramming Wizard multiple times with Calculate Averages turned off.
Just keep running it each time it ends. 

**Expected behaviour**
Would expect same measurement on each corner at each run.

**Screenshots**

[image]
[image]
[image]
[image]

**Version (please complete the following information):**

GD32 UBL-ProUI
Ver 2,1,3 MRiscoC 20230624 - 10:08

**Additional context**
The number of clicks/ rotations of the encoder wheel to run/ run again is frustrating.  Could it stay ‘within’ the Wizard loop, and merely have a  ‘Run Again ‘  option once it has completed and told you 
‘Raise/lower XYZ corner’ 
Even remove the ‘Confirm/cancel’ step.. pretty pointless really.

npaisnel, 2023-07-08:
I have re flashed the firmware and run again.
Yes confirmed, it increases on each pass..
An Auto Home sorts it / resets it to the same figures per the start image above

classicrocker883, 2023-07-09:
yes thanks for confirming. I have noticed this and wondered why it was like this.

this only happens with calculate average off - finding true values?

npaisnel, 2023-07-09:
So not just me or my dodgy machine !

npaisnel, 2023-07-09:
Your username …. 883    do you happen to ride a Harley ?

classicrocker883, 2023-07-11:
haha you guessed correct! yes I used to have a Sportster 883, which I traded for a chevy truck. but now I ride a Buell Blast, which is almost like a Sportster. you ride?

npaisnel, 2023-07-11:
3 of the buggers now.., 
92, with 3750 miles… Barb find
2004 with bigger tank for Europe touring
2007, live out door everyday Bike

did have a Heritsge too.. but crashed that in Czech Republic a few years back 

no decent truck any more , did have ‘82 Toyota HJ60 for about 25 years.. little Suxuki Vitara and a small Kia Picanto town car now

classicrocker883, 2023-07-13:
I never did think a Harley would be something anyone rode in Europe because they're loud and "American Muscle" like to show off, I always think of Europeans living small and quiet - "to themselves". so that's very cool to hear another Harley Davidson rider.

My father actually just got himself a 2007 or 2008 Sportster 1200. Earlier, he sold his I think it was a Street Glide or Softtail - which he got from insurance money after crashing in his Night Train - which he won before at a raffle.

never been in an accident myself but If I had been with him that day I probably would have as well since it was another drivers fault.

classicrocker883, 2023-07-13:
as for the Trammingwizard - I updated the code a bit, I'm not able to get the error issue to come up. nothing in the code seems to jump out and say what the issue was. Everything works fine for me I can upload a version for you to try.

I'm wondering about your Z offset. it is very High. usually it should be around 0.00, like +/-1.50. also, I notice the leveling is Active, correct? the icon for Z offset 17.85, bottom right looks like the one for when leveling is active, the lines are red and should be white when it's disabled.

try to disable active leveling which I think is under Mesh settings. in the update code I Disable leveling and I think that fixed it.

npaisnel, 2023-07-14:
I actually live on a small island, Jersey, just off the coast of France

https://maps.app.goo.gl/AZkWbJgi7jGBD16r9?g_st=ic

This was my crash, on way to Harley Anniversary Rally.  Harley’s very popular in UK/ Europe

https://www.krimi-plzen.cz/a/vazne-zraneny-motorkar-u-horsovskeho-tyna/

npaisnel, 2023-07-14:
> as for the Trammingwizard - I updated the code a bit, I'm not able to get the error issue to come up. nothing in the code seems to jump out and say what the issue was. Everything works fine for me I [...]

Run with Active Levelling disabled and Calculate Averages off
Wizard reports level all corners +/- 0.01 of the figures below..but multiple runs do still increase.
-0.02
0.02
0.04
0.07
0.13
0.14
0.18
0.20
0.23
0.26

Power off and back on again and figures return to the initial -0.02 figure.
So it does still increase, so if you did have a version you could upload or e-mail to me neilpaisnel@gmail.com

Z-Offset. yes it is around 3mm but is that jsut a random figure, a result of the fitting of the BL-Touch sensor and its operation.
I have just move the head to the bed and the tip of the probe whilst retracted is fractionally less than the thickness of a piece of PLA..so 1.6 mm or so 

With the probe deployed, prob just touches the bed with nozzle 4.6mm above the bed.

I see the spec for the Geeetech 3-D touch V3.2 sensor is 2.3-4.3 mm Tip to sensor tip, so yes a little closer than ideal..but not by 3mm out .  I'd need to print a new fan / Extruder / Sensor holder to adjust that, as this is the setting as it sits 
The shroud/ sensor mount I used is a Thingyverse one  SATSANA with a fixed mount.

So not able to change the height of the tip clearance..so maybe this is the issue..Sensor not running in its ideal zone..

Might have to search again for another one and endure the agony of another 10 hour print..and the multiple restarts associated with getting it to work

classicrocker883, 2023-07-16:
I'm just reading that story now, sorry to hear that happened like that! air lifted by helicopter? if that happens to anyone here that is big news. you are healed and okay now?

must be nice to be on an island I can only imagine. also you live near the D-day Normandy landings? have you been? I like the WWII stuff it's very cool, I would like to go one day.

about the `trammingwizard`, I have been testing out the more recent code changes and I can't recreate the issue where it increases in value. the thing is there isn't anything I can see in the code which would account for it. that is I had originally thought the same thing happened to me, I thought I was getting a wrong reading, but it did not increment after each time. it was that certain spot on the bed which was different height from the spots around it - tiny piece of filament - which made me think that the trammingwizard was off because building the bed Mesh made it seem tilted because of the offset value from tramming.

I hope what I just said makes sense - - anyway I'm saying that this could be an issue with the probe itself rather than firmware. but I will send you the updated version for you to try out to see, hopefully it's not the probe. because I am unable to recreate the same issue so if it is the firmware it is hard to find out exactly. but we will see once I upload that new file for you.

looking at the values listed above which change, there seems to be a slight pattern, but it's not exact. I do remember that I had been getting wrong readings if I used paper to tram. such as for the Z offset. if the Z offset is not correct, it may account for the different values increasing when you tram, and then resets when you rehome.

try changing the Z offset and see, by then I'll post the new firmware update. if you want a true Z offset, this is what I do and suggest to you - start by going to the Z offset wizard. 
then Home Z. select Center nozzle, it should bring the nozzle tip to the bed. look in between the nozzle and bed - there should be no gap. also, it should not be pushing down too far. so to reset the Z offset, have it to 0.00 or any value where it is just above, where there is a gap. then slowly bring down the nozzle until it Just touches the bed surface. if you put paper in between, this can mess up the value, but you won't scratch anything this way because it's only going up and down, and just make sure it doesn't go too far down. (optionally - what you can do with paper (or feeler Guage) is measure the thickness, do the sort of paper tram thing feeling when it grabs, then subtract the thickness from that Z offset value.)

then you should have a true Z offset. this can be why the values are off from each trammingwizard run. let me know if that has changed anything.

classicrocker883, 2023-07-16:
https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-1t3

it should be posted here, you can test the updated firmware, let me know how it goes

npaisnel, 2023-07-17:
Yes fully healed, Tibial Nail fitted.  Helicopter off the road then, Citation jet back to home.    The joke is I fly air ambulance fixed wing for the island..and they dod not come and get me..got another company to do it! 

D-Day beaches, yes just across the water.  When I ws flight instructor for the local aero club that was one of our trial / demonstration lessons.  90 minute flight from the island across the Normandy peninsula, out across the beaches and the the still visible wrecked ships, the Mulberry harbours, back via St Mere Eglise, where the American paratrooper got hung up on the church spire.  They still hang a parachute from the church spire in remembrance.

npaisnel, 2023-07-17:
New firmware on 
Adjuster nut inset adjusted
Z probe offset re set 
Manual tramming done
I have a 0.009 mm feeler gauge
I use the ..look along the bed with a lamp method ..to see the reflection of th tip in the glass. 
been around the corners probably 5 times..Bed tramming spot on using feeler gauge/ view the reflection/ no gap.  
Z- offset set.  Auto Home Centre/ Adjust z-offset  -2.31
It use to be up at 3.somethinig  ..but I have made a new mount with 1.25 mm higher mount point  S the probe is within the 2.3 -4.3 misplacing recommend from the extruder tip

But the incremental increase still continues..see following 

I do have a spare new probe to fit on the Geeetech that I have not yet fitted .  Can try that later.

npaisnel, 2023-07-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

npaisnel, 2023-07-17:
[image]
[image]
[image]
[image]

npaisnel, 2023-07-17:
The new firmware. 
three times now, since the successful Auto Bed levelling run, it has frozen at the point below
Encoder wheel does not move to cancel, and clicking for confirm Also does not run the bed levelling

[image]

npaisnel, 2023-07-17:
Just trying the GD32 UBL Pro UI

npaisnel, 2023-07-18:
A few Normandy Pictures for you.

last one you can see the line of Mulberry harbour units that were floated across and then sunk

 [image]
[image]
[image]
[image]

classicrocker883, 2023-07-19:
oh wow are you flying that plane??? that is amazing I can see the beaches.

it looks like your BL touch isn't working properly, the deviation should not be that much and that would explain why it increases every run. did I mention before about using a ferrite core?
<p>[image]</p>
<br>

it helps any electric interference which may be causing the over deviation. 
as for the knob not working for confirm or cancel I think I can fix that I know what may have happened.

classicrocker883, 2023-07-19:
I had no idea about the wreck ships still visible, or the paratrooper story. I used to study as a history major in college before I went into engineering - that is why I would have to go there to know all the true history, which doesn't really get retold elsewhere.

classicrocker883, 2023-07-19:
trammingwizard corner values are increasing just about the same as the deviation... interesting. where it says extra probing, try from 0, do 2. 

if you select 1 it will revert to zero, unfortunately since the May update 2 times multiple probing just doesn't want to work like it did before. So I rename the option to "Extra Probing" instead of "Multiple Probing" as to not be confusing. I would like the double probe back, however that doesn't work right since the update, so it's either 0 - no extra probes, or 2 or more - adds 2 or more extra probes. so having it at 2 means it probes 3 times. and 3 times is just too slow. even with high speed HS mode.

technically slow (not HS high speed) is more accurate than fast. but you experienced slow (HS off/ unchecked box) to be less accurate? 

ideally, like before, it should be like this: probes 2 times - first time fast, second time slow, return average.

for Extra Probing, instead it is either 1 fast, (option 0), or 3 slow (option 2). 
when I tested trying with (option 1), it would not raise the probe first before deploying, the code is in the ProUI library which I am unable to edit because it is precompiled... it's possible to reverse engineer to remake the code, but that is an endeavor I'm not qualified to take, even so it would take ages to figure out and in the end it would not be the same. so I had to make do with what we have... and that was to leave option 1 out.

classicrocker883, 2023-07-19:
from a guide 
> Step2 and 3: [image]
> 
> [image]
> 
> The most important step is the grounding of the gantry, as shown in the first picture. If the problem persists, replicate the fix in the second and third pictures too.
> 
> Check the grounding is a success by performing an electrical resistance check between the case of the meanwell power supply and the screw holding the electromnics enclosure closed.
> 
> Good luck buddy!

npaisnel, 2023-07-20:
Yes, I fly the local air ambulance now... King Air B200

https://en.wikipedia.org/wiki/John_Steele_(paratrooper)

https://archeologie.culture.gouv.fr/epaves-debarquement/en/les-rois-de-la-ferraille

Lots to reply to ...and I am now helping with creating a contour map for someones local fields..

Ferrite cours..yes I have some somewhere...

Grodun gin the beam..hmm Interesting .

Twin lead score on Aquila...not jsut yet.... 

The twin extruder to a single..that is on another machine ..a GeeeTech i3Pro C..now a B...that has twin lead screws..jsut added 8mm 4 start screws and given each stepper its on wn deriver ..from tehgspare extruder driver...  also relocated the y end stop to give 18 mm more travel...changin to single extruder has given eh ned extra width too...  Just need to upgrade the bed next...need to get a piece of glass cut and create a thicker base unit..and get rid of those dumb paper clips that hold the glass on ...

Will try and get back to his when Inext have time..but field contour mapping the priority this week......unless I get called in to fly ..that is the advantage of doing air ambulance..my time is my own until the bell goes and I gotta rush out to fly.

Thanks for the links......gotta fly ..literally ..

npaisnel, 2023-07-20:
oh does the Aquila support twin drivers for the z lead screws or do you have to parallel the Z ..like the GeeeTech was originally ?  Having twin drivers and Auto Align gives more driver power.  Yes I have fitted anti backlash on the lead screws...

classicrocker883, 2023-07-21:
hey I posted an update to the firmware,
[you can try your version here](https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-2b)

I'm still unable to recreate the issues you been having such as the Start Cancel for auto build mesh freezing on the screen. is that still an issue after trying the newest release?

classicrocker883, 2023-07-21:
> oh does the Aquila support twin drivers for the z lead screws or do you have to parallel the Z ..like the GeeeTech was originally ? Having twin drivers and Auto Align gives more driver power. Yes I have fitted anti backlash on the lead screws...

no the Aquila doesnt have two Z drivers. just one for each axis including E, the extruder. so yes it is connected in parallel. I really dont have an issue with this. Auto Align would be a good feature, but without an extra driver its not available. What I was able to do was take the motor from the X axis and move it two my 2nd Z motor. that way both motors are identical. I also increased the voltage ref because they are in parallel. and because of the anti-backlash nut, I should probably increase the voltage ref more since the motors aren't that warm to touch. the more work the motor needs to do, the more power is needed.

I believe there is a youtube video demonstrating exactly how much power. Victor Bared -[Steppers too hot? How to set the correct stepper voltages](https://www.youtube.com/watch?v=2m5voPdFzEc)

npaisnel, 2023-07-21:
I have just downloaded it ...I'll test in the next few days.

On you8r Aquila..can oyu connect to a computer via USB and drive it / send commands via something like repeater Host?  
 I have been able to connect and Monitor what it is doing but have never had any control.

I have d/loaded the GD32 UBL Pro UI which is what I have been using up until now.
Buy kitchen workbench is cover in a stripped GeeeTech i3 prussa clone at the moment. Trying to get a bed glass cut to maximise print area.

I have just set up Auto Align on the GeeeTech.  and purposely turned one stepper motor half a turn out..so 4mm off try ..it sorts and correct that witt about 6 iterations.

Looking at video s next...

npaisnel, 2023-07-21:
``I have not kept notes on how to re create that Start freeze..so will ahve to read back up the thread and see what I said at the time of the error.

Are you on Facebook ? I use that a lot ...mainly messenger.. Neil Paisnel...

npaisnel, 2023-07-22:
I just do not know any more.
Another 5 hours wasted on this F*^** tu
[image]
rd of a machine.

Bed Tramming shows level, mesh created…. Z offset set.   Saved mesh to all slots.. 

then it goes and does this..  gouging the extruder through the bed at one corner.  

Manually move the head with steppers disabled and lower to z zero… with print head hanging over the side of the bed, and Z 0.00 puts the nozzle about 1.3mm below platform height.  

fuck knows…

can this machine be driven via USB so I can send commands from Repetier ? 

[image]
[image]

npaisnel, 2023-07-22:
It is almost as if Z-offset some times gets added to itself...

I have just done a good 0.2 mm 20x20 mm flat squares..had to manuyally use the Tune Z-offset menu to get it to adhere 
Yet when I go in to Menu afterPrint finsihwsed. Do an Auto Home and centre and then go to the bed Offset wizard, if I use 0.09mm feeler gauge under the extruder...the gauge gets stuck at anything above -1.31 mm offset.  If I set the 2.45 you can see the z stepper pulling the gantry down on the LHS and the RHS pivoting up ..with the extruder as the pivot point..

A manual bed levelling with the feeler gauge shows the corners level 
an auto tramming wizard with Calculate averages selected, shows all corners green with 0.00.  yet it tells me to adjust the level as it is not level..

npaisnel, 2023-07-22:
[image]

classicrocker883, 2023-07-22:
ull try to answer best I can, yes you can control the printer through USB, I'm not familiar with Repitier, but I do know you can with Cura, octoprint, Voxelab Voxelmaker (I think since it's 3rd party firmware), Slic3r, 3dprintsos, I think Prusa Slicer. but I usually use Pronterface or Octoprint.

to reset EEPROM go to Control menu, then Advanced settings. reset settings should be next to Load settings. or you can issue the M501 or M502 command I forgot which. I know M500 is to save and M503 is view settings.

So you are having issue with the nozzle and the print bed? If the nozzle is off the edge of the bed, you may have to adjust home offsets. I myself haven't needed to use the home offsets but it should be easy to figure out. there's an option in the firmware (I think in the control menu) where you set that current position as home which is X0 Y0 Z0. 
I know you can set the values in Cura as well. 

also, if you need more information check out Mriscocs repository which this one is a fork of. look under the Wiki tab. Because I know there are many other Gcode commands which you should be able to issue that are custom only to this firmware. 

as for the nozzle up and down... you'll need the correct Z offset. but if you're having issues with the bl touch probe then it will be difficult getting everything correct. 

and yes I have Facebook I can message you there. 

don't forget Reddit has a good community you can post any issues there someone is always ready to answer. r/VoxelabAquila and r/Ender3V2firmware

classicrocker883, 2023-07-22:
that is a good observation that the corners show 0.00 yet say to lower front left. for whatever reason it selected that value. I'm sure I could figure out how to write the code so it says what it should say. I do know when you start trammingwizard after the first time it will display something like this before the first corner is probed. was this during that time or this was the result at the end?

classicrocker883, 2023-07-22:
I'm still amazed you fly planes that is something I wanted to do for a long time. I was once I  the civil air patrol, which is a US air force auxiliary. it's basically the military for under age 18.if you don't know about it you meet once a week, do drills, learn aviation, get to ride in  black hawk helicopters or Cessna single prop (rarely). I wasn't in it for long but it would have been cool to become a fighter pilot.

i still have flight simulators which are awesome. one is a ww2 combat simulator and I was kicked from playing once cause they couldn't get a hit on me so they thought I was cheating. lol...

but I just read about that paratrooper caught in the church steeple. what a story. I had no idea he was portrayed in that movie.

npaisnel, 2023-07-23:
> I'm still amazed you fly planes that is something I wanted to do for a long time. I was once I the civil air patrol, which is a US air force auxiliary. it's basically the military for under age 18.

We ahve similar here..the Air Cadets and other similar branches for the other forces too.

npaisnel, 2023-07-23:
> ull try to answer best I can, yes you can control the printer through USB, I'm not familiar with Repitier, but I do know you can with Cura, octoprint, Voxelab Voxelmaker (I think since it's 3rd party firmware), Slic3r, 3dprintsos, I think Prusa Slicer. but I usually use Pronterface or Octoprint.

Yet my Version of Cura on the Mac will not control it, neither will Repetier..yet Repeater does control the Geeetech i3 with the GT2560 board.    have to have a play again.
> 
> to reset EEPROM go to Control menu, then Advanced settings. reset settings should be next to Load settings. or you can issue the M501 or M502 command I forgot which. I know M500 is to save and M503 is view settings.

Ah yes, that Reset..I was looking for an actual menu item with the word EEPROM in it as not he straight Marlin 2.1.2.1

> 
> So you are having issue with the nozzle and the print bed? If the nozzle is off the edge of the bed, you may have to adjust home offsets. I myself haven't needed to use the home offsets but it should be easy to figure out. there's an option in the firmware 

Ah no, no problem with it ..I purposely set the X max setting to over sized, then micro steped the X motion to see how far it went. think 234.5..it hits the end as I think 236.  I wanted to be able to lower the nozzle BELOW the bed whilst monitoring the probe tip.  So i coudl accurately check the measurement of the tip to nozzle clearance.     I am still yet not sure if that figure is the Z-Offset ..or if the Z-offset is the distance between nozzle and BLTouch Trigger point..which is different.

I have since tried three different identical probes, added ferrite cores and looked at grounding   all produce zero difference.

> also, if you need more information check out Mriscocs repository which this one is a fork of. look under the Wiki tab. Because I know there are many other Gcode commands which you should be able to [...]

I think I am just expecting too much form a cheap machine that is after all badly designed, cantilever design, single Z screw, platform supported in the middle.  If this was a piece of mechanical / engineering workshop kit, I'd be take-in it back and say not fit for purpose..cheap crap ..refund please 

> don't forget Reddit has a good community you can post any issues there someone is always ready to answer. r/VoxelabAquila and r/Ender3V2firmware

I am thinking we have both wasted enough time on this for now..  I have the GeeeTech to sort too.  Got twin Z steppers on that, that I have just got working with Marlin 2.1.2.1 with both steppers on individual drivers channels, and the alll three BLTouch probes works fine on that ...Auto Align etc.  Want to look at parallel wiring the Z-stop switch this morning ..and do a job that has been cluttering up my kitchen since March...replace backup battery in Garmin GPS296,   3-d printer mods ahve taken over and all sorts of other stuff was shelved.

Now time to shelve this printer and leave for another day!

Thanks for your help

speak On FB later..

classicrocker883, 2023-07-23:
Yes it is called "Restore Defaults". 

Z-offset really has not much to do with the distance between the probe tip and the nozzle. for instance, just imagine youre homing the Z axis without the probe, using the stop limit switch that is to the side infront of the Z axis motor. say you go to Home, the nozzle will move all the way down to the bed: Z=0. the z-offset is there to account for any differences to "offset" the home value. you can also bring the stop limit switch up and down, which will also change the Z=0 home position, right? then you can adjust the Z-offset still so the nozzle reaches the bed.

okay now take that information and apply it to the probe. it acts exactly the same way, except it is located on the hotend extruder and takes its measurement in the center of the bed. the probe tip extends/retracts so that the nozzle can reach the bed/doesnt run into the bed. the length of the tip isnt crucial, just as long as the previous sentence can happen.

say you put Z-offset as 0.00, and Home the machine to X0Y0Z0, and the nozzle doesnt actually touch the bed, you use the Z-offset to account for that difference. dont need to take anyother measurement in consideration.

npaisnel, 2023-07-23:
I'd still say the Z-offset when using a probe mounted on the extruder is totally to do with the probe tip (and the activation distance) and the extruder tip 

With a Z stop switch , the Z- zero position is fixed by the switch position.
With the Z-stop end of travel being the activation of the probe, and the probe being fixed to the extruder, the Z-Stop and therefore Z home 0 is always at the same point with regard to he extruder and the bed.  
If you  screw the adjusters down fully, Z0 is as low as it can go..get a silver marker pen and mark the vertical tracks.  Now unwind the bed adjusters till there is very little tension on the springs... and do a Z- home again.  
The Z- probe will still stop the Z travel at exactly the same position above the bed.  The distance it stops above the bed is defined/ determined by when the Z-probe gets 'kicked' by the bed.  Its relationship between it and the extruder has not changed.  You could lift the bed 20mm and the Z home position will still read zero, but the gantry will be 20mm higher 

With a probe mounted on the gantry, the Z- Offset is a fixed distance based onto distance between probe activation point and extruder tip.  The Z stop is moving with the extruder 
 
As the gantry lowers with the probe deployed, it will always activate at the same point above the bed.

I was trying to work out the relationship between z probe max deployment length and its activation point

npaisnel, 2023-07-23:
the BL Touch sensor, they recommend 2.3-4.3mm ( from memory) probe tip to extruder tip distance.

these are hall effect sensors, so not true on/off like a micro switch.  I'd liek to know when the circuitry within the switch that is monitoring the motion determines that contact has been made....  I'd guess the range the manufacturer specifies is to give greatest accuracy with triggering.  if the probe hits the bed as the gantry is lowering gand the tip is out side the specified range..then does it have to move some degree, to get within the specified range...before the activation 'trips'.. dunno.

classicrocker883, 2023-07-24:
hey @npaisnel i uploaded a special firmware for you to try out
you can find it in the latest Aquila release - with your name on it! [here is the direct link](https://github.com/classicrocker883/MriscocProUI/releases/download/2.1.3d-2c/Neil_GD32_UBL-ProUI.bin)

so I fixed the mesh viewer so its more accurate instead of limited to 0.20. and the trammingwizard where it used to say to lower (corner) even though its all 0.00 and level. that is fixed.

classicrocker883, 2023-07-24:
[image]
[image]
[image]

this is what the bed Mesh should look like - the values (more or less).

and the deviation as well. I ran the M48 test several times and I got both these same exact values. it was very consistent.

npaisnel, 2023-07-25:
> 

I’ll try it in a bit.
Thanks
On a day off today then back on call from Wednesday.

npaisnel, 2023-07-25:
> well you arent wrong in your description. I mean to say that although there relationship between probe tip and nozzle tip has an "offset", for all intensive purposes the printer doesnt take that physical value into consideration per say. when we are talking about "Z-offset" using the printer firmware, that is the relationship between where the printer's Z Home position and the actually nozzle tip. that is what I mean by Z-offset.
> 

Great, so the Z offset in firmware is related to the Home position… but the home position is found by the printer… by using the Z -Stop of the probe. So, if you were to move the probe say 2mm lower, further away from the tip, then when you do an Auto Home’ the new home position after moving the probe is now 2mm higher… So the Zoffset would need to now be 2.55mm

Have managed to get USB Control now, as I said on FB, it was the USB 2 front ports it did not like.  USB 3.0 add on board works perfectly.

> just like my printer setup, i set my Z-offset to -0.55. the "Home" position for the probe is 0.55mm downward from the nozzle tip. now I dont know what the length of the tip is, so I was saying it can be as long as it wants. the gantry can travel as far until the probe is triggered, which happens to be -0.55mm from the nozzle tip. because the Probe itself is in a fixed position, so too is the triggered position.

Is your  probe a 3-D BL Touch or Creality one ? Just been reading about the difference.  Creality uses and optical pin sensor inside where the 3D brand uses a Hall sensor.   Looking at your images, it does totally look like a machine/ accuracy issue… bringing me back to it being a poor machine or similar.   I have swapped out the BLT between three probes and all give same inconsistent results. 

The recommendation for my sensor says that with the pin stowed it should have a clearance of 2.3-4.3mm higher than the nozzle. 
via Cura and USB I  used M401 to manually deploy, then lowered z axis until it just touched.  My probe needs to then go another 1.4 mm until it triggers.

> 
> but as soon as I say swap my nozzle, the Z-offset can change. it may not be -0.55 because the nozzle tip is in a different physical position.
> 
> I dont have a 3D printed mount for my probe, it is a CR-touch and I used the included brackets, so in my case it works well and the probe tip to nozzle tip offset isnt a factor for me to worry about.

Ah.. CR-Touch… so likely the optical one..  hence your accuracy.

> 
> I guess that is where the confusion is... Z-offset relates to the Z axis of the Nozzle and Bed, just as if changing Home offsets for X or Y. maybe we should call the other "probe tip-offset" which relates probe tip and nozzle tip. has nothing to do with bed because that is the "Z-offset". fair enough?

:
I still have it in my mind that z-offset is directly related to probe to nozzle offset, simply because 

This is my thought process

1) Offset relates to Home position 
2) Home position relates to Z - stop trigger point
3) Z-Stop Trigger point relates to BLT mount position.
4) BLT Mount position fixed in relation to nozzle .

Therefore 
move BLT mount position, 
trigger point moves ( it is fixed at 1.4mm after probe bed contact) 
Home position then also moves
So z-offset has moved same distance as probe mount point has moved.

I am wondering if this confusion has arisen because your CR-Touch probe does not have a fixed trigger position… being optical… so you can adjust separately..

Very impressive level mesh, 

don’t think I will achieve that without new heated bed / glass and sensor…. but as you saw from the chess pieces, once I have first layer down, all is good enough, it does work.  

I tried sending the G26 command to print the mesh layer test print…. Something you can add in firmware ? 

> 
> anyway while you can adjust that in software, like in the slicer, I never had to myself. I always adjusted the Z-offset using the LCD interface on the printer. I hope that helps.

classicrocker883, 2023-07-27:
> I tried sending the G26 command to print the mesh layer test print…. Something you can add in firmware ?

that is enabled and should work. except just G26 wont cut it, youll need a few more parameters as explained [here at the Marlin site](https://marlinfw.org/docs/gcode/G026.html)  
OH, you know what?? I had G26 disabled because there simply wasnt enough room with the ProUI.  
You can try the *_UBL-NoPro* version which has a lot of other options I simply could not fit otherwise.  
The ProUI of the *_BLT-* version has it enabled, however for some reason the mesh doesnt save after restart - that is something I dont know if its meant to do that or not, but im trying to work on that little issue.

also there is a way to visualize the mesh, this website can explain it. I also believe M503 will also display the values in a grid through the terminal. Apparently G29 can too.  
https://3dwork.io/en/visualize-3d-mesh/

> So, if you were to move the probe say 2mm lower, further away from the tip, then when you do an Auto Home’ the new home position after moving the probe is now 2mm higher… So the Zoffset would need to now be 2.55mm

kind of... you'd be right if you meant (negative) -2.55mm. so *Negative* Zoffset values **Lowers** the Nozzle, *Positive* values **Raises** the Nozzle [from the Home 0.00 position].

>via Cura and USB I used M401 to manually deploy,

there is a way to do that through the LCD screen itself. It is under the Leveling menu, Probe settings. right above `M48 Probe Test` you will see `Deploy Probe` | `Stow Probe` | `Reset Probe`  
They do what you expect, *Reset* is basically only used if the other commands arent working such as if it runs into a fault like if you were to manually trigger it or something like that.

>Ah.. CR-Touch… so likely the optical one.. hence your accuracy.
...  your CR-Touch probe does not have a fixed trigger position… being optical… so you can adjust separately..

yes the CR-touch is a bit different than the others, I never had a problem with it. I'm not sure how you mean doesnt have a fixed trigger - that I can adjust seperately. As far as I can tell I dont think im able to do any sort of adjustments regarding that. its fixed and rigid as you can expect. as far as I know it works just like all the other types. 
- Gantry Raises
- Probe Extends
- Gantry Lowers
- Probed gets Triggered
- Gantry Raises 2nd time
- Then moves *XY* so Nozzle in place of Probe tip (also depends what its called to do)

>Therefore
move BLT mount position,
trigger point moves ( it is fixed at 1.4mm after probe bed contact)
Home position then also moves
So z-offset has moved same distance as probe mount point has moved.

yes I supposed that is one way of putting it. If you move the BLT mount (up or down), that in turn changes where the printer thinks Home position is, because the trigger point has moved. and since it has moved in relation to nozzle, Z-offset should be changed to account of that difference.

I'm not so great with recording video but as soon as I can I will film the process I go through hopefully give some insight explaining my reasoning and steps. is there anything else before I start?

and I wanted to say make sure the bed surface is absolutely clean. I use alcohol to clean mine before every print and never have an issue. I suppose having all kinds of stuff on the bed isnt all that bad because I've seen some pictures of other peoples printers and its really atrocious. they used glue and hairspray and what not and it is caked on. i almost feel sorry for them. its like those nasal inhalers for allergies. yeah it helps, but eventually you become dependent on it.

the thing is, its fine if you dont use a bed probe i suppose, but once you do how do you get around all that junk throwing it off? maybe they dont, i dont know.

I did try tape and glue, and it does help...but its not necessary. im not saying this about you, but if anyone is ever going use that stuff, its best to put down a layer of tape first to prevent it from building up on the bed. I found thats what worked for me (for a while) when I couldnt get PETG to stick. I think it was because the bed was kind of new, it wasnt "broke" in yet. So after using it a bunch and cleaning it before every print, things just started to STICK.

and remember the bed has TWO sides to it. you can always flip it over and give it a try.

classicrocker883, 2023-07-28:
https://www.youtube.com/watch?v=W8ouBPnRV4s

this should shine some insight to the situation. granted hes using paper, so the nozzle will still be the thickness of the paper above the bed. so then you just subtract the paper thickness and the nozzle should just be touching, otherwise its no good if there is even a small gap.

classicrocker883, 2023-08-03:
check this out, sounds just like what you experienced 

https://github.com/mriscoc/Ender3V2S1/discussions/976

npaisnel, 2023-08-04:
This bed has a coated and non coated surface, 

seems close enough now and just need a fine tune as I start a print.

yes, it does sound similar.