classicrocker883/MRiscoCProUI #58: [BUG]  Bed mesh unuseable.

Vondanta, 2023-07-27:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Got a stock N32 Aquila with a 3dtouch and got everything working exept anything related to mesh. If i start a auto mesh it homes and tries to find z endstop outside of buildplate causing the mesh to stop. Bed is 235x235 so i'm not sure what's going on here. Using IS version atm if that's relevant.

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

N32

### Add-ons that could be involved

3d touch

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Vondanta, 2023-07-27:
^ clarification: it's the first mesh point that fails.

classicrocker883, 2023-07-28:
you set the bed to 235x235? how about the Mesh Inset?
what I can tell is that because you set the bed to 235x235, it thinks its a larger bed.

I have an Aquila and it Physically cant move past 230 in X and 220 in Y.
granted I use bed clips and the printable area remains 220x220.

I'm guessing you have a PEI sheet so you can print at the full bed size?  
So what I would do is pace around the bed to see how it lines up. It could be that the X or Y end stop limit switches offset where the printer thinks the bed position is.  
In other words, you may need to set Home Offsets to what works for you.

but as I was saying before that, check where your bed limits are. set the printer to move to each size Max and Min position. note where the Nozzle lines up.  
for example: X0 Y0, is the nozzle on the bed or off the bed?  
X235 Y235, same question on bed or off bed?

and if one side is on the bed, how far away is it from the edge? because if you're to use the FULL 235 square bed, the nozzle should be at the edge of each Max and Min - X and Y coordinate. if not, you can either adjust the Home Offsets, or even the Physical bed Size. if say the nozzle is at the edge and not off it, then for building a mesh you can adjust the Mesh Inset, by bringing it in a tad (increase Min inset, decrease Max inset).

Vondanta, 2023-07-29:
> you set the bed to 235x235? how about the Mesh Inset? what I can tell is that because you set the bed to 235x235, it thinks its a larger bed.
> 
> I have an Aquila and it Physically cant move past 230 in X and 220 in Y. granted I use bed clips and the printable area remains 220x220.
> 
> I'm guessing you have a PEI sheet so you can print at the full bed size? So what I would do is pace around the bed to see how it lines up. It could be that the X or Y end stop limit switches offset where the printer thinks the bed position is. In other words, you may need to set Home Offsets to what works for you.
> 
> but as I was saying before that, check where your bed limits are. set the printer to move to each side (X+Y) Max and Min position. note where the Nozzle lines up. for example: X0 Y0, is the nozzle on the bed or off the bed? X235 Y235, same question on bed or off bed?
> 
> and if one side is on the bed, how far away is it from the edge? because if you're to use the FULL 235 square bed, the nozzle should be at the edge of each Max and Min - X and Y coordinate. if not, you can either adjust the Home Offsets, or even the Physical bed Size. if say the nozzle is at the edge and not off it, then for building a mesh you can adjust the Mesh Inset, by bringing it in a tad (increase Min inset, decrease Max inset).

I truly appreciate you taking your time to get to me with a thought out message!
For anyone reading this the issue was a setting called "HS" <- could be called something else, i'm unable to check atm. I think it's a setting that should be off if using a non genuine bltouch. Since mine was a 3dtouch from China it makes sence to me.

Once again, thanks for getting back to me!

I do have one question, does it have thermal runaway protection? -cant test for a while as to why i'm asking rather than testing.

classicrocker883, 2023-07-30:
hey no problem. "HS" stands for High Speed, as in BL-Touch High Speed. all it does is increase how fast it probes up and down. this option can be set in the 'Probe Settings' menu, in the 'Leveling' Menu.  
It really is something that _should **Not**_ cause any issue on or off, but if that *is* causing something to not go right (when its enabled or disabled) then please - by all means take note of the steps to reproduce it and give as much detail describing what is happening. try HS both on or off and record the differences. That way it would help me figure out and solve the issue.

I say that because I do not have any other bed Probe other than a CR-touch and those things are legit bullet-proof. never had an issue.

Now I am wondering, your issue is fixed?? Turning off `Enable HS mode` solved it?

You're going to have to explain to me because I don't see how that can be related to the original issue. So to start you said the probe was finding itself off of the bed when probing the bed to build a mesh. That doesn't seem related to the Probe's rate of speed. Because `HS Mode` relates to the up and down motion - Z axis. What you described in the issue is X/Y plane. Sorry, did I miss something?

and yes to answer your question 
> does it have thermal runaway protection?

this firmware does have thermal runaway protection enabled - for both Hotend and Bed. I would be careful attempting to test this out. you could research into it and find a safe way to do a test. I do know how to do it the quick and easy way but takes risk of damaging something so I wont get into that here.

closed 2023-11-03 by github-actions[bot]