classicrocker883/MRiscoCProUI #60: Aquila X2 N32 screen jumbled and unuseable

Vanilla009, 2023-07-30:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

After flashing Aquila_N32_Default-NP-ProUI-IS.bin to the Aquila X2 with an N32 board, the screen is jumbled up, making it entirely unusable.

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila N32

### ProUI?

ProUI

### Bed Leveling

Default - No Probe / No Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

An image of what my screen looks like.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
The output from M111 S247 simply gave me endless echo:M105 messages.

classicrocker883, 2023-07-31:
you will need to flash the firmware on your LCD  
[this will show you how](https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display)  
or you can check [this readme as well](https://github.com/classicrocker883/MriscocProUI/tree/2023-July/display%20assets/Aquila%20Display%20firmware)

Download the ZIP to the repo and navigate to [here within the folders](https://github.com/classicrocker883/MriscocProUI/tree/2023-July/display%20assets/Aquila%20Display%20firmware/Firmware%20Sets)

choose one of the DWIN_SETS, I would stick with Original or Voxelab Red
[here is a Direct Download](https://minhaskamal.github.io/DownGit/#/home?url=https://github.com/classicrocker883/MriscocProUI/tree/2023-July/display assets/Aquila Display firmware/Firmware Sets) link I created so you do not have to download the whole repo if you dont want to.

Vanilla009, 2023-07-31:
Thanks for the quick response! Your solution worked perfectly. What size should the thumbnails be? They aren't working for me.

classicrocker883, 2023-08-01:
> Thanks for the quick response! Your solution worked perfectly. What size should the thumbnails be? They aren't working for me.

thumbnails for the G-code model preview when you select to print?

they should be 200x200, I should make a note suggesting that, but I was able to test using 230x180 and it worked fine, stretched, but fine.

what are you using for slicer? you followed the directions of putting the script in the correct folder?
another thing you need is Python installed. and it says you need to use `pip` to install Pillow

Vanilla009, 2023-08-11:
I'm using Prusaslicer, Python installed, with pillow, 200x200 size, png. Prusaslicer throws an error - I have tried reinstalling Python and updating Pillow. I'm using Python 3.11.4. 
[image]
Here is the error message Prusaslicer gives me.

classicrocker883, 2023-08-12:
that address doesnt show up for me. I have python3.11.4 installed here  
(open up CMD - command prompt, and type the following)
```
C:\Users\Andrew>where python
C:\Program Files\Python311\python.exe
C:\Program Files\msys64\mingw64\bin\python.exe
C:\Users\Andrew\AppData\Local\Microsoft\WindowsApps\python.exe
```
for whatever reason I seem to have two different ones installed
```
C:\Users\Andrew>python --version
Python 3.11.4

C:\Users\Andrew>python3 --version
Python 3.10.12
```
```
C:\Users\Andrew>where python3
C:\Program Files\msys64\mingw64\bin\python3.exe
C:\Users\Andrew\AppData\Local\Microsoft\WindowsApps\python3.exe
```

`python` and `python3` are different (for just me at least)

you should add one of these paths to the *PATHS* environment variables. there is some documentation written up on it if you dont know how to do that.

classicrocker883, 2023-08-12:
[image]

your slicer should look like this,

first image is in `Printer Settings` tab - General  
then `Print Settings` tab - Output Options

closed 2023-10-13 by classicrocker883