classicrocker883/MRiscoCProUI #62: Aquila x2 when trying to save a mesh, the printer reboots and resets all settings [BUG]

PolcovnicGauss, 2023-08-03:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

when i try to save heightmap in firmware HC32_BLT-ProUI.bin
then it just reboots when trying to save, if you use the firmware HC32_UBL-ProUI-MPC.bin
 or HC32_UBL-ProUI.bin
 then the printer starts reading the height map, after which at the 37th or 38th point out of 49 it simply exits the process of reading the mesh in the menu, while it does not restart and does not save the mesh

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila HC32

### ProUI?

ProUI

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

PolcovnicGauss, 2023-08-03:
I'm also not sure if linear advance works if there were no special changes on the models to change the k-factor

classicrocker883, 2023-08-03:
yes there is currently work being done to fix that issue. it will take some time for tests to be done but hopefully soon we can have a release to be good to go.

PolcovnicGauss, 2023-08-04:
I also tried to use HC32-MM-ProUI, I even managed to print the part, but after 5-10 minutes the temperature started jumping, the display said that the temperature was 206/230, the hotend started to crackle, after a couple of seconds the temperature on the display was updated and became 278 /230

PolcovnicGauss, 2023-08-04:
> yes there is currently work being done to fix that issue. it will take some time for tests to be done but hopefully soon we can have a release to be good to go.

It turns out now there is not a single stable firmware?

classicrocker883, 2023-08-04:
> > yes there is currently work being done to fix that issue. it will take some time for tests to be done but hopefully soon we can have a release to be good to go.
> 
> It turns out now there is not a single stable firmware?

they should be stable aside from the issue when saving. that is the only real reported issue.

When you had the temperature jump that sounds more like something with the thermistor and not firmware.  
the reason why it jumped and crackled was it readi a low temp, and started to increase the temperature. but since the temp wasn't exactly reading back correctly it heated up higher than it meant to.

So I would definitely check your thermistor, look into reseating into the hotend, or replacing it. If it were me, those things are cheap, and you should replace it right away.  
You should think about upgrading to a cartridge style which requires a heater block to support it. again these parts are reasonably affordable and have no trouble finding it.  

[WINSINN 3D Printer HT-NTC100K Thermistor](https://www.amazon.com/dp/B07F3SMRLJ?ref=ppx_pop_mob_ap_share)  

I recommend either [High Temperature Plated Copper by ChowThink](https://www.amazon.com/dp/B0B12FX9F1?ref=ppx_pop_mob_ap_share)  
or  
[High Temperature Plated Copper by POLISI3D](https://www.amazon.com/dp/B08NVTJM4S?ref=ppx_pop_mob_ap_share)

classicrocker883, 2023-08-04:
https://github.com/shadow578/Marlin-H32/issues/37#issuecomment-1662582007

this is the process of the fix going on. I will be experimenting with the `core-rewrite` before making a release.

PolcovnicGauss, 2023-08-05:
> > > yes there is currently work being done to fix that issue. it will take some time for tests to be done but hopefully soon we can have a release to be good to go.
> > 
> > 
> > It turns out now there is not a single stable firmware?
> 
> they should be stable aside from the issue when saving. that is the only real reported issue.
> 
> When you had the temperature jump that sounds more like something with the thermistor and not firmware. the reason why it jumped and crackled was it reading a low temp, and started to increase the temperature. but since the temp wasn't exactly reading back correctly it heated up higher than it meant to.
> 
> So I would definitely check your thermistor, look into reseating into the hotend, or replacing it. If it were me, those things are cheap, and you should replace it right away. You should think about upgrading to a cartridge style which requires a heater block to support it. again these parts are reasonably affordable and have no trouble finding it.
> 
> [WINSINN 3D Printer HT-NTC100K Thermistor](https://www.amazon.com/dp/B07F3SMRLJ?ref=ppx_pop_mob_ap_share)
> 
> I recommend either [High Temperature Plated Copper by ChowThink](https://www.amazon.com/dp/B0B12FX9F1?ref=ppx_pop_mob_ap_share) or [High Temperature Plated Copper by POLISI3D](https://www.amazon.com/dp/B08NVTJM4S?ref=ppx_pop_mob_ap_share)
> 
> 
> the reason why you should replace it right away is it may be compromised. those glass things break easily. Once I started to heat my bed up to 100° for the first time to print ABS, and this was after at least a year of printing PETG before at 80°, hours and hours at that temperature. but once it got to 100 I went to start a print and the thermistor broke. this then caused some short in the mainboard chip and make the display freeze. the printer would turn on but immediately freeze after a few seconds. the main chip would feel super hot to touch right after starting up, which told me the thermistor caused a short.
> 
> there are no protections or safe guard against shorts in the thermistor or hotend. so be very careful not to touch the exposed wires when it's powered on.
> 
> that's why it's recommended to swap to the cartridge style thermistor because the glass ones are too cheap and unreliable, even for the Bed (which uses the same glass one).

there are no problems either in the thermistor or in the heater, I disassembled and checked in advance, before installing your firmware, the printer worked with the factory firmware from aquila, more than one kilogram of plastic was printed and everything was fine, I installed your firmware - a problem with suddenly jumping temperature, again I installed the factory firmware, everything works fine, I have already printed half a kilogram, not a single mistake ..

classicrocker883, 2023-08-09:
I'll take note of the issue, there is a core-rewrite in the works right now to make it better. I'll do my best seeing what code is related, Hopefully the next release will solve the issue.

for now I may be able to have a JyersUI working with the HC32 rather than the ProUI.

it is more stable and shouldn't have any issues.