classicrocker883/MRiscoCProUI #63: linear advance on aquila x2[FR]

PolcovnicGauss, 2023-08-04:
### Is your feature request related to a problem? Please describe.

The rest of the firmware does not work properly, I need a simple but stable firmware without bells and whistles

### Are you looking for hardware support?

Aquila x2 main board HC32

### Describe the feature you want

Can you compile the most simple and stable firmware, with a simple interface for aquila x2, so that it has linear advance?

classicrocker883, 2023-08-09:
yes as I mentioned in the other post I am working to getting the JyersUI to work for the HC32. it's the NoPro essential but known to be stable.

not to mention it will be to the most upto date Marlin version. look forward to seeing that sometime this week.

classicrocker883, 2023-08-09:
I want to say that Linear Advance **should be enabled ** for the HC32.

I think however there was a slight mistake with the option not showing up. However you can still set the value using the M900 gcode.

PolcovnicGauss, 2024-01-13:
there is a problem with linear advance again, it seems to work, but when I try to print the calibration code from here https://marlinfw.org/tools/lin_advance/k-factor.html, every time the K-factor increases, the printing speed drops, This can be seen even visually, what could be the problem?