classicrocker883/MRiscoCProUI #65: [BUG]  When printing menu 'Hotend temp graph' not closing when press button 'Continue'

Weresk, 2023-08-09:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When press button 'Continue' it should close window and it does not.

1. Start print something
2. Press 'Tune' menu
3. Choose 'Hotend temp graph'
4. Press 'Continue'

### Printer Model

None

### Your Mainboard

Aquila GD32

### ProUI?

NoPro

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Tried with ProUi and No ProUi - same result.

classicrocker883, 2023-08-09:
hmm. I'll get back to you I may have a fix before the week ends

classicrocker883, 2023-08-11:
I was able to make it work how it should. I can have another release soon with other planned updates

Weresk, 2023-09-22:
Again repeats this error on last september build.

classicrocker883, 2023-10-25:
does this still happen? so far I have not had an issue, but if you can let me know how it might occur, like from a filament runout, or M600 pause, ect..

classicrocker883, 2023-12-06:
hello, I am wondering if this issue or is there anything else I can help with?

classicrocker883, 2024-01-19:
I have since fixed the issue. when printing, if this graph popup is open and say another popup occurs, like homing or pause, this will then return the menu.

closed 2024-01-19 by classicrocker883