classicrocker883/MRiscoCProUI #66: [BUG] Prints getting cancelled again

Arson31, 2023-08-11:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Old gcodes prior to flashing new firmware seems to work with alex's firmware or MRiscoC's. After flashing this firmware any gcodes sliced from Cura 5.4.0. will only home and will cancel print. I'm not sure if the start gcode in Cura is causing the problem but it can print older gcode just fine.
[gcodes.zip](https://github.com/classicrocker883/MRiscoCProUI/files/12325140/gcodes.zip)

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

Octoprint, Cura

### ProUI?

ProUI

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Using the latest Aquila_GD32_MM-ProUI.bin

classicrocker883, 2023-08-12:
this sounds almost like this [issue here](https://github.com/MarlinFirmware/Marlin/issues/26144)  

which has led to this [pull request here](https://github.com/MarlinFirmware/Marlin/pull/26158)

I'll be posting an updated release soon, and this should be fixed in it.

Arson31, 2023-08-12:
@classicrocker883 [This workaround](https://github.com/classicrocker883/MRiscoCProUI/issues/41#issuecomment-1600065609) resolves the issue without having to redo the mesh again.
> 
>     1. Build mesh
> 
>     2. Save Mesh
> 
>     3. Power off printer
> 
>     4. Power on printer
> 
>     5. Go to Control/Advanced Settings/Load Settings
> 
>     6. Then go to Mesh settings and view mesh
>

classicrocker883, 2023-08-12:
for some reason the mesh saves do not load at startup.

Arson31, 2023-08-13:
For now I simply add M501 after homing in my start gcode in my printer settings from the same [post](https://github.com/classicrocker883/MRiscoCProUI/issues/41#issuecomment-1600065609) without having to go through the list of steps:
>     G28 ;Home
>     M501; Load settings
>     M420 S1 ;Pull your mesh bed level:
>     G1 Z15.0 F6000 ;Move the platform down 15mm
>     ;Prime the extruder
>     G92 E0
>     G1 F200 E3

KimmoHop, 2023-12-09:
I added this to Configuration_adv.h to specifically loading settings (again) and to prevent losing mesh if I change some other setting. It's a cludge, but I didn't find any reason in source code, why mesh is not initially loaded :/

`#define STARTUP_COMMANDS "M501"`