classicrocker883/MRiscoCProUI #68: [BUG] Unintended release updates?

ThomDietrich, 2023-08-23:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Hey, not a bug with the firmware. Thanks for your work!

I've subscribed to releases in your repository. Good idea in theory but since then I get four release notifications per day. Looks like you regularly replace releases by the daily build!? Is that intended? To me this feels like more like a bug in your CICD toolchain.

Thanks for checking!

### Printer Model

Creality Ender-3

### Model Type

V2

### Your Mainboard

Creality v4.2.7/4.2.2 F103RET6

### ProUI?

NoPro

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-09-11:
Ohh shoot I had no idea that was a "thing" that you could do that. and I apologize for the late reply, been super busy.

so for the releases, yes those are Unintended to be the latest release. I should add some kind of marker or note saying "TEST" or something like that. but if the release has no description box, simply ignore it. also there may not be any assets either.

you can also look at the README and at the very top you will see something like `Aquila Build Configs :: passing`

if it says `failing` then chances are those releases arent the ones meant to be downloaded.

ThomDietrich, 2023-09-12:
All good, thanks.
The issues stopped some time after my message. You probably fixed it without knowing.
Cheers!

closed 2023-09-12 by ThomDietrich