classicrocker883/MRiscoCProUI #75: [BUG] - Freeze after choosing to load more filament

TheKarls, 2023-09-26:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When i start to load filament it load just a bit (I think it's calibrated on a direct extruder) but i I choose "continue" to extrude more filament the extruder start to extrude, stop a bit after and then the ui freeze and you can only turn off the printer.

In a normal situazion you just restart, push the filament to the hotend by hand, run again the load filament function and then choose continue but if you fail to load when you change filament color in the middle of a print time you waste all the process

### Printer Model

Voxelab Aquila

### Model Type

x2

### Your Mainboard

Aquila HC32

### ProUI?

ProUI

### Bed Leveling

Default - No Probe / No Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Austinj78, 2023-09-26:
I've experienced this, too.    I've got a Monoprice Joule printer, which, as I understand it, is a (barely) rebranded Aquila.  Aquila Main Board 1.0.3, with the HC32 chip.  I'm running the HC32-MM-JyersUI bin that was released 3 weeks ago.

If you need more details about the printer itself, I'm happy to dig into that again; I thought I remember the Joule being an original Aquila, not an Aquila X2, but I recently started questioning if I remembered that right...  I AM sure about the main board markings and the chip, though.  

Also happy to test whatever would be helpful!

classicrocker883, 2023-09-26:
the JyersUI is a work in progress. I wasn't sure which is better for the HC32... ProUI or Jyers.

I would rather the ProUI because it has more features, the main issue was it would freeze with saving/ store settings however it did save and hopefully we could one day fix that issue. for now I'll try to get either one working without issue, but it seems now the ProUI could be the way to go.

either way. I will reupload new releases. I would try the ProUI first, either way keep me posted with what works or what doesnt.

classicrocker883, 2023-09-26:
UPDATE
---

you can try the new released versions here 
[for ProUI](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-HC32-1)

[for JyersUI](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-HC32b)

TheKarls, 2023-09-28:
Thank You! I'll try it  in like a week because in the meantime I also broke he printer :P I need to waith for the parts

TheKarls, 2023-10-23:
> ## UPDATE
> you can try the new released versions here [for ProUI](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-HC32-1)
> 
> [for JyersUI](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-HC32b)

Hi! I want to try the latest version of the firmware but the one released 4 days ago doesn't have the X2 version. Which I have to download?

classicrocker883, 2023-10-26:
whoops that was supposed to be with the HC32, I just re-released it so it should be fine to download.

however, with the latest build there is an important note to make. one of the issues with it saving is fix, but another issue came up.

the printer will restart over when pressing the knob in, this can be fixed by removing the SD card.

however since you may need the SD card to print, you should be able to using a USB - like through Octoprint or a slicer or Pronterface.

so again, once you flash the firmware, after it loads you will see that pressing enter with the knob will cause a restart. so just remove the SD card and you should be fine.

if you don't like this and wish to print using the SD card, then by all means try the earlier previous versions.

a fix is in the works and hopefully all the issues will be in the past.

if anyone does use the new version, I'd like to know if at all any issues remain or anything new pops up.

TheKarls, 2023-10-30:
Yeah I'll prefer to use the sdcard, could you please link me the previous release? just to be sure

classicrocker883, 2023-10-31:
that version is here:  
[2.1.3e-HC32-2](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-HC32-2)

<br>

otherwise you can try the one before that which is  
[2.1.3e-HC32-1](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-HC32-1)

classicrocker883, 2024-01-19:
I have uploaded a new build that includes the different UI's, using a Vanilla base Marlin. so any issues please post in the Discussion of that release, and also if you want at the Issues tab of repo:MarlinFirmware/Marlin

There is JyersUI, CrealityUI, and a NoPro version of ProUi

closed 2024-05-03 by github-actions[bot]