classicrocker883/MRiscoCProUI #77: [BUG] LCD Backlight Timeout and Encoder Tick menu interfere with each other

Nazar78, 2023-10-02:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

I noticed this behaviour where when I enabled the tick/beep from the advanced settings menu, return to it again and noticed the LCD timeout value (max 99) now showing out of range values 266 sometimes 267. And if I set the LCD timeout value on the menu, now the tick got unchecked and of course the beep disabled. Strangely the LCD still turns off when it's due.

I had some timer/pwm code + custom config changes ported over from the old Alex version including the ones from PlatformIO and thought that could be the cause. It's been awhile since I messed with the firmware. So I tested with the latest release Aquila_N32_BLT-ProUI.bin and that too gave the same outcome.

After few testing I figured out uncommenting TICK_ON_DEFAULT finally fixed it. The tick/beep can also be disabled after that and no further issues with the menus (except the known thumbnail preview issue, will try to look at it). Unfortunately I can't simply open a PR for this, probably will close this bug report too, because I'm unsure where is the relation between these two that caused the issue.

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila N32

### Add-ons that could be involved

BL-Touch

### ProUI?

ProUI

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Nazar78, 2023-10-10:
The only fix here is to uncomment the `TICK_ON_DEFAULT` in the `Configuration_adv.h`, turn off in the menu if you prefer no beeps.

Managed to compile the latest working release of [Ender-3 V2/S1 Professional Firmware 20230904](https://github.com/mriscoc/Ender3V2S1/releases/tag/20230904) with this variable still commented out. So the fault lies in the outdated private libs which the owner has its support dropped for maple builds.

More discussion and help needed to continue supporting @classicrocker883 fork. Closing this for now.

[image]

classicrocker883, 2023-10-11:
I'd like to reopen the issue if it still persists.  

If I understand correctly...  
so when `TICK_ON_DEFAULT` is enabled (uncommented) this is supposed to have the lcd Beep when the knob is pressed.

but when it is disabled (commented out) it leaves the lcd Beep off.

that said when it is enabled then the issue is fixed?

`TICK_ON_DEFAULT` is not available with *Ender-3 V2/S1 Professional Firmware 20230904*. that is something I made and is only in this Fork. it is meant to act like `SOUND_ON_DEFAULT`.

I think now it has been fixed so turning sound off doesnt effect the Beep. before the tick/beep was included in the sound being on, now it should be completely separate.

this seems like it should be a quick fix anyway. I'll take a look.

Nazar78, 2023-10-12:
> If I understand correctly... so when `TICK_ON_DEFAULT` is enabled (uncommented) this is supposed to have the lcd Beep when the knob is pressed.
> 
> but when it is disabled (commented out) it leaves the lcd Beep off.
> 
> that said when it is enabled then the issue is fixed?

The issue I noticed was not with the actual audible tick/beep but the outcome of the displayed menu. 
1. Checked `Encoder Tick`, it's now audible, go back then return to the page again, now  `LCD Backlight Timeout` shows 266 or 267.
2. Now reset `LCD Backlight Timeout` to i.e. 10, go back then return to the page again, now `Encoder Tick` got unchecked and no longer audible.

This goes on infinitely.

And yes once I uncommented the `TICK_ON_DEFAULT`, the LCD Backlight Timeout and the Encoder Tick menu works as they should.

> `TICK_ON_DEFAULT` is not available with _Ender-3 V2/S1 Professional Firmware 20230904_. that is something I made and is only in this Fork. it is meant to act like `SOUND_ON_DEFAULT`.

> I think now it has been fixed so turning sound off doesnt effect the Beep. before the tick/beep was included in the sound being on, now it should be completely separate.

Thanks, think I got confused comparing several configs lol. I wasn't aware of this as all along I thought the beep and tick was referred to the same thing. I opened this issue not because of the physical sound/beep/tick but instead the weird menu display outcome when using between these two options.

Anyway yes it's a quick fix, for me at least by uncommenting the `TICK_ON_DEFAULT`, not sure if you have a better way, coding wise, to fix it.

classicrocker883, 2023-10-12:
I managed to fix this issue. I'll post the commit, as well as a revision to the firmware files. basically I changed the menu to have it as an Unsigned Char, vs a Signed Short Int. as well as changed it to be more like setting and applying the LCD brightness (code wise) because it is more like it.

closed 2023-10-12 by classicrocker883