classicrocker883/MRiscoCProUI #81: [BUG] TJC display screen artifacts

vw72, 2023-10-16:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

I replace my Voxelab Display with an Ender 3v2 display that is the TJC model. The following artifacts exists

When autohomeing the printer, whether through the menu or through the gcode, it appears that Chinese characters are displayed on the screen.  The homing works correctly.

When loading gcode with a preview file embeded, the preview on the display looks like a bunch of lines.  I am using 230x180 for the image.

While it appears that the mesh is being used as the z value changes as the print head moves, the letter "Z" no longer flashes as it did with earlier versions.

These artifacts are present regardless of the slicer (Cura, PrusaSlicer, OrcaSlicer).

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

TJC display 

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

vw72, 2023-10-16:
This is with the October firmware release.

classicrocker883, 2023-10-20:
I do believe in the head MRiscoC fork repo something like this was mentioned, but you can also try using 200x200 for gcode preview, because that is what they were set at. but for TJC displays 180x180 might be what they are supposed to be. I will look into this and link the info to it.

If you are able to 

>When autohomeing the printer, whether through the menu or through the gcode, it appears that Chinese characters are displayed on the screen. The homing works correctly.

this is something I may not be able to figure out right away.

>While it appears that the mesh is being used as the z value changes as the print head moves, the letter "Z" no longer flashes as it did with earlier versions.

this should be working for all other displays, I'll have to look into this as well. I mean technically if everything is working normally on the DWIN display, it should for TJC. I have to ask then, for the runout sensor (even if you dont have one installed) does that flash as well? you go into menu Prepare > Filament Management > Filament Settings 
> Runout Enable. and also if it doesnt flash, change Runout Active to **HIGH** or **LOW**

then you should also see one of the lower middle icons flashing.

classicrocker883, 2023-10-20:
I almost forgot, that for TJC displays, it may require a certain parameter to be defined in the source code.

but to fix your issues, this should do it **https://github.com/mriscoc/Special_Configurations/releases/tag/tjc**

this isnt my configurations or anything, but you can use this in the time being.

vw72, 2023-10-20:
Making the requested changes to the Runout sensor (setting it active and to
HIGH and also setting it to LOW), did not cause any flashing.  Before
working further on this, let me install the BLT version to make sure that
the issue isn't caused by the UBL version not installing properly on the
GD32 board. I'll report back.

vw72, 2023-10-20:
I couldn't try these because there wasn't a GD32 version.

vw72, 2023-10-20:
I installed the BLT PROUI version from here
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-2-2-test

And while it fixed the scroll direction issue, it did not fix the blinking
issue regardless of how the runout sensor was set.  In addition, it doesn't
retain the BLT mesh after shutting down the printer (so the mesh has to be
built each time the printer is used).

vw72, 2023-10-20:
Using Aquila-TJC_GD32_BLT-ProUI.bin (2.1.3e-2-2-test)

Does fix the scrolling problem. But does not fix the display problems.
However, even though the Z doesn't flash when printing, the z height
displayed at the bottom of the screen does indeed change depending on the
mesh at the point the printhead is printing. So the leveling does work.
I've attached a screenshot.

The chinese characters are still displayed when homing, whether through the
menus or in gcode.

classicrocker883, 2023-10-26:
sorry I do not see an attached screenshot.

there may not be anything I can do for why random characters show when homing.

actually, it may be because I since replaced the image with something else. but that was with the DWIN_SET, which I'm not sure how to update for TJC_SET. I suppose what I can do is change it so when TJC is enabled it will go back to the stock image as before.

as for why the Z isnt blinking im not sure. but does other things blink? like each axis coordinate when its not homed

X ` - ? - `  Y  ` - ? - `  Z  ` - ? - `
X `     `  Y  `     `  Z  `     `

classicrocker883, 2023-10-26:
okay I see thanks for the update. if I am able to update the TJC_SET like I did with the DWIN_SET I will and it all should work as normal. for now I should be able to fix the homing image thing, and i'll try to do something about the Z icon blink situation.

classicrocker883, 2023-10-26:
[image]
[image]

at the very least, the icons should be changing like so. can you see the difference?

otherwise it must have to do with whatever icons are in the TJC_SET. you did update the LCD and flash it using `tjc_update.zip` in the **display assets** folder?

vw72, 2023-10-26:
The image is too small for me to see. :(I did flash the TJC_SET.  I wonder if I pull the icons file from the DWIN_SET and replace the icons in the TJC_SET if that will resolve it?  Regardless all the TJC_SET has is the icon file. JoeSent from my iPhone

classicrocker883, 2023-10-27:
they are different. TJC has just one file in TJC_SET - *tjc.tft*

DWIN_SET's icon file is *9.ico*, and comparatively DACAI's (dacai_update.zip) icon file is *9.zico* which I would somehow like to update those icon files but sadly cant find information how.  for DWIN its very easy, there is a tool which allows you to split .*ico* files and extract .*jpg* images from it. conversely, allows you to combine .*jpg* into .*ico*.

so ill have to see how the whole tft files work for TJC. otherwise look out for a new revision to the firmware build. ill post any other fixes pertaining to this LCD type.

classicrocker883, 2023-10-27:
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-2-2

heres a new release, not much I can do for the Z offset flashing, but the homing character artifacts should be fixed

vw72, 2023-10-27:
This fixed the homing character problem!

Thanks,

Joe

classicrocker883, 2023-10-30:
https://github.com/classicrocker883/MRiscoCProUI/wiki/3D-BLTouch

this is how the blinking *should* look like. maybe there is something about the TJC display that makes the "Select" box not show well. you know when you say are in the Move menu and select an Axis to move, there should be a lighter color background over the values. you can also change the color to maybe a brighter color/shade to show better in the color settings.

vw72, 2023-12-23:
I figured out what was causing the chinese characters!  Using the December noproui version, it displays the characters, but the proui version does not.  I believe this is related to the icons.  In the noproui version screens with the chinese characters don't show the icon while in the proui, it does. I think the code in the nonproui is trying to display a non-existent icon.  This is with the TJC display, so I don't know if it is applicable to other displays.

classicrocker883, 2023-12-28:
its possible that `TJC_DISPLAY` is enabled/disabled using the -NoPro or -ProUI. ill have to take a look through to see if its within the bounds of the `PROUI_EX` extra features. another thing is I have since updated the ICON_'s, and some may not actually exist in the LCD's firmware. I do know there is an updated firmware version for TJC displays. another thing is `TJC_DISPLAY` code could be enabled within the `PROUI_EX` library, which unfortunately I cannot change anything about.

this makes me wonder if the current MRiscoC version is better in that regard, better for TJC displays. I don't use that because it has a different `PROUI_EX` library that's not compatible with Maple chips. I mean I *could* use it with the -NoPro, and possibly with the -ProUI but some options may need to be disabled, and some things may not work correctly.

classicrocker883, 2024-01-26:
@vw72 hey just recently Mriscoc put out a fix for the `TJC_DISPLAY` regarding LCD brightness. can you confirm that setting the LCD brightness value in Control - Advanced settings doesn't work properly? like if you change the value between 1-100, 100 is max, and if you go for 101-200 it resets/loops.

because I have a fix in an update coming up soon.

vw72, 2024-01-28:
Was the TJC_Display fix in the test firmware you had me run for the fix for
when cancelling the bed mesh?  If so, the LCD brightness can scroll up to
40 and works as expected. It does alter the brightness very well upto that
point but doesn't scroll beyond.

classicrocker883, 2024-01-29:
> Was the TJC_Display fix in the test firmware you had me run for the fix for when cancelling the bed mesh? If so, the LCD brightness can scroll up to 40 and works as expected. It does alter the brightness very well upto that point but doesn't scroll beyond.

yes that was the intention. Mriscoc put out a fix on his repo doing that. I saw the dialog between another user and this was a good fix. 

I mean  I suppose I can lower pass 6, but they experienced the same kind of odd behavior when it was <255. and also <100.

well I'm glad we were able to bang out these issues