classicrocker883/MRiscoCProUI #82: [BUG] Display scrolls backward

vw72, 2023-10-16:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

With the October GD32 PROUI UBL firmware, it appears the control knob turns in the opposite direction than expected.  This could be an artifact of using a Creality Display in lieu of the default Voxelab Display.

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

Creality Display

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-10-20:
I will suggest using [this build for Ender-3](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-2-ender3-1)

however, I just noticed you are using the PROUI UBL, which means you should be able to use all the other GD32 versions, except for the UBL because that just will not fit the 256k chip of the Aquila. in otherwords, the Aquila cannot flash files larger than ~228k (because it is 256k - bootloader = 228k).

so, for now you can use the BLT version for Ender-3, or continue to use the reverse encoder until I am able to make a **.bin** file for you. bare with me, ill post it up soon.

classicrocker883, 2023-10-20:
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3e-2-2-test

you can test this and seee

vw72, 2023-10-20:
This fixes the scrolling backwards issue.

closed 2023-12-06 by classicrocker883