classicrocker883/MRiscoCProUI #88: Settings not being saved[BUG]

bigmick110, 2023-12-01:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Had no issue installing the firmware and used the Voxelab Red DWIN set for the screen. Everything went well inserting all my settings after adjusting Tram and Z offset and saving everything as I went. I initially had the Probe offset setting set as a +positive digit instead of the correct -negative digit which threw me off for a bit. My first mesh was all green lights so i was really happy with that....Everything working well and started printing the usual test prints which all came out surprisingly great first try. By this time it was getting late so I shut everything down for the night feeling very positive and quite happy with myself. Upon starting everything up the next morning I was ready to print something more substantial so I uploaded an Articulated Dragon print, did all my usual pre-print bed clean, homing, pre-heat and hit start. Straight away I noticed issues so stopped the print and started to investigate the problem.....Found that most, if not all, my settings I had saved the night before had been lost and returned to, I presume, the stock settings. Lucky enough I had written everything down so started reinstalling all my settings. Then I cam across the issue of when I set the Z-offset and saved this setting every other setting I had put in was lost again. I then tried installing them in a different order....Z-Offset first and when I changed the mesh points from 7x7 to 4x4 my z-offset would revert back to zero. I tried reinstalling the stock voxelab firmware and DWIN set then reinstalling your firmware again and it was exactly the same. I even tried using Jyers firmware but for some reason half of the needed settings were missing from the menu (Specifically the custom bed sizing which I needed because of the Sprite). I was using a Netac 8GB Micro Sd (Speed Class 10) Card so I figured I'd change that and see what happens......This new 128GB Ultra High speed Micro Sd card has just arrived so that's where I'm up too. I'm about to reinstall everything again using the new Micro sd card.

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila HC32

### Add-ons that could be involved

Creality Sprite Extruder/Hotend w/CR Touch, Creality Duel Z-Axis kit

### ProUI?

ProUI

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-12-06:
This is a known issue that is hopefully going to be fix very soon.

The JyersUI does not include many of the features you find in the ProUI. 

Do you find the settings do save in Jyers version?

Also, what is the date of release for the ProUI which you used? because there is a more recent release which settings _do_ save, however there is an issue using the SD card itself, but you can still use the printer and print using the USB hooked up to a computer with a slicer/Pronterface or Octoprint.

bigmick110, 2023-12-24:
Yeh....There was another version update I missed. It was only a few days old and it worked a treat. Most issues solved....just a few minor bugs every now and then but I'm sure your on top of it !! Many thanks though !! Tried the link to the new H32 update and it just took me to the good old "404" not here anymore page so I guess it's not quite ready or something ?

classicrocker883, 2023-12-26:
> Yeh....There was another version update I missed. It was only a few days old and it worked a treat. Most issues solved....just a few minor bugs every now and then but I'm sure your on top of it !! [...]

I've been working on getting the release up to date. it should be working now

closed 2024-04-03 by github-actions[bot]