classicrocker883/MRiscoCProUI #90: Print finishes after heat up and doesn't print anything

Claresh07, 2023-12-07:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Not sure if it's a firmware problem or not. When I stat a print the bed and hotend heat, moves to home then print goes to 100% and cool down starts. I also tried alex's firmware to same result.

### Printer Model

Voxelab Aquila

### Model Type

S3

### Your Mainboard

Aquila N32

### ProUI?

ProUI

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-12-20:
this shouldn't happen, even with Alex's. I think the issue is the Slicer.

there may be something how you sliced, or with the slicer program, or even the SD card.

I would try using a known good slicer, with a known good model to print, and a known good SD.

it could also be your Start Gcode.

Claresh07, 2023-12-20:
Hey, thanks for getting back to me.
 I started having the problem with the stock firmware too, I did a full format of the sd card, and it was working again. 
In summary, it was probably a faulty sd card (and your firmware is more sensitive than the voxelab)

closed 2023-12-26 by classicrocker883