classicrocker883/MRiscoCProUI #92: [BUG]  Cancelling build mesh requires restart before being able to rebuild another mesh

vw72, 2023-12-23:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Install December firmware - preheat printer and auto-build mesh.  Press cancel during the mesh building.  Subsequent attempts to auto-build mesh start but stop immediately with mesh cancelled message.  Reboot printer and mesh building works as expected until cancelled, then you need to restart again to build a mesh.

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

BL Touch

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm using a Creality display with the TJC chipset on the Voxelab, but I don't think it is related to this issue.

classicrocker883, 2023-12-26:
I'll take a look into this.

classicrocker883, 2024-01-19:
do you know if this had been fixed with the most recent release? if not I can make a test file with a change which might help

vw72, 2024-01-20:
I’m not at the printer tonight but will test it in the morning and report back. Sent from my iPhone

vw72, 2024-01-20:
Actually, I'm in the middle of a big printjob. As soon as it is done I'll
install the latest version and see if it still is a problem.  (Version I
reported it on was 20231221)

vw72, 2024-01-21:
The problem with cancelling building the mesh keeping you from being able to go back and build a mesh persists on the January release. Once you cancel the building if the mesh, you need to restart the printer to be able to build the mesh. JoePs. On TJC still can’t preview gcode regardless of which slicer created the gcode and image, including Cura, Prusaslicer and Orcaslicer. Sent from my iPhone

classicrocker883, 2024-01-21:
if im not mistaken, some TJC displays may not work with gcode preview. I'm actually not sure theres not much info I can find about it [Reddit post about this](https://www.reddit.com/r/Ender3v2Firmware/comments/18eybew/help_with_mriscoc_ender3v2_firmware/)

I think for TJC you may need to use 180x180 , try that in the slicer.

classicrocker883, 2024-01-21:
are you able to compile your own?  
in **src/lcd/e3v2/proui/dwin.cpp**

look for and add this to the function:
```diff
  void DWIN_LevelingDone() {
    DEBUG_ECHOLNPGM("DWIN_LevelingDone");
    #if HAS_MESH
      #if ALL(PROUI_EX, HAS_BED_PROBE)
        ProEx.LevelingDone();
      #else
        Goto_MeshViewer(true);
      #endif
    #endif
+   HMI_ReturnScreen();
  }
```

vw72, 2024-01-21:
I’ve tried 180x180 and 160x160 for both jpg and png.  It used to work prior to the December firmware. It gives the message 200x200 preview not found or something like that regardless of what size I choose. I think previously it just said no preview found. I’m going to try reverting back to the the October or November firmware and see if I can get it still works there. It’s not show stopper just a nice feature to have. JoeSent from my iPhone

vw72, 2024-01-22:
Currently I’m not in a position to compile it. My work around is just not to cancel the mesh or if I do just restart the printer and reheat the bed. It’s a minor inconvenience. Sent from my iPhone

classicrocker883, 2024-01-26:
Hi I am very close to this fix. So far I have been able to cancel normally, and re-level without restarting.  
One thing I'm hung up on is when you press the button to cancel. it seems as though the button must be pressed at the right moment (like when its "not busy").

its a bit tricky to figure out since the code in question resides within the `proui` library - which are compiled files you cant just open and change, actually you cant even read them because its protected in a way.

so soon, I'm so close. also I mentioned in another comment asking about the TJC display, if there is an issue setting the value that isnt between 0-100. or 101-200. setting brightness should be linear. and I should have a fix next release.

and if you had any other recommendations what I can do that would help.

classicrocker883, 2024-01-26:
> The preview works with TJC on the 20231027 version of the firmware

oh no... I think this is when we had fixed the issue regarding the preview - anytime you would click it more than a few times it would freeze and restart. what I can do is when TJC is enabled, I'll have it use that code set which worked for that.

vw72, 2024-01-26:
I can’t get to it today (Friday) because I’m in the middle of a big print.
I should be able to text it tomorrow (Saturday). I’ll report back
afterwards. I’d it located in the regular GitHub area?

Joe

vw72, 2024-01-26:
That would be great but if I’m the only one noticing the problem, it might
be easier on your end to just say the preview doesn’t work on TJC. It’s
nice to have but by no means a show stopper.

classicrocker883, 2024-01-26:
I have something you can test, its TJC-GD32_UBL-ProUI.bin
it also has some code I'm testing canceling the build mesh.

let me know if the gcode preview works.  
[Aquila-TJC-GD32_UBL-ProUI.zip](https://github.com/classicrocker883/MRiscoCProUI/files/14066927/Aquila-TJC-GD32_UBL-ProUI.zip)

vw72, 2024-01-28:
This did not fix the gcode preview but it did fix the cancelling the build
mesh. Also, unlike the other TJC versions, the control knob is scrolling
backwards on this one.

With the exception of the preview, from what I can tell, the difference
between the TJC and the DWIN, from a user perspective is the direction the
control knob scrolls.  Is that something that can be placed in a setting
under advanced to keep you from having to compile two versions?

Joe

classicrocker883, 2024-01-29:
oh that's right I forgot about it being TJC, Aquila displays have it reverse and I forgot to comment that line out.

well that sucks about the gcode preview... I had it reverted back since before the change, like u said was back in October. maybe I missed something.

When u mention when it worked, were u using 200x200 thumb size in slicer? have you tried Both Cura and Prusaslicer?

did I mention the slicer scripts have been updated? perhaps if you updated your slicer, maybe the scripts were lost/deleted. or maybe the new scripts aren't working properly for the TJC.

anyway I should have another release build any day. 
I think I improved the mesh canceling a tad since.

classicrocker883, 2024-02-10:
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-2

here is the new release. ill make a list of changes later, let me know if anything needs adjusting

closed 2024-02-10 by classicrocker883