classicrocker883/MRiscoCProUI #93: [BUG]  Mesh Fade setting

vw72, 2023-12-24:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

December Rev 1 release. It appears that for the ProUI UBL version, the Mesh Fade now defaults to 2, whereas it used to be 10 (I think).  Not really a bug, but most recommendations I find online are to set it to 10 and then to decrease it by 2 until you find the sweet spot for your printer and settings.  Not really a bug, but a change.

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-12-26:
thanks for pointing this out, I think this is an actually bug with Marlin. because the default fade height should be 10 as it always was.

```y
#define DEFAULT_LEVELING_FADE_HEIGHT 10.0 // (mm) Default fade height.
```

looking through the code, since its been updated it appears there are some inconsistencies in the settings.

vw72, 2023-12-26:
Another oddity is if you manually tram the bed using the probe (but not the tram wizard itself) and then build a mesh, the four corner values are quite a bit different between the two methods. Not sure what’s going on with that. JoeSent from my iPhone

classicrocker883, 2023-12-26:
the corner values are different when building a mesh because those are different points. the 4 corners are adjustable by changing the bed screw inset. just like the mesh by the grid (5x5, 3x3...) so all the points arent going to line up.

I just checked the Mesh Fade value, are you sure youre not confused with the Tilt? the Tile value is 2. the Mesh Fade is default 10.0. which is a float not an integer.

vw72, 2023-12-26:
It’s the value above the tiltAs for the corner differences between tramming and the mesh, sometimes the values are significantly different, like as much as +-.100. JoeSent from my iPhone

classicrocker883, 2023-12-27:
next time you are at your printer, take note of the X Y position for the corners in the tramming, and the X Y position for the same point of the mesh. compare the two to see if they are indeed the same. also, in the tramming menu, measure the Z value using the probe for any point. do it multiple times, you should find the value is probably not exactly the same each time. it should be within a certain tolerance which is measured using the M48 probe test. if you havent done one, try and see if that is within correct tolerance.

if the points are close, but not the same value, its not uncommon for it to be +/-0.100. thats generally max acceptable difference. meaning its not uncommon to have *some* warping of the bed, and +/-0.100 between points should be fine.

why there's discrepancy between the same point between different measurements can come down to several factors; the Z screw rod - it may have some wobble or could use lube, if you are using dual Z rods or 1 Z rod causing sag/misalignment, the XYZ steps/mm need calibration.

so try the M48 test and check your tolerance, and see if the points in question are indeed the same in the X Y plane. and if possible, through terminal you can do a probe test to a specific point, "G29 P4 [R points] [X position] [Y position]" I think its just "R1" for 1 point. check the g-code G29 how to do so https://marlinfw.org/docs/gcode/G029-ubl.html

to make it easy, there is another way. you can do it as follows, G42 "I#" is the Column and "J#" is the Row, then G30 - does a probe:
> G42 I3 J3 ; Move to Back Right for 3x3, or Center for 7x7
> G42 I7 J7 ; Move to Back Right for 7x7
> G30 ; probes point

closed 2024-04-03 by github-actions[bot]