classicrocker883/MRiscoCProUI #95: [BUG] H32 Temperature won't stablize, even after PID tuning.

smhanov, 2023-12-25:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Printer: Voxelab Aquila X2 H32 with TJC display
Build: H32 December: Change ID ce4f406e28462598da43140e5a545cc46d6653fe

When running with official Voxel lab firmware, after PID tuning, temperature fluctuates between 199.7 to 200.3 degrees.
With MRisoC firmware, after PID tuning, temperature fluctuates between 196 to 206 degrees and does not stabilize, so print will not start.

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila HC32

### Add-ons that could be involved

BLT

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

smhanov, 2023-12-25:
[image]
PID tune and heat to 200 degrees using H32-December firmware.

classicrocker883, 2023-12-26:
this looks more like a PID calibration issue. I mean after PID tuning, temperatures should stabilize and the graph should remain flat.

there must be some kind of variable causing this kind of fluctuation. when the printer is heating for a print, is the part cooling fan on? it could be that, having to do with your start G-code.

otherwise, what you can do is change your PID settings to a different/default setting, then **re-run** the _PID tune_
```r
Kp  23.00
Ki  2.00
Kd  70.00
```

classicrocker883, 2024-01-02:
it is possible that the H32 board is different regarding heating. for instance official Voxelab firmware may be tuned differently in PID settings.

have you tried Alex's H32 firmware??

anyway in **Configuration.h** the PID settings can be changed:
```y
PID_MAX  255      // Limit hotend current while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
PID_K1     0.95   // Smoothing factor within any PID loop
```

a lower `PID_MAX` could give a smoother result, same for adjusting the `PID_K1`.

are you able to compile your own firmware? how about try changing these values in the code and see if that helps.

or what you can do is reseat your thermistor or even get a [new one like these copper cartridge ones](https://www.amazon.com/dp/B09WDR1M5Q/ref=sspa_dk_detail_0?pd_rd_i=B0BTYHJLPJ&pd_rd_w=uzqAH&content-id=amzn1.sym.eb7c1ac5-7c51-4df5-ba34-ca810f1f119a&pf_rd_p=eb7c1ac5-7c51-4df5-ba34-ca810f1f119a&pf_rd_r=2X1Z8XV25TGZPVWFRHWG&pd_rd_wg=d98Oi&pd_rd_r=ebd01458-2c28-472f-8513-67539f7fec93&s=industrial&sp_csd=d2lkZ2V0TmFtZT1zcF9kZXRhaWw&th=1)  
you would have to get a new heater block, but with the copper ones they read temps faster and should improve PID tuning a ton.

vica153, 2024-01-02:
I have been having a similar issue for a couple months on my G32. PID tuning works fine, but when printing, or even preheating, temperatures fluctuate enough to cause the printer to error out. Usually after 1-2 restarts, the temps will actually stabilize and it prints fine.

I don't see how the different firmware would affect this. Assuming the PID logic is same.

smhanov, 2024-01-02:
I'll try adjusting that soon and the Alex firmware. I can recompile. 

Note that the same result happens MPC tuning as well using H32-December. Is there something common to both tuning methods that would cause the temperature oscillation? I verified the wattage of the heater for MPC using a resistance meter.

Here is what I know so far:
- Voxelab firmware: Works fine
- MRiscoCProUI firmware: temperature swings after both PID and MPC tuning. Tried replacement thermistor as well with same result.

classicrocker883, 2024-01-04:
MPC and PID are two different methods for tuning, it might make more sense that what may be causing the fluctuations can be something else temperature related. its hard to say exactly what it is because I'm able to achieve steady temperatures, the only time when it wasnt steady was before a PID tune. I know one time a PID tune kept giving me bad results, so I reset the `Kp` `Ki` `Kd` values from the menu (which can also be done in the Config files precompile). you can always try Zero values and do a few cycles for tuning.

looking at your Octoprint graph again, the Line does not seem right at all. I mean besides at Target temp, it looks jagged especially at the beginning and end of the graph.

if anything it *could* be H32 board/software related rather than Marlin related. otherwise it can always be hardware related, such as the hotend - the 24v 40w heater cartridge or the heater block itself.

Ivoz, 2024-01-10:
You can try different number of runs on the PID tune. That can get it to arrive at slightly different numbers and one set might work better. You can even try manually lowering the D value by 25% continually to see if that smooths off the response

closed 2024-05-03 by github-actions[bot]