classicrocker883/MRiscoCProUI #96: [FR] hi

Kristina321, 2023-12-28:
### Is your feature request related to a problem? Please describe.

Hi you do greatest firmware, thanks a lot!!! but you delete powerloss function, in my country  power is off every day and i cant normal work with my aquila, can you help? i have N32 blt 

### Describe the feature you want

power loss function very need

classicrocker883, 2024-01-01:
sure thats no problem. there *is* powerloss function available in the -MM (Manual Mesh), or -NoPro versions.

but you are using UBL? if so i would just need to remove some of the features. if you can tell me which ones you dont need, like the Hotend/Bed Temp Graph, homeoffsets, anything else.

classicrocker883, 2024-01-02:
[N32_UBL_ProUI_PwrLss.zip](https://github.com/classicrocker883/MRiscoCProUI/files/13807753/N32_UBL_ProUI_PwrLss.zip)

hello I uploaded the firmware with PowerLoss enabled. this is ProUI UBL for N32, let me know if this is what you need.

I had to disable some things to make it fit:
`HAS_ESDIAG`          // View End-stop switch continuity  
`HAS_CUSTOM_COLORS`   // Able to change display colors

Kristina321, 2024-01-06:
Thanks a lot for you help and happy new year for you and you family =))) battery insert is very hard for me, a think i cant do that.
 i have voxelab aquila s2 with cr touch (analog of blt) and print head- sprite pro from creality. can you do this fore my printer? sorry for my bad english. happy holliday =))

Kristina321, 2024-01-06:
i dont use graph of temp and colors redactor

classicrocker883, 2024-01-07:
yes I uploaded the firmware file you can download from the previous comment
it is the N32_UBL .zip file.

or I [uploaded one here ](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-1a)

but I will let you know that I am currently working on some fixes, you can use this for now and print, but soon come back for the updated version. I'll post a comment to let you know.

Kristina321, 2024-01-12:
Ok, thanks, i will wait=))

classicrocker883, 2024-01-14:
I am having an issue with github actions compiling and uploading the versions. ive exceeded my rate limit testing. hopefully sometime today there will be a new release! 🥇

Kristina321, 2024-01-14:
Oh, it`s great!=)

Kristina321, 2024-01-15:
> I am having an issue with github actions compiling and uploading the versions. ive exceeded my rate limit testing. hopefully sometime today there will be a new release! 🥇

Hi again =) Sorry, i cant found version for my aquila s2 N32 with sprite, how name it have?

classicrocker883, 2024-01-15:
> > I am having an issue with github actions compiling and uploading the versions. ive exceeded my rate limit testing. hopefully sometime today there will be a new release! 🥇
> 
> Hi again =) Sorry, i cant found version for my aquila s2 N32 with sprite, how name it have?

Hello, there is an issue building for the release.

https://github.com/classicrocker883/MRiscoCProUI/actions/runs/7523650900/job/20477329532

as you can see `Error: API rate limit exceeded for installation ID 23315627.`

this means when you try using github Actions too much it stops any further actions, to prevent overload on the system. I'm only allow a certain number per hour.... hopefully within an hour it should be ready!

Kristina321, 2024-01-15:
Oh, good =))) I thought you forgot of me, when i see new releese and dont find firmware =)))

classicrocker883, 2024-01-15:
> Oh, good =))) I thought you forgot of me, when i see new releese and dont find firmware =)))

https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-1
it should be all ready. if you dont see it within a few minutes of me posting, just wait a couple it will be there

classicrocker883, 2024-01-19:
> Hi, i set firmware Aquila-SpriteT13_N32_UBL-ProUI-PWRLOSS and i cant found powerlost function, weare it place? and i think, if you do firmware for sprit extruder, you can set e steps 424,9 for standart setting

the power loss function is in the advanced settings from the Control menu. or you can enable it from Tune Menu when printing.

I will automatically have that setting enabled next build update, that was something I overlooked. and I hope you don't mind, but I will set E steps to 425.0 because it is a round number and I want to make it available for all users, just remember to change it for yourself.

closed 2024-01-28 by classicrocker883

classicrocker883, 2024-02-10:
hi i uploaded a new release.
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-2

I made changes for you, I hope that helps, if anything needs adjusting let me know