classicrocker883/MRiscoCProUI #97: [BUG] Can't get a UBL Mesh to complete/save

inspectionsbybob, 2024-01-02:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

I loaded up the December release... Aquila_GD32_UBL-7x7-NoPro-MPC.bin
got most of the setup things done...
Went to create a new Mesh....
Had problems in that I had to Zero the Stored Offset as the first time it missed the Bed with the probe by 1mm everytime.
Got it to touch the bed (49 times) but it hung after completing the 49th probe... nothing.
Hit the Select button a few times and got some response, but no option to save or anything else.

Can't get a Mesh to save...

Tried a Tram first... no difference...

What am I missing?

### Printer Model

Voxelab Aquila

### Model Type

Aquila G32

### Your Mainboard

Aquila GD32

### ProUI?

NoPro

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Loaded what I think is the right version.
My Aquila just has BL-Touch added and everything else is stock.

H E L P

inspectionsbybob, 2024-01-03:
Ok, I just loaded the Linear version (Aquila_GD32_BLT-5x5-NoPro-MPC.bin) 
and got the same behavior... Mesh building is broken.

I flashed the Software
Rebooted
Reset all my offsets
Tried to do an "Auto Mesh" and it did the FIRST location and then aborted with no message on the screen, but Pronto Interface showed "Probing Failed"

inspectionsbybob, 2024-01-03:
One More Comment.
I rolled back to NOVEMBER 2023 (Aquila_GD32_BLT-5x5-NoPro-IS-MPC.bin)

And that was completed and saved!  

I didn't change anything else, I just flashed some OLD code, so something broke.

classicrocker883, 2024-01-04:
thanks for bringing this to my attention. I will try to see what's going on. I know that the bilinear/BL touch (BLT) version saving the mesh doesn't exactly work as intended and thought I had that issue fixed.  

but it seems like with **NOVEMBER 2023 -BLT-NoPro-IS-MPC** that is how it should be.  
until we get to **DECEMBER 2023** then if it is **-UBL** or **-BLT**, it is the same issue?

inspectionsbybob, 2024-01-04:
I didn't do much testing beyond what I did as I had a project waiting on that printer.  The community software I had been running finally got stuck in a PID fight and would no longer stabilize.

the behavior of the later releases was inconsistent.  It seemed to vary depending on the order I did things. One issue with -all- the releases was that it would not create a mesh if the z-offset had not been set.  Once I went and put in a z value, it would finish probing, other wise is would try to probe from too high up and abort when it didn't touch the plate. 

Having not played with UBL I though its probing pattern was weird, started in a corner and worked a triangle, and then moved to another corner... sometimes it would finish and get stuck where it would normally give the "save" prompt, obviously it couldn't save as the mesh was corrupted, but the UI hung there.  

Still working may way around the UI as it is very different, not wrong, just different so I am having to unlearn the old UI. Setting the Z-offset does seem a bit  buried in the UI though. As the "create mesh" and "set Z-offset" are so related/tied together -I- think they should be closer.  Confirming the -Z- is one of the most common UI tasks on my machine (it seems to drift) so I visit that section routinely.  Having to go so deep, then back out, go deep in another menu, back out... etc... seems arduous.

One good note: MPC is like -W O W-.  Once it calibrated, all on its own, it heated up with no overshoot, no oscillation, no issues and stayed rock solid, no oscillation, during a LONG print.  That is why I am going through this upgrade.

classicrocker883, 2024-01-06:
I am working on making a ProUI version with MPC enabled. I'm thinking whatever issue for saving could be related with the ProUI extra features, because I know when saving the mesh there are two different functions it calls for. I might have to do some testing, but another thing is the mesh inset. this has an effect of where the probe can actually reach. basically if you move the X axis as far right as it goes, the probe doesnt reach the end of the bed. so theres no way for the probe to get all the points if the mesh takes up the whole bed.

so naturally for 7x7 grid, it may not reach the last row, so it can only probe 42/49 points. if you edited the Maximum X mesh inset to a smaller number - which can be limited by the Probe's X offset - then you should be able to probe all the points. the thing with that, if you wanted to print on that side (in that area it cant reach) it doesn't use the adjusted leveling values. but with a full bed mesh it fills in any unprobed points with what it *thinks* it should be and it does so pretty accurately. and thats how the firmware is by default.

as for how it probes in a weird pattern, I'd like to work on that. there is a way to make it go in order, row by row, but I haven't got around to figure out exactly. 

as for the UI layout, I tried making it like Alex's which is what most of us are used to. the original ProUI - Mriscoc's layout to me is all over the place and bunched in one place. I try to spread it out to what makes sense. setting the Z offset is a bit of a tossup. because those without using a Probe can still set their offset, so I put it in the Prepare Menu. which is where Tramming is, and the Z offset and Tramming go hand in hand.

>One issue with -all- the releases was that it would not create a mesh if the z-offset had not been set. Once I went and put in a z value, it would finish probing, other wise is would try to probe from too high up and abort when it didn't touch the plate.

I'm not quite sure what this is about. it shouldn't matter what the Z offset is for it to probe. so the Z offset is the relationship between the probe tip (where it thinks the bed 0.00 is) and the actual position of the nozzle (where the nozzle and the bed meet). so basically when Z is 0.00 the nozzle should just be touching the bed. if not, it is too high.   
I mean, before anything you should be setting the Z offset to the correct true value. then Tram the bed to level it. then do an auto bed level - probe for mesh. otherwise you won't get the right values.

im open to suggestions though, if something is out of place or could use change let me know. another user here I spoke with has helped me make some good changes that I probably wouldn't have thought of myself.

and I actually haven't given MPC a try yet. I imagine its great. i'm just not sure how to go about it. for me, yeah I do get a couple oscillations when heating with PID, but it quickly gets to the right temp with a good PID tune, and doesnt change at all during printing - like u said rock solid.
I did have an issue once setting the right PID, and what really helped me was resetting the default values. even to Zero them all out can help the autotune.

inspectionsbybob, 2024-01-06:
On all the releases I tried, I had the same issue with my "first" attempt
at mesh building. It got the center of the bed height fine using the probe
(the G28 portion) but then tried to probe from too high up and failed
either on the first probe, or on the last depending on the releasedd, with
the same un-useful message of "probe failed"

I tied tramming first... same result.

It was only AFTER I manually re-set the z-offset to anything was it happy.

One thing I wanted to try, but couldn't find in the UI was re-initilize the
EEPROM.  I was told long ago that using M502  wasn't the same thing as a
"re-initialize" and my other printers have BOTH a "load defaults" and
"re-initialize EEPROM" options.  Suspect the z-offset was corrupted when
flashing a new version.

On tramming... it doesn't work as well as other marlin implementations, or
I haven't figured out how to use it. On others I go to the first corner,
measure and it will always show ZERO.  The next corner will show me the
OFFSET from the original corner. I can make adjustment and re-measure
easily (re-measure is the default action at that point) once I get it close
enough (+/- 0.05 for me) I can move to the next corner.

Bob S.

inspectionsbybob, 2024-01-06:
I have a couple more observations from trying to dial in my Aquila today.

1) After the MPT Calibration, the printer is left in an unusual state.  I
tried printing right after the calibration and nothing happened.  It was
waiting for the Bed/Nozzle to come up to temp, and the heaters NEVER came
on.
I did a manual M500 and power cycled the printer and now it is happy.

2) Z-Offset is still weird.  I ran the Z-Ofset wizard immediately after
running the MPT tune and got one number.
I re-ran it after the power cycle and got a DIFFERENT value.  Lastly, I
still had to adjust the Z manually (Tune) from -0.80 to -1.04 but the
screen still said -0.80.  I can't save during a print so I will have to
remember to do an M500 when it is done printing the Temp tower.

3) There needs to be a "save" nearer to the Z-offset wizard.  Had to go all
the way out and back down into control to find a "save"

4) already wrote about the Tramming wizard.

Hope these comments help.

Bob Sisson

> e thread.Message ID: <classicrocker883/MRiscoCProUI/issues/97/1879579788@
> github.com>
>

inspectionsbybob, 2024-01-06:
One More... sorry... and this may be Marlin not you.

MPT looks great, but I am having issues.
Calibrated, and then ran some manual tests of extrusions watching the heater kick in early and the temperature staying rock solid.

When I tried to print a Temp tower things went bad FAST.  Everytime the bed heater kicked in (The Aquila is Bang-Bang)  there would be a spike, and the temperature would go NUTS.  I had it set to start at 230degC and work down.  It went from 230 to 218, to 240, back to 220 within moments and on the same layer.  Needless to say, the print was trashed, with the nozzle going from oozing to almost clogging within inches and on the same layer.  

Not sure what is going on, but it has been repeatable.

Any thoughts or things you want me to test?

This is the NOVEMBER 2023 release, so let me know when you have a 2024 version to try...

classicrocker883, 2024-01-09:
when you mean "MPT" do you mean "MPC"?

so there are a few things I'm trying to understand what you mean. for instance are you using something like Octoprint or Pronterface to connect through the USB and give it these commands?

for instance:
> Lastly, I
still had to adjust the Z manually (Tune) from -0.80 to -1.04 but the
screen still said -0.80.  I can't save during a print so I will have to
remember to do an M500 when it is done printing the Temp tower.

did you get this using the LCD menu?  
so when you adjust the Z-offset, it saves automatically. the screen gives a status saying "settings stored" after pressing the knob in to confirm the change.

and if you want to Reset or Load EEPROM defaults, that is in **Control > Advanced Settings**

for Tramming Wizard, that basically works kind of like the auto leveling mesh build. it takes the value (offset) of the Home value when it Home's Z. so by having different Z-Offset values or re-Homing Z will give you different results for each corner.

the goal for that is so all 4 corners are as close to 0.00 *or* the same value for **Center**.  
What I like to do is first find the Z-offset, this is done in the Z-Offset Wizard, click on Center Nozzle onto Bed, the nozzle move to 0.00 right? this is where it *thinks* the nozzle and the bed meets, so the Nozzle **should** be just touching the bed surface.

if it is not, then you adjust the Z-Offset until it is. no paper or gauge or anything in between necessary, just the bare nozzle and bed touching and as long as its not down further or moving left or right it is fine.  
- *if you find it is already touching the bed, I would raise the nozzle up with the Z-Offset, then lower back down just to make sure it isnt down to far, but just as it touches the bed.

once that is done, then in the Tramming Wizard, you can adjust the bed. you can do this using the bed probe, or manually which is exactly like you did with the center of the bed, but with each corner - instead of changing the Z-Offset (height of Z Axis), you adjust the bed knobs.
- *technically, you can adjust each corner where the printer thinks the Bed Screw is. this is in **Control > Advanced Settings** as "*Bed Screw Inset*"

Generally I find doing this manually without the probe first is better (with "*Manual Tramming*" checked), but you can always use the probe and either the Tramming Wizard (or with the "*Manual Tramming*" UNchecked).

inspectionsbybob, 2024-01-09:
Let me start -CLEAN- from the current current release and re-document any issues know that I sort-off understand the procedure differences I need to account for.

closed 2024-01-09 by inspectionsbybob

classicrocker883, 2024-01-09:
> Let me start -CLEAN- from the current current release and re-document any issues now that I sort-off understand the procedure differences I need to account for.

im in the middle of working out a couple issues trying to get MPC to work without a compiling error. so I will have to repost anything from this year/month

classicrocker883, 2024-01-09:
update
---
it appears I was able to fix the error for now. the odd thing is I have no issue compiling normally through VSCode, however once I use Github Actions to auto compile all the files, only the ones with MPC enabled it acts as if another parameter isn't also enabled.

I was able to make it work for now, my concern is the bed PID tune.

with MPC enabled let me know if there are any issues doing a PID tune for the Bed. (since MPC is only for the Hotend, the bed still uses PID).

[Here is the most recent release you can test if you like ](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-1-test3)

inspectionsbybob, 2024-01-09:
Still got a few task to complete for the Real World.

Bob Sisson

inspectionsbybob, 2024-01-09:
Ok, I downloaded what I THINK is the correct version and Fashed it.

My Machine is:
G32
BLtouch ABL
and I want MPC so I downloaded
"Aquila_GD32_BLT-5x5-NoPro-MPC.bin"

Not sure about NoPro but this was as close as I could get.

I flashed
I rebooted, UI came up fine.
Reset Probe Offsets
Manually set Z-Offset, and confirmed it was correct with another reboot and
check Z

Tried to do a "Build Mesh"
It did the G28 Z
Moved to the First position and probed once while showing "Bilinear
Leveling" on the UI and...  Back to the UI after doing 1 probe, The Screen
still says "Biinear" at the top.  If I attempt to generate a mesh NOW, it
does the G28 Z sort of (only 1 probe) and then goes to try the corner and
never gets low enough for the probe to touch.

REBOOT and try the same thing....

It does a FULL G28 Z, two probes, one fast, one slow
Goes to the corner, probes the corner properly.... and STOPS

Without touching anything else, any attempt to generate a mesh now just
repeats with the proper G28 Behavior.

When It finishes the corner probe it lifts, pauses and then lifts again
before returning to the Menu.

Did I get the right version, or am I missing something else... is there
another step I need to do before tying to create a mesh.  My Bed is pretty
well trammed already but I can try seeing ig going through that procedure
makes any difference.

Give me a direct email, and I will send you videos of the behavior...

Bob Sisson