classicrocker883/MRiscoCProUI #99: [BUG]  ABL-BLT not working properly

inspectionsbybob, 2024-01-09:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Ok, I downloaded what I THINK is the correct version and Fashed it.
"Aquila_GD32_BLT-5x5-NoPro-MPC.bin"
My Machine is:
G32
BLtouch ABL
and I want MPC so I downloaded
"Aquila_GD32_BLT-5x5-NoPro-MPC.bin"

Not sure about NoPro but this was as close as I could get.

I flashed
I rebooted, UI came up fine.
Reset Probe Offsets
Manually set Z-Offset, and confirmed it was correct with another reboot and check Z

Tried to do a "Build Mesh"
It did the G28 Z
Moved to the First position and probed once while showing "Bilinear Leveling" on the UI and...  Back to the UI after doing 1 probe, The Screen still says "Biinear" at the top.  If I attempt to generate a mesh NOW, it does the G28 Z sort of (only 1 probe) and then goes to try the corner and never gets low enough for the probe to touch.

REBOOT and try the same thing....

It does a FULL G28 Z, two probes, one fast, one slow
Goes to the corner, probes the corner properly.... and STOPS

Without touching anything else, any attempt to generate a mesh now just repeats with the proper G28 Behavior.

When It finishes the corner probe it lifts, pauses and then lifts again before returning to the Menu.

Did I get the right version, or am I missing something else... is there another step I need to do before tying to create a mesh.  My Bed is pretty well trammed already but I can try seeing ig going through that procedure makes any difference.

Give me a direct email, and I will send you videos of the behavior...

Bob Sisson

### Printer Model

Voxelab Aquila

### Model Type

Aquila

### Your Mainboard

Aquila GD32

### Other Mainboard Type

I only know it as G32 not GD32

### Add-ons that could be involved

BL touch

### ProUI?

NoPro

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Please let me know what to try in the way of tests to see what it is doing.

I also can give you access to my PRONTO interface so You can play with system remotely 
(Its actually Repetier and Has a Camera as well)

classicrocker883, 2024-01-10:
go to one of the commits made by me, ex.:  
`https://github.com/classicrocker883/MRiscoCProUI/commit/65c3dd34d65d1c890be71d7cac25eb1a09d38cfc`  
copy the URL and add "`.patch`" to the end, you should see the email at the top

that is very cool I didnt know I could be able to access remotely that way. you can do that if you like - send access, I just dont know when I would be able to get around. I'd like to try anything I can on my end first.

the *-BLT* / Bilinear version is one I have focused on less, since there are a couple issues/or lets say _inconveniences_ - regarding that.  
for that reason, I do suggest trying _-UBL_ / Unified Bed Leveling.  
the thing about this, is it takes more up more memory space, so if I were to enable _-MPC_ for it, I would have to disable a couple other things, but these things are like: being able to change the UI color theme, or viewing End Stop diagnostics - stuff that doesnt effect printing in anyway.

but what I'm able to do is leave you with some options you can try out. I'm going to make a version with the most recent MRiscoC that just came out the other day. the UI is very different, but it may have some new code and fixes.

It just came out so I haven't gone through it just yet, which I would fork over anything that makes this one better.

inspectionsbybob, 2024-01-10:
Hum...
Will load up a UBL and see how it behaves...
-My- opinion is that once -a- bed leveling system works well, it should not
be needed but only occasionally.  Temperature control is something I use on
every print.
Other things like changing the color of the UI seem are silly cosmetic
things that I wouldn't use.  I do 99% of my setup via my version of
OctoPrint (Repetier) so don't use the UI much at all.

What I want is a system I can turn on, send file to it and have a high
level of confidence that it will print successfully.  I have my start file
set to warm everything BEFORE doing a G28 then a G29, a purge, a prime, and
then a print.  If I get a good first layer, I have good success unless
supports are involved.

Anyway, let me go download and flash....

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2024-01-11:
Ok, I downloaded the Recommended firmware.
Flashed it.
Rebooted
M502
M500
Re-set probe offsets
Set Z-Offset

TRIED to do a Build-Mesh.
It took the Bed center correctly. (G28 Z ??)
Then moved up (back) and to the right and almost probed.  Missed the bed by
about 2mm.  And then it probed again, and again... I turned it off at 7/49
It never moved from that first probe position!

NEXT Version ??

Bob Sisson

classicrocker883, 2024-01-14:
as soon as the issue im having doing the github Actions are worked out, I can then compile the new release version. hopefully sometime today.
I will upload **UBL** with _-MPC_, I recommend using that one.

classicrocker883, 2024-01-14:
>Then moved up (back) and to the right and almost probed.  Missed the bed by
about 2mm.  And then it probed again, and again... 

so the probe missed the bed, like it was off to one side?
have you updated your Probe Offsets??

it may be different for everyone.  
the probe should be touching the same point as with nozzle
https://www.youtube.com/shorts/FKvPU2nwdts

another thing is changing the Mesh inset. by making the Min X/Y inset bigger. or Max smaller, may help align where the probe goes.

inspectionsbybob, 2024-01-14:
Nope I had changed all the probe offsets. It was over the bed, and the
probe deployed, the carriage started down, and stopped before the probe
could reach the bed.

Question. Is there a "Z" probe offset? I re-loaded some old software as I
needed to print something so I can't go check.  If there is, how do you
choose a value?

What I am used to is the G28 command determines where the center of the bed
is in the z-plane with a double probe (fast then slow). The printer now has
a reference, so it knows -roughly- where the bed is when it moves to the
corner and without any further adjustment, probes the location to determine
the Delta.

My opinion is that the initial probe height should be the rough trigger
height of the bed center. It KNOWS that so I should not need to enter it.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

classicrocker883, 2024-01-15:
>Question. Is there a "Z" probe offset? I re-loaded some old software as I
needed to print something so I can't go check.  If there is, how do you
choose a value?

yes, I think this is why you are having difficulty. Z probe offset can be changed through "Z-Offset Wizard" in the "Prepare" menu.
sorry if there was any confusion.

Adjusting the Probe Offset for Z was removed from the "Probe Settings" menu, in "Level" Menu, which was next to the ones for X + Y. because it is redundant.

you can also edit the value using `M851 Z` + the value. so like `M851 Z-0.15` for a negative Offset, same goes for **X**|**Y**

https://marlinfw.org/docs/gcode/M851.html

I may have mentioned this somewhere else but the ideal way to level everything is first get your Z-Offset, using the wizard in "Prepare". I mean before anything you should go through the menu's settings in both "Control" > "Advanced Settings", and anything in "Level". then go to "Prepare" > "Z-Offset Wizard". follow the instructions, which should automatically move the nozzle to the center, and lower it to the bed. once there (or near), change the Z-Offset value using the knob, its better not to like paper in between and let the nozzle itself just barely touch the bed. once that happens, you should have a true Z-offset. a way to double check this is to either redo the "center nozzle on bed" or tell the printer to move to Z0.00, like `G1 Z0`. there should not be a gap, but also the nozzle should not be pushing down on the bed, except only just barely be touching.

then go ahead and perform the "Bed Tramming Wizard". I like to do it Manually first, which is basically like the same thing for Z-Offset, except when the nozzle goes to each corner, turn the knobs until it is just barely touching the bed, like for center. you may have to rehome Z and do this one more time for better precision. then by doing an auto Tram with the probe, the values don't have to be exactly zero, close enough is fine, thats why I like to do it Manually first and double check with the probe.

again if the values are off, you may need to adjust the Z-Offset value. then go ahead to the Level menu and perform an Auto Level procedure. you should have a mesh that is ready to print.

I have noticed that even if the mesh is +/-0.10 (max difference), you should be fine printing without auto bed leveling enabled (G29/M420)

I should be having a Build Release within an hour or two.

classicrocker883, 2024-01-15:
new release should be ready, finally

if you havent, check out the WIki https://github.com/classicrocker883/MRiscoCProUI/wiki/Calibration-Guides

inspectionsbybob, 2024-01-15:
Sorry, I tried several versions and they would never finish Flashing.

Had to go back to an Alex version.  That Flashed fine.

Just to confirm a MOSTLY STOCK Aqiuila should use an Aquila 422 Version
The last one I tried was Aquila_422_BLT-ProUI-MPC.bin

It would bring up the garbage screen, blink, and blink and then nothing.
If I reboot at this point, the VoxeLabs logo comes up and freezes.

Alex's version would blink, blink, then show a progress bar across the
bottom, and the UI would come up.

UPDATE:

I got a version to load...

Still won't probe properly.
Flashed Aquila_GD32_BLT-5x5-NoPro-MPC.bin
Loaded defaults to clear any garbage
Power Cycled to clear...
Re-Set the Probe offsets  (it kept them even after a flash and load
defaults and reset printer??)
Went to Build Mesh
Did the G28 right
Went to the left front Corner... It started to probe, stopped short of the
bed and aborted.

Bob Sisson

inspectionsbybob, 2024-01-16:
If you send me a direct email I will send you a video of what I am seeing.

I can also ZOOM with you and walk through what I am doing in realtime....

Bob Sisson

classicrocker883, 2024-01-19:
hi sorry what was your email?
also, the thing with these boards is they cannot flash a firmware larger than 228k, nor can they support non-Maple, that is the ones labeled for _-422_, which use an STM32. if your Aquila doesnt have that chip, it will not flash. only Maple environment is supported for GD32/N32

classicrocker883, 2024-01-19:
I think that when you hit load defaults, it resets the values so anything saved will be reverted.

sometimes some settings *do* save from the previous even after a new flash.

inspectionsbybob, 2024-01-19:
Ok, I know I have a "G32" board, but I don't know about the STM32.

I know that when I flashed the Alex firmware, I had to look for the "G32"
version.

I know that YOUR GD32 versions work about 80%, it's that 20% left that is
the problem, of course.

It may be that your version is pickier than Alex's was about the base
hardware...

Bob Sisson

classicrocker883, 2024-01-20:
> Ok, I know I have a "G32" board, but I don't know about the STM32.

you mentioned flashing the "422" version. that is meant specifically for if you had replaced the board with a creality 4.2.2 board, just like if it were a creality 4.2.7 board, you would use the "427" version. this is why some of the versions would not work. 

there are similarities and differences between the G32 aka GD32, and STM32 (which is on 427 and 422 boards).
1. 422/427 have 512k flash memory, GD32 has 256k.
2. GD32 and N32 use a Maple type bootloader. STM32 do not.
3. 422/427 firmwares are compiled using standard non-Maple. they will not flash on GD32 or N32.

that said, firmware .bin files 228k and under will work, because max is 256k (= bootloader + firmware.bin). if you have a "G32/GD32" use firmware that have only "GD32". 

have you tried the most recent release?   
Try the ProUI version with MPC

inspectionsbybob, 2024-01-20:
Got grandaughter today, so it will be later...

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2024-01-21:
Just flashed Aquila_GD32_BLT-ProUI-MPC.bin 20240116-7:17

Startup was far better....
Everything worked as it should... Garbage and flashing, then the status bar
and then a restart...
Then I did an M502 followed by a M500
reset the Probe Offsets
Went through the Z-offset wizard
Got a VERY different value of 6.10 when the old software was a 1.10 so
there is an offset somewhere
Tried to build a Mesh  FAIL!

Tried again, reflashed
Re-Homed before the Z-offset, manually lowered the Nozzle to 0.0 before
going to the Zwizzard.
Got the -1.10 I expected, had High hopes...
...Build Mesh
Did the SINGLE center probe, went to the corner, extended the probe,
lowered and stopped maybe .1mm from the bed and aborted.

OK, had a BIT of time to explore...

If I AUTOHOME, it probes the middle of the Bed TWICE.
Went through again, verified the Z-offset, DId several AutoHomes and Home-Zs
It probed the Left corner correctly! Yea, we are on a roll.  It moved to
position #2, lowered slowly, and aborted... It hadn't deployed the probe at
all!

The Home-Z seems inconsistent in how it behaves.  SOMETIMES it probes
twice, sometimes only once.  In trying to watch what it is doing, the first
Z-probe finds the rough offset, and the second is supposed to set it
exactly.  Everything is based off that center-Z.  No second probe, and we
don't really know where the bed is.

From what I can see, it is starting too HIGH for the probe to trigger
before it aborts. About the Z-offset too High.  When It is done with the
G28, it should go to 10mm.  It goes to 11.1mm, which is about 1.1mm too
high...  If I do a G1 Z0, it goes to the right location, just kissing the
bed.

The following is a terminal capture of a G28 followed by an attempted G29

Send:18:49:51.546: Slow command added:G28
Send:18:49:51.546: N21 G28
Recv:18:50:03.805: X:155.00 Y:134.00 Z:11.10 E:0.00 Count X:12400 Y:10720
Z:4440
Send:18:50:03.806: N22 M114
Recv:18:50:03.813: X:155.00 Y:134.00 Z:11.10 E:0.00 Count X:12400 Y:10720
Z:4440
Send:18:51:18.127: Slow command added:G29 V4
Send:18:51:18.127: N23 G29 V4
Recv:18:51:26.643: X:155.00 Y:134.00 Z:11.10 E:0.00 Count X:12400 Y:10720
Z:4440
Recv:18:51:26.646: G29 Auto Bed Leveling
Recv:18:51:26.664: Probing mesh point 1/25.
Recv:18:51:36.283: Error:Probing Failed
Recv:18:51:36.288: X:55.00 Y:34.00 Z:11.10 E:0.00 Count X:4400 Y:2720 Z:4440
Send:18:51:36.320: N24 M114
Recv:18:51:36.346: X:55.00 Y:34.00 Z:11.10 E:0.00 Count X:4400 Y:2720 Z:4440

PLEASE send me a message with a direct email.

Bob (at) Sissonfamily (dot) com

I can send you videos of what is happening, or let you terminal access to
the printer.

That way we don't load up GitHub with conversations...

Bob Sisson

classicrocker883, 2024-01-21:
what probe are you using? BL-touch, CR-touch, 3D-touch?

sometimes there may be issues using these. apparently there has been reports of the CR-touch having issues. I have a cr-touch and have not experienced any problem.

I couldn't tell you why you are having the issue but yes definitely send a video in email. I sent you one back.

inspectionsbybob, 2024-01-21:
3D touch.   I never had any problems with Aquila or Alex firmware.  If I
run the M48 test it does fine.

Bob Sisson

inspectionsbybob, 2024-01-22:
> so when it is saying 11.10 that is the minimum clearance of 10mm + offset.
>
> it appears your Z offset is +1.10
> this seems not correct. usually if you have a probe the offset will be (-)
> negative.
>
> lets say you reset the Z offset to 0.00. G28 Z should bring the nozzle to
> the center. do G1 Z0, observe the relation between the nozzle and bed. the
> nozzle should be above the bed slightly, right?
>
> this is another way to change the Z offset. you want a negative number for
> Z offset - you want it until the nozzle reaches the bed, then stop and
> record the offset. you may have to enter M211 S0 to turn off soft endstop.
> remember to do M211 S1 after this is done.
>
> so you can manually tell the printer to move down with M290 Z-0.1. it
> might help using paper, do this until the paper doesn't move. whatever
> number Z is at should be your Z offset. it should be a negative number and
> don't forget to take away the thickness of the paper which is usually 0.10mm

Ok...
a G1 Z0 takes me to the right height, but that isn't used by the leveling
routine, apparently.

Let me do some more testing.

When I use the Z-offset Wizard, it starts at 0.0 typically (depends on if
the Flash wiped it).  I do a Home Z, and then a Z-center, and then
Z-offset.  I lower the Head until it drags hard.  That SHOULD be ~ -1.10mm
which it did feel right.  an M500 to save the offset.  Then a try at a Mesh
Build, which is where it normally fails.  Typically I can tell how quick it
is going to fail (spot #1 or Spot #2) based on the Z-Home.  If it does ONE
probe it will fail immediately.  If it does TWO probes, it will fail on the
second position.

I typically use a piece of PostitNote, which is 0.08mm thick.  I lower the
Nozzle in the Z-offset UI, which changes depending on which printer I am
using, until it drags med-hard.  When I print a single layer BRIM it
measures 0.23mm which is pretty close to the 0.2mm target, so I know my
routine works (and it works on my other 2 printers running Marlin)

Bob Sisson

Cr-opas, 2024-01-23:
I really like your version of the firmware, but I have the same problems, one day I started printing and almost broke the extruder nozzle because the offset for some reason went wrong, before this incident several prints went through without problems, the 3D touch does not work, the same problem I can’t build a mesh, I feel like after the next firmware some data remains in memory from the previous firmware. Also, the printer does not correctly determine the area of the bed when manually setting the bed level; all points are shifted forward by about 4 centimeters. I tried to compile the firmware myself from your sources (I need a maximum bed temperature of 120 degrees, and presets for a larger amount of ABS, PETG, Nylon, PLA filament), the same problem. Aquila x2 N32.

classicrocker883, 2024-01-25:
@inspectionsbybob were you able to send me the video via email? if that doesn't work send it to the same screen name `@gmail.com`.

>the 3D touch does not work, the same problem I can’t build a mesh, I feel like after the next firmware some data remains in memory from the previous firmware.

@Cr-opas in what way does your probe not work?? it is possible that sometimes data does roll over.

> Also, the printer does not correctly determine the area of the bed when manually setting the bed level; all points are shifted forward by about 4 centimeters. I tried to compile the firmware myself [...]

for bed temps I might be able to increase it next build. as for points being shifted, this might be due to the mesh_insets. after starting, go to the leveling menu, and Probe settings, look for Mesh Insets. make sure those values can cover the bed. its possible that the "workaround" I made hasn't loaded properly. so what happened originally, the mesh_insets would automatically shrink based on the probe offsets. this is so all the points can be reached by the probe, basically moving the points inward so the probe can reach. this isnt right because you want the mesh to be able to take up as much as the bed as possible, so I implemented a workaround. the mesh_insets should "max" out covering most of the bed. you should be able to adjust the inset if necessary, defaults should place a 10mm inset from the edge.

you can test this by going to the edit mesh menu, set the probe to any point along each edge, and in essence you should have a perimeter 10mm inside of the bed. this is for a couple reasons, 1. being you dont want to probe exactly on against the edge. you can always adjust this to account for any bed clips that might be in the way.

also, are you using the "*_BLT*" version or "*_UBL*". someone had pointed out in the _BLT version that the mesh may not load correctly from starting, and that either manually loading settings or reset defaults fixes it.

inspectionsbybob, 2024-01-25:
i can try to send it again.

Does the Bltouch work?  I didn't know there was any difference in the
drivers, especially since it works fine on my other Marlin printer.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

classicrocker883, 2024-01-25:
there has been reports of the probe not working correctly. however that was in the past and should have been fixed then. I have a CR-touch and not had a problem, but some others report issues using this.

These Aquila's might be a clone of Creality Ender-3V2, but there are still differences which sometimes there is an issue with the probe.

just make sure the probe is plugged in correctly, some come with wires with split connectors, like having one with 2 wires, and one with 3, instead of the usual 5 in a row on just 1 connector.

inspectionsbybob, 2024-01-25:
All I know is that it worked with the Voxelabs and Alex software just fine.

The only reason I am moving away from Alex's build is I want MPC as
temperature control has been my big problem forever.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

classicrocker883, 2024-01-26:
oh one thing I forgot to mention was try disabling HS mode - High Speed mode. its in the Probe settings, bottom setting below M48. I bet you that might be it, thats something that has been an issue. You can also try changing "Extra Probing". its the same as "Multiple Probing". extra meaning additional probes. there was some weird thing going on since the ProUI_EX features were updated, it used to work normally as in it probed twice. but I set it to 0, because 1 doesnt work with HS mode. you can set it to another value (like 2) if you want but it will probe 3 times +.

classicrocker883, 2024-01-27:
> > I can confirm with HS mode disabled bed meshing and tramming wizard work.
> > Bug or feature? ;-)
> 
> HS is an enhanced feature, some probes (clones of the original Antclabs BLTouch) are incompatible with HS mode.

https://github.com/mriscoc/Ender3V2S1/issues/1150#issuecomment-1819808321

inspectionsbybob, 2024-02-03:
Ok, I have a few less busy days....

What is the latest version to try?

I want to start with JUST BLT and as large a Grid as I can get.

If that levels, and will hold a temperature I will add features...

So which release to start with GD32-BLT-XxY- then what

Bob Sisson

classicrocker883, 2024-02-08:
hi yes busy days for sure, I'm working on a few fixes, and so that we are able to adjust the mesh and grid array in the NoPro version, like the ProUI with all the extra features.

i've been trying to get the multiple/extra probing to work like it is supposed to, that is being able to set any value 0-4. where 0 - probes once, 1 - two times, 2 - three times.....so on. 

the latest right now is for January. I would go for one of the UBL versions. which is just an enhanced BLT. here is one w/MPC
[Aquila_GD32_UBL-ProUI-MPC.bin](https://github.com/classicrocker883/MRiscoCProUI/releases/download/2.1.3f-1/Aquila_GD32_UBL-ProUI-MPC.bin)

you can also use a "NoPro-#x#-" file, like 7x7. are you using a different size bed? because to be frank, you don't really need to go higher than 7x7 on a bed this size. the reason why 7x7 is ideal is because the probe offset limits how far right it can probe, so unless you lessen the X max mesh_inset, the last row cannot actually be probed. but the firmware is able to "fill in the blanks" accurately. so its a good compromise because you are able to get the most probed points without sacrificing more than 1 row. like if it were 8x8 or 9x9, at least the last two rows wont be probed.. again it can be filled it smartly.

and you can always check the actual offset manually if need to.

inspectionsbybob, 2024-02-08:
Great!
I was using the Alex 9x9 with success.

I will download it and give it a try soon.

Bob Sisson

classicrocker883, 2024-02-10:
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-2

new build here. there has been a few fixes, ill do a complete list later, but real quick I updated the Multiple Probing so that it actually works as it should. 1 probes once, 2 two times, so on...

trammingwizard works better as well, cancel bed leveling... for the _-NoPro_ versions, I was able to make it so you can adjust the grid_array, 3x3 5x5 ect..., mesh_inset's, zprobe speed...

let me know how everything is, any changes to be made. also if you still have issue with BL touch, disable HS mode in the Probe Settings menu, under Level menu

inspectionsbybob, 2024-02-15:
Ok, first problem....

MANUAL leveling doesn't deploy the probe
AND
It doesn't recognize the probe offsets. Bottom left, where I start, was too
far left so It was off the bed,

I mostly use auto leveling so I can move forward with this...
Bob Sisson

classicrocker883, 2024-02-16:
> That is more than all the previous versions have done.

this is good news!

> Ok, first problem.... MANUAL leveling doesn't deploy the probe AND It doesn't recognize the probe offsets. Bottom left, where I start, was too far left so It was off the bed, I mostly use auto leveling so I can move forward with this...

Manual leveling is not supposed to deploy the probe. the purpose is to check the bed level without the probe at all. if you are familiar with (_-MM_) Manual Mesh Bed Leveling then you would know what that is for. basically instead of using a probe, you're able to find the level of the bed by the distance of the nozzle from the bed.

for instance, using a piece of paper to gauge the distance. the Manual leveling part when having a probe is to double check in a sense. or if the probe cannot reach certain parts of the bed, manual leveling is a way to check the level using the nozzle.

so basically you take the piece of paper (i actually dont use anything, no gauge no paper, I have the nozzle come straight to the bed surface and thats how i know the level, anyway...) place it under the nozzle and the point you with to probe, and find the "sweet spot" and record the distance like you would tramming a corner of the bed. then you would subtract the thickness of the paper from the measurement and thats your new value.

since you have a bed probe this really isnt necessary, but its there to check any of the grid points if you wish.

inspectionsbybob, 2024-02-16:
Ok, I may be confusing trimming with leveling.

The best version I have used looked the same, up to the behavior at the
corner. the first corner it -MEASURED- became the reference and then you
could go to any corner and it would tell you the delta, no paper.  I used
it to get the bed really really close ( +/-  .05) and THEN used the ABL.
Then a G28 Z, then set my offset with paper.

Bob

classicrocker883, 2024-03-01:
hello I gipe everything has been working as it should since that new update. I think the previous issue could have been due to a small change in the code which changed how "Center Nozzle on Bed" behaved, which would explain the main issue, the Z offset distance was registering too far.Z
34
basically Center nozzle in bed should have places the nozzle in the center, and also lower it to nearly touching the bed surface. you can see the differences between firmware versions.

inspectionsbybob, 2024-03-01:
I have been running the basic version for a week or so with no other issues.

I will try a MPC version next…  thought the version I downloaded said it had it but can’t find any reference to it in control or prepare. when tried to invoke the test with an Mcode it said not found

closed 2024-06-03 by github-actions[bot]