classicrocker883/MRiscoCProUI #100: VERY DANGEROUS: hot end temperature is read incorrectly [BUG]

PolcovnicGauss, 2024-01-12:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

firmware HC32_Default-NP-MPC and HC32_Default-NP. I waited half a year until this problem was solved and everything is in the same place, when the hot end is turned on, the temperature reaches 160-170 degrees and then starts jumping here and there, please do not ask to check the heater or temperature sensor, since everything is fine with them and at stock Everything works fine with the firmware and without any glitches!

### Printer Model

None

### Your Mainboard

Aquila HC32

### ProUI?

ProUI

### Bed Leveling

Default - No Probe / No Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-01-13:
is this the same as this one here? https://github.com/classicrocker883/MRiscoCProUI/issues/95

the HC32 is something I don't have that much control over. those issues I want to direct you over to https://github.com/shadow578/Marlin-H32/issues

otherwise you might get someone else who might know exactly how to solve it over at MarlinFirmware, now since it has incorporated the HC32 board
https://github.com/MarlinFirmware/Marlin/issues

to be honest, everything *should* be working regarding any sort of issues like this. if not, I will try to do my best, but here my specialty is _UI_ related, concerning the LCD menu - its layout - and any other **ProUI** features based on **MRiscoC's Professional Firmware**.  
I won't be much help regarding anything HC32 specifically. like I said shadow578 would be much more help concerning this, because if this is not even Marlin code related, then it is specifically with the board and it's SoC (the main HC32 chip platform/framework).

that said, my guess is it has to do with that, and not so much with anything in my control, or if it is, I wouldn't know where to start.

<br>

so please make a post about it over in MarlinFirmware and shadow578's repo. link this issue's URL to that as well so I may be able to keep track.

PolcovnicGauss, 2024-01-13:
no, in my case the problem is different, the temperature rises adequately to 160-170 degrees, after which it starts jumping here and there, maybe up to 200 and then drops sharply, I pulled out the SD card and tried it a few more times, with MPC firmware, maybe by accident, maybe not but once I managed to successfully calibrate, I tried small prints, there seemed to be no problems

classicrocker883, 2024-01-14:
ok it sounds like i can close this then

closed 2024-01-14 by classicrocker883