classicrocker883/MRiscoCProUI #101: [BUG]

PolcovnicGauss, 2024-01-13:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

as it was raw and unfinished, it remains as it was in the summer and now, I put it to print, the part is printed until about 7-8 layers, after which the printer freezes and does nothing. There is also a problem with linear advance: when I set the k value, the higher it is, the greater the effect, the printer stupidly reduces the print speed, what the hell is this?!

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila HC32

### ProUI?

ProUI

### Bed Leveling

Default - No Probe / No Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

PolcovnicGauss, 2024-01-13:
if I print via a USB cable, then printing proceeds, but after 10 minutes the printer itself finishes it, turning off the heating and fans, while commands from the PC go and printing proceeds without heating, what a waste

classicrocker883, 2024-01-13:
it seems as though something is triggering that to happen maybe the LCD timeout, which is set to 10. or maybe the hotend idle timeout, which really should keep it heated if printed. that is set to 6mins.

try putting the LCD timeout to 0  or to some greater number and see if that actually works.

I'm going to post a Vanilla version so it's just bare Marlin, perhaps also a CrealityUI and JyersUI for testing.

I mentioned this in another HC32 issue to also post to shadow578 and at MarlinFirmware on their issues tab.

PolcovnicGauss, 2024-01-15:
I flashed HC32_Default-NP-ProUI.bin
 instead of the version with MPC, it began to work normally, even if you leave the screen timeout for 10 minutes, but there is another error: if you turn the encoder a little faster or try to change the settings while printing after some time, the printer freezes and stops printing

PolcovnicGauss, 2024-01-15:
> it seems as though something is triggering that to happen maybe the LCD timeout, which is set to 10. or maybe the hotend idle timeout, which really should keep it heated if printed. that is set to [...]

it wouldn’t be bad to release firmware with simple Marlin, which would have a normally working linear advance, that’s essentially all I need, I can’t compile it myself, my PC doesn’t support vs and via arduino ide I don’t know how to connect the HC32 core for compilation

classicrocker883, 2024-01-15:
- you can always use github codespaces to compile. creak a fork of this Repo into one of your own. in the branch drop down select HC32-January, then youll see a big green Code button,  go to codespaces and start one up. a Web VScode page should start. then go to the configurations and edit whatever you need to. a bunch of things to initialize will happen, just ignore other popups you might see, you may need to say yes to install or reload (doesn't actually install on your computer just the webpage).  
once that is done you can simple build, its a check mark near the bottom of the page along the bottom toolbar. it should start compiling after. once it is done and successful, navigate to the left pane, where all the folders are  
go to **.pio/build/HC32F460C_aquila_101/** youll see a firmware_.bin_ file. right click and download. 

thats your firmware.

classicrocker883, 2024-01-15:
> I flashed HC32_Default-NP-ProUI.bin instead of the version with MPC, it began to work normally, even if you leave the screen timeout for 10 minutes, but there is another error: if you turn the [...]

I think that was an actual issue from before, I thought I had solved it.

I can try some things like changing the encoder response time.

PolcovnicGauss, 2024-01-16:
What to do with the fact that when the linear advance coefficient increases, the printer slows down the printing speed? despite the fact that the speed is set to the same for the entire part, on calibrated parts with each increase in linear adnvance the printing speed becomes less and less

classicrocker883, 2024-01-16:
> What to do with the fact that when the linear advance coefficient increases, the printer slows down the printing speed? despite the fact that the speed is set to the same for the entire part, on calibrated parts with each increase in linear adnvance the printing speed becomes less and less

i'm not sure about that why. is linear advance supposed to do that anyway, to compensate?
I mean, it should slow down when moving in corners, but youre saying throughout the whole print it is doing that?

[here is a new release](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-HC32)  
lets see if this makes a differences.

classicrocker883, 2024-01-19:
I just created something I'd like you to try. it is different UI versions, with using Marlin. I did make some changes, but nothing like before. so please try the different versions if you can, let me know which one works for you.

also, its important to know if the issue you are having - or any other issue at all using the "ProUI"

closed 2024-01-19 by classicrocker883

classicrocker883, 2024-01-21:
I just realized that Ejerk is not recommended to be less than 10 with linear advance. change your E jerk setting in Motion in Control to at least 10