classicrocker883/MRiscoCProUI #110: Timer never starts again after screen turns off [BUG]

acousticpunker00, 2024-02-01:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

The print timer will stop counting as soon as the screen goes to sleep, and never updates again even after you wake the screen back up.
[image]

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

Aquila N32

### ProUI?

ProUI

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-02-07:
in my testing I havent been able to reproduce this. so what I did was manually turn off the screen when its printing. and set the timer to like 1 minute instead of waiting. upon returning it was still counting. so when this happens, I know that turning the knob alone might bring back the display, but it doesn't allow any interface unless you actually push in the knob.

can you confirm that? see if theres a difference when turning the knob vs pushing in the knob.
also, you can put `M111 S8` in the beginning of the start Gcode, this enables DryRun mode. basically it pretends/thinks to print without actually heating or printing.

acousticpunker00, 2024-02-08:
Looks like you're right. On the second button press, the timer updated.
I can't believe I didn't try that before, but maybe it wasn't working on the last firmware, but is now on the new one.

Thank you!

closed 2024-02-08 by acousticpunker00