classicrocker883/MRiscoCProUI #113: When FT_Motion enabled Print doesn't start and Homing failure [BUG]

Weresk, 2024-02-13:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When enable FT motion with command M493 S11 A37 B37 D0 P1 K0.18  homing failure (as if he doesn't see endstops)

### Printer Model

Voxelab Aquila

### Your Mainboard

Aquila GD32

### ProUI?

NoPro

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/classicrocker883/MRiscoCProUI/files/14270095/Configuration_adv.zip)

classicrocker883, 2024-02-16:
I'm not familiar with `FT_MOTION` at all. I do know it is Experimental.

I would try using this without `PROUI_EX` enabled and see if that changes anything.

You can try testing the [Marlin source code](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.1.x.zip), and transfer over your configuration files (*Configuration.h**, **Configuration_adv.h**).
- where you find in the first config `#if ENABLED(DWIN_LCD_PROUI)`, just change that to `#if 0` or whatever so it's not pre-compiled.
- you may have to change `#define HEATER_0_MAXTEMP 315` to `275`.

classicrocker883, 2024-02-16:
you can also check the Frequency for stepper motors, any of the variables you can change, such as `ftMotionMode_t`/`dynFreqMode_t`

Weresk, 2024-02-16:
> I'm not familiar with at all. I do know it is Experimental.`FT_MOTION`
> 
> I would try using this without enabled and see if that changes anything.`PROUI_EX`
> 
> Also, what about Linear Advance or Input Shaping? do these need to be enabled or disabled?
> 
> You can try testing the [Marlin source code](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.1.x.zip), and transfer over your configuration files (_Configuration.h_*, **Configuration_adv.h**).
> 
> * where you find in the first config , just change that to  or whatever so it's not pre-compiled.`#if ENABLED(DWIN_LCD_PROUI)``#if 0`
> * you may have to change  to .`#define HEATER_0_MAXTEMP 315``275`
> 
> if you get the same issue using that, post a new **Issue** over there and copy the link to this one if you can. I'm betting this is more a Marlin issue rather than to do with ProUI

I get an error when compile
 Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void setZOffset()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:2118:22: error: 'PROBE_OFFSET_ZMIN' was not declared in this scope
     setPFloatOnClick(PROBE_OFFSET_ZMIN, PROBE_OFFSET_ZMAX, 2, applyZOffset, liveZOffset);
                      ^~~~~~~~~~~~~~~~~
Marlin\src\lcd\e3v2\proui\dwin.cpp:2118:41: error: 'PROBE_OFFSET_ZMAX' was not declared in this scope
     setPFloatOnClick(PROBE_OFFSET_ZMIN, PROBE_OFFSET_ZMAX, 2, applyZOffset, liveZOffset);

classicrocker883, 2024-02-17:
>I get an error when compile
Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void setZOffset()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:2118:22: error: 'PROBE_OFFSET_ZMIN' was not declared in this scope
setPFloatOnClick(PROBE_OFFSET_ZMIN, PROBE_OFFSET_ZMAX, 2, applyZOffset, liveZOffset);

using Marlin source code? this is because things are a bit different. I requested a change to be made regarding this error, hopefully the **PR** can go through. so this is what you'd have to do.

in **Marlin\src\inc\Conditionals_LCD.h** line ~1490

```y
/**
 * Conditionals based on the type of Bed Probe
 */
...
  #undef PROBE_OFFSET_ZMIN
  #undef PROBE_OFFSET_ZMAX
```
`PROBE_OFFSET_ZMAX/MIN` is disabled when `!HAS_BED_PROBE` / when `MESH_BED_LEVELING` (_-MM_).

just comment out those two `#undef` from that line

Weresk, 2024-02-18:
> comment out

The same error or I misunderstood what to change.
[image]

classicrocker883, 2024-02-20:
no that is the right #undef to comment out. i dont get why its not defining either. you can just do this in **Configuration.h** line ~1670
```diff
-//#define PROBE_OFFSET_ZMIN -20   // (mm)
-//#define PROBE_OFFSET_ZMAX  20   // (mm)
+#define PROBE_OFFSET_ZMIN -20   // (mm)
+#define PROBE_OFFSET_ZMAX  20   // (mm)
```

<br>

because in **src/inc/Conditionals_post.h** it is:
```y
#if ANY(MESH_BED_LEVELING, HAS_BED_PROBE)
  #ifndef PROBE_OFFSET_ZMIN
    #define PROBE_OFFSET_ZMIN -20
  #endif
  #ifndef PROBE_OFFSET_ZMAX
    #define PROBE_OFFSET_ZMAX  20
  #endif
#endif
```

meaning since you have `MESH_BED_LEVELING` and if its not defined already, it should define itself.

however, in my repo I have changed this to:
```y
#if ANY(BABYSTEPPING, PROBE_SELECTED)
  #define HAS_ZOFFSET_ITEM 1
#endif

#if ANY(PROBE_SELECTED, HAS_ZOFFSET_ITEM)
  #ifndef PROBE_OFFSET_ZMIN
    #define PROBE_OFFSET_ZMIN -20
  #endif
  #ifndef PROBE_OFFSET_ZMAX
    #define PROBE_OFFSET_ZMAX  20
  #endif
#endif
```

you can copy this snippet over if you'd like and see if that works

Weresk, 2024-02-20:
> no that is the right #undef to comment out. i dont get why its not defining either. you can just do this in **Configuration.h** line ~1670
> 
> ```diff
> -//#define PROBE_OFFSET_ZMIN -20   // (mm)
> -//#define PROBE_OFFSET_ZMAX  20   // (mm)
> +#define PROBE_OFFSET_ZMIN -20   // (mm)
> +#define PROBE_OFFSET_ZMAX  20   // (mm)
> ```
> 
> 
> because in **src/inc/Conditionals_post.h** it is:
> 
> ```yacc
> #if ANY(MESH_BED_LEVELING, HAS_BED_PROBE)
>   #ifndef PROBE_OFFSET_ZMIN
>     #define PROBE_OFFSET_ZMIN -20
>   #endif
>   #ifndef PROBE_OFFSET_ZMAX
>     #define PROBE_OFFSET_ZMAX  20
>   #endif
> #endif
> ```
> 
> meaning since you have `MESH_BED_LEVELING` and if its not defined already, it should define itself.
> 
> however, in my repo I have changed this to:
> 
> ```yacc
> #if ANY(BABYSTEPPING, PROBE_SELECTED)
>   #define HAS_ZOFFSET_ITEM 1
> #endif
> 
> #if ANY(PROBE_SELECTED, HAS_ZOFFSET_ITEM)
>   #ifndef PROBE_OFFSET_ZMIN
>     #define PROBE_OFFSET_ZMIN -20
>   #endif
>   #ifndef PROBE_OFFSET_ZMAX
>     #define PROBE_OFFSET_ZMAX  20
>   #endif
> #endif
> ```
> 
> you can copy this snippet over if you'd like and see if that works

Updated to the latest Marlin firmware,   Exactly the same behavior, after enabling M493 S11 A37 B37 D0 P1 K0.18 failure homing and rattling motor sound, as if it doesn't see endstops.

classicrocker883, 2024-02-21:
you can try commenting out `OLD_ADAPTIVE_MULTISTEPPING` from **Configuration_adv.h**
```diff
-#define OLD_ADAPTIVE_MULTISTEPPING 1
+//#define OLD_ADAPTIVE_MULTISTEPPING 1
```

or if its not there or enabled, try with it.

if that doesnt change anything then you should post [the issue on their issues tab](https://github.com/MarlinFirmware/Marlin/issues/new?assignees=&labels=Bug%3A+Potential+%3F&projects=&template=bug_report.yml&title=%5BBUG%5D+%28bug+summary%29)  
someone might know more about this

Weresk, 2024-02-21:
> ```diff
> TERN_(ENDSTOP_INTERRUPTS_FEATURE, endstops.update());
> ```
Oh my God! Did everything written above except  disabling ADAPTIVE_STEP_SMOOTHING.
Checked homing, it works. Now need to find the calibration instructions, how to calibrate new LA and Shaping.

Weresk, 2024-02-21:
I was happy early, the same problem, it dont start printing https://github.com/MarlinFirmware/Marlin/issues/26614?ysclid=lsw70odr9u441942616

classicrocker883, 2024-02-22:
> I was happy early, the same problem, it dont start printing [MarlinFirmware/Marlin#26614](https://github.com/MarlinFirmware/Marlin/issues/26614)

did you try each thing seperately?

classicrocker883, 2024-02-22:
>> I notice what standard is zv was enabled, and maybe because of it, ft_motion did't work properly.

>We don't prevent Input Shaping & Fixed-time-based Motion Control from being compiled together & made available at the same time, but maybe more testing needs to be done around these.

_Originally posted in https://github.com/MarlinFirmware/Marlin/issues/26614#issuecomment-1874383844_

it was recommended to uncomment/disable `EDGE_STEPPING` from **Configuration_adv.h**
from this post here: https://github.com/MarlinFirmware/Marlin/issues/26683

im not sure what else the problem is, so it Homes okay now but it just wont start print?  
I wonder, what if you removed G28 from the start gcode, and manually auto home before the print.

Weresk, 2024-02-23:
> > > I notice what standard is zv was enabled, and maybe because of it, ft_motion did't work properly.
> 
> > We don't prevent Input Shaping & Fixed-time-based Motion Control from being compiled together & made available at the same time, but maybe more testing needs to be done around these.
> 
> _Originally posted in [MarlinFirmware/Marlin#26614 (comment)](https://github.com/MarlinFirmware/Marlin/issues/26614#issuecomment-1874383844)_
> 
> it was recommended to uncomment/disable `EDGE_STEPPING` from **Configuration_adv.h** from this post here: [MarlinFirmware/Marlin#26683](https://github.com/MarlinFirmware/Marlin/issues/26683)
> 
> im not sure what else the problem is, so it Homes okay now but it just wont start print? I wonder, what if you removed G28 from the start gcode, and manually auto home before the print. or vice-versa. try not auto home before print.

Experimentall way I understood  if insert in starting Gcode M400 before M493 S11 D0 printing starts...

Weresk, 2024-02-24:
I'm tired of trying out the new firmware, I can't adjust the LA because values no usable, https://github.com/MarlinFirmware/Marlin/issues/26642 
I printed test Ringing Tower STL and I can't see the differences( Maybe ft motion doesn't work...
[image]
[image]

classicrocker883, 2024-02-24:
I do not even know how FT_Motion works. is it supposed to replace Linear Advance + Input Shaping? 

it says this **PR**-[FT_MOTION : M493, reporting now shows the same values as is in the menu](https://github.com/MarlinFirmware/Marlin/pull/26643)
changing
```diff
-if (dynamic) SERIAL_ECHO(" scaling: ", p_float_t(ftMotion.cfg.dynFreqK[X_AXIS], 8), F("Hz/"), z_based ? F("mm") : F("g"));
+if (dynamic) SERIAL_ECHO(F(" scaling: "), p_float_t(ftMotion.cfg.dynFreqK[X_AXIS], 2), F("Hz/"), z_based ? F("mm") : F("g"));
```
i do not understand this change precision from 8 to 2.  
`p_float_t(ftMotion.cfg.dynFreqK[X_AXIS],` `8` -> `2`

meaning 8 decimal places to 2. so should K value be like K0.000006 or K0.6? because normal linear advance K0.6 is fine, but FT_Motion is different?

Weresk, 2024-02-25:
// Check endstops on every step
-IF_DISABLED(ENDSTOP_INTERRUPTS_FEATURE, endstops.update());
+TERN_(ENDSTOP_INTERRUPTS_FEATURE, endstops.update());

What should I do? Leave this line (IF_DISABLED(ENDSTOP_INTERRUPTS_FEATURE, endstops.update());)  ?

classicrocker883, 2024-02-29:
> // Check endstops on every step -IF_DISABLED(ENDSTOP_INTERRUPTS_FEATURE, endstops.update()); +TERN_(ENDSTOP_INTERRUPTS_FEATURE, endstops.update());
> 
> What should I do? Leave this line (IF_DISABLED(ENDSTOP_INTERRUPTS_FEATURE, endstops.update());) ?

I think it should be `TERN_` inplace of `IF_DISABLED`.

because it doesnt make. why would there be `endstops.update()` if `ENDSTOP_INTERRUPTS_FEATURE` is disabled.

KimmoHop, 2024-03-26:
> I'm tired of trying out the new firmware, I can't adjust the LA because values no usable, [MarlinFirmware/Marlin#26642](https://github.com/MarlinFirmware/Marlin/issues/26642) I printed test Ringing Tower STL and I can't see the differences( Maybe ft motion doesn't work...

What was FTM mode in those tower tests? 11? Enabled (1) just means it uses FT motion for stepping, but no input shaping.

I tried printing with FTM once. Start G-code enabled shaping right after after homing, but printer skipped next line - _going back to front-left corner_ - before printing purge line :/ It was some time ago, though it should had been fairly usable and working, but I have better things to do (Tinkercad, doomscrolling FB stories :) ) in my life ;)

> unfortunately this doesnt work on ProUI, for this to work a menu function would need to be made, it shouldn't be too difficult to create, just need to know which parameters.

I wrote some bad menu code for setting FTM parameters. Selecting values from discontinuous (uncontinuous? incontinuous? noncontinuous) range (0, 1, 10, 11 etc for FTM mode - I wonder **why on earth** they did it that way) is not quite easy, even less if you want to show labels :D ProUI (and probably E3V2 in general?) has kind of weird menu/parameter system, where starting values, shown values and values to be saved all have separate handlers. Plus best parts of menu value handling is in closed source, though same things should be possible with public interface and some elbow grease.

Weresk, 2024-03-27:
> What was FTM mode in those tower tests? 11? Enabled (1) just means it uses FT motion for stepping, but no input shaping.
>
> I tried printing with FTM once. Start G-code enabled shaping right after [...]

S11  mzv shaping

yrabbit, 2024-06-15:
I felt that there was something very simple there - like forgetting to check endstops :smile: 

https://github.com/MarlinFirmware/Marlin/pull/27179

classicrocker883, 2024-06-16:
I'll close the issue since that **PR** was merge which should fix the issue here. 

> I felt that there was something very simple there - like forgetting to check endstops 😄
> 
> [MarlinFirmware/Marlin#27179](https://github.com/MarlinFirmware/Marlin/pull/27179)

I will be merging this and other updates for the June release so look out for that soon

closed 2024-06-16 by classicrocker883