classicrocker883/MRiscoCProUI #115: Hotend and board fans stopped booting after I used "change filament" during a print

Big-Channie, 2024-02-24:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Hotend and board fans stopped booting after I used "change filament" during a print.

### Printer Model

Voxelab Aquila

### Model Type

X3

### Your Mainboard

Aquila N32

### Add-ons that could be involved

BLtouch

### ProUI?

ProUI

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm running the "Aquila-TJC_N32_BLT-ProUI" build of the Mriscoc firmware for my Aquila X3, and after using the "filament change" function during a print (my spool of blue pla+ ran out), the hotend fan started acting funny (turned off while changing filament, back on when resuming print, then off for a few short periods during print). Came back later in the day and now neither the board fan or the hotend fan will turn on when booting up my printer (usually they are both on at all times).

I've seen some printers out there will turn off the hotend fan while idle, then back on when heating, so I tried heating up the hotend but neither of the fans would start.

I checked the board for loose wires or a fuse or something, but no luck. Tried the "reset printer" function in the control menu as well. Everything else seems fine. I'll probably try re-flashing my firmware, but I don't want to lose all my settings and changes.

Big-Channie, 2024-02-24:
Update, I've tried flashing a few different firmware versions with no success, so this may have caused a short in my board.

closed 2024-02-25 by Big-Channie

classicrocker883, 2024-02-29:
to reflash the firmware you should rename the file if you havent done so already. otherwise it will see the same file name and not update.

there can be a few different reason for a short to be caused, but using a function or flashing the firmware should not cause this. I would suggest taking the hotend fan off and look at the wires which connect to it, should be under the fan sticker. sometimes these arent soldered so well and can come loose.

also, thermistors can also break down, I would recommened upgrading to a cartridge style instead of the glass bead ones because they are easy to break. even the one that goes under the bed. I had my original bed thermistor break and cause a short in the board just by heating to 100ºC when attempting to print ABS for the first time.

if you have a multimeter, that is a for sure way to test for any wires which may be loose or shorted, double check all points where they are phyically connected