classicrocker883/MRiscoCProUI #117: [BUG] Aquila. NoPro version does not move the carriage while creating a level map

yrabbit, 2024-03-22:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

"Auto Build Mesh" should check all mesh points, but actually stays in the first position position, although it draws what moves on the mesh.

Observed only in Aquila_GD32_UBL-NoPro-MPC.bin
The Aquila_GD32_UBL-ProUI-MPC.bin version works as expected.
But it's too redundant for me and I constantly end up in this small menu of small buttons, so I would like to use NoPro.

### Printer Model

Voxelab Aquila

### Model Type

Aquila 

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

BL-Touch

### ProUI?

NoPro

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

NoPro leveling:
https://github.com/classicrocker883/MRiscoCProUI/assets/6075465/b3c95bfa-735f-4536-baa3-c4e8d4680885

Version:
https://github.com/classicrocker883/MRiscoCProUI/assets/6075465/ec197e97-ceb9-4b90-9be4-4a7245605e3a

NonPro log

```
echo:DEBUG:
ok
echo:busy: processing
echo:busy: processing
X:155.00 Y:122.90 Z:10.00 E:0.00 Count X:12400 Y:9832 Z:4000
ok
Mesh zeroed.

Bed Topography Report:

    ( 10,210)                                      (210,210)
        0       1       2       3       4       5       6
 6 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 5 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 4 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 3 |  0.000   0.000   0.000 [ 0.000]  0.000   0.000   0.000
[... 278 lines cut ...]
Bed X: 0.00 Y: 0.00 Z: -0.07

Bed Topography Report:

    ( 10,210)                                      (210,210)
        0       1       2       3       4       5       6
 6 |   .     -0.055    .       .       .       .       .
   |
 5 |   .     -0.045  -0.067    .       .       .       .
   |
 4 |   .     -0.064  -0.066    .       .       .       .
   |
 3 |   .     -0.073  +0.022    .       .       .       .
   |
 2 |   .     -0.057  -0.056    .       .       .       .
   |
 1 |   .     -0.080  -0.065    .       .       .       .
   |
 0 |[  .   ]   .       .       .       .       .       .
        0       1       2       3       4       5       6
    ( 10, 10)                                      (210, 10)

Probing mesh point 12/49.

```

classicrocker883, 2024-03-25:
thanks for pointing this out. ill have to check it out and see what I can do.

classicrocker883, 2024-04-01:
@yrabbit , when you say 
>But it's too redundant for me and I constantly end up in this small menu of small buttons

can you explain? what is it you needed to do and what happens instead? maybe there is something I can change to make it better.

yrabbit, 2024-04-01:
Never mind. I’m just not used to the fact that the cursor can move anywhere other than these main four points, but time has passed and I’ve become quite comfortable with the Pro interface and this has ceased to be a problem.

classicrocker883, 2024-04-01:
[Aquila-GD32_UBL-NoPro-fix.zip](https://github.com/classicrocker883/MRiscoCProUI/files/14818702/Aquila-GD32_UBL-NoPro-fix.zip)

here are a couple firmware files i compiled, hopefully "fix1" does the trick, otherwise you shouldn't have an issue with "fix2" the only difference is "fix2" is 7x7 grid and cannot be changed

classicrocker883, 2024-04-01:
that is a main feature I like about the ProUI, having those extra toolbar icons on the main menu, they can be changed to different shortcuts.

well if you wouldn't mind if you ever get a chance to check the 2 "-fix" firmware files i made that would help solve the issue for everyone else.

yrabbit, 2024-04-01:
Certainly.
I checked these firmware. The news is disappointing.
There are simply draconian restrictions on video, so I provide links to testing
Fix-1:
https://yrabbit.cyou/extra/UI-Fix1.mp4
Fix-2:
https://yrabbit.cyou/extra/UI-Fix2.mp4
after which I restored that old version:
https://yrabbit.cyou/extra/UI-OldPro.mp4

So far I have one complaint about the Pro - no matter what I do, testing the surface begins with large indents from the edges.

classicrocker883, 2024-04-06:
thanks for trying, I believe I have found why that was the issue, and I should have a fix in the next build release. 

but if you want to change the distance from the bed edges, go to Mesh Inset menu, it should be in Level > Mesh Settings > Mesh Inset.  
you will see Max / Min for X | Y.  
this will change where it probes the bed surface from the edge.

yrabbit, 2024-04-06:
I did it. That’s not how it works. It seems like there’s a huge 40mm indent somewhere else. I just can't find it.

https://yrabbit.cyou/extra/insets.mp4

classicrocker883, 2024-04-06:
oh I see, your Mesh Inset is too little. you can try this: set to Maximum, and if it is too close to edges then bring it in (increase Min, decrease Max).

or set to default which is 
  Min | Max
X: 10 | 220
Y: 10 | 220

yrabbit, 2024-04-06:
I think I entered the z-probe offset correctly. Somewhere inside the program there is a condition that I do not fulfill and it resets the mesh to some default one - because it does not respond to changes in max/min.

https://yrabbit.cyou/extra/80.mp4

https://yrabbit.cyou/extra/50.mp4

classicrocker883, 2024-04-08:
for Max/Min, I think you need to save settings after changing. so once you change it do "Store Settings"

yrabbit, 2024-04-08:
I turn the printer off and on to make a video, so I have to save the settings :)
You probably don’t have an Aquila printer anymore, or you don’t have the original G32 model, right? :)

classicrocker883, 2024-04-08:
I have the original but swapped the mainboard with 4.2.7 board. so your mesh inset isn't changing??

try this. set the Maximum | Minium X Inset value to 80 | 190 (or values a high for Min and low for Max), then go to Edit Mesh menu, select "Move to Point" checkbox, and the X point to the highest it can go. Y point doesn't matter but can be something in between. but you should observe the printhead **Not** go as far *right* as before. same thing with X point at 0, it should **Not** go as far *left* as before.

yrabbit, 2024-04-08:
It does not care :)

I would look at the prices for the 4.2.7 board, will it definitely work there?

https://yrabbit.cyou/extra/xmin.mp4

classicrocker883, 2024-04-08:
it shouldn't matter what board, because original board works equally the same. so you dont have to get a new board. I only had to get one because mine broke.

here is a firmware which might work, I changed the parameter where it is saved so maybe it should fix. let me know if not.
[Aquila_GD32_UBL-ProUI-meshinset1.zip](https://github.com/classicrocker883/MRiscoCProUI/files/14901853/Aquila_GD32_UBL-ProUI-meshinset1.zip)

yrabbit, 2024-04-08:
Doesn't want to respond to xmin = 80.
https://yrabbit.cyou/extra/new.mp4

In principle, as long as I am in this area in the center, it prints perfectly! My problems start when I want to print something large - I know that the bed is curved :)

[image]

classicrocker883, 2024-04-09:
ive uploaded a recent update for April [you can find here](https://github.com/classicrocker883/MRiscoCProUI/releases/2.1.3f-4)
I may change some things and reupload at later time, but everything should be working as it should if you wanted to try the new firmware

yrabbit, 2024-04-10:
Sorry for the delay - it took many hours to print one piece.

A new version
NoPro (not working):
https://yrabbit.cyou/extra/f-4-non.mp4

Pro (works, but only with a rigidly compiled fixed mesh):
https://yrabbit.cyou/extra/f-4-pro.mp4

Pro insets test (as before):
https://yrabbit.cyou/extra/f-4-pro-insets.mp4

Even though you say that nothing depends on the motherboard, I still found where to buy it inexpensively second-hand, although I’ll have to wait a month and a half.
Since you have such a printer and you obviously check the creation of the mesh on it and everything works fine there, then I’ll wait for the new board for now since I don’t see that anything has changed with my GD32.

classicrocker883, 2024-04-11:
there may be one thing I forgot to mention and that is HS Mode, or High Speed mode. this has sometimes caused issues with the probe and you can disable it in the Level menu under one of the probe settings. 

I have also noticed something strange in the NoPro version, that is being able to change the grid points array like 7x7, 3x3, ect. so when first selecting to change, you may notice there *is* a difference between the Pro versions, ProUI gives it as `#x#` (eg.`7x7`), but NoPro shows it as the a single number, but the strange thing is when you change this, it jumps to `9` and then you select to change. but when reselecting to change, it doesnt do this again, only acts as normal.

even after saving/restart, this odd effect doesn't happen again. I wonder if this has something to do with the NoPro version not working correctly regarding making the mesh.

so if you could, in either version but mostly in NoPro, take note of when changing the grid points and see if changing this makes a difference, that is if you haven't already.

classicrocker883, 2024-04-11:
I can confirm it is firmware related and not the board. I just checked on my machine and it is the same issue. so it's nothing to do with the board. I know the mesh Inset had worked at one point and when flashing mine with updated firmware I see now the problem.

if I had to guess when updating the files like merging with Marlin, so I'll have to find out when and where...

yrabbit, 2024-04-11:
I realized about HS immediately when I received a map of the heights of the Himalayas instead of the bed levels, so this is the first thing I turn off after flashing the new version :)

Well, about the 4.2.7 board - it has more RAM, so one can play with the size of the printer command prefetch buffer (BUFSIZE and Co), which *theoretically* can improve the printing of complex curves consisting of many segments. And it has more FLASH, which will extend the relevance of the printer in light of new Marlin software improvements.

And yeah, I read your advice from two years ago about installing this board, so I'm prepared for differently positioned holes, etc. :)

classicrocker883, 2024-04-12:
update: I fixed the issue, it stemmed from me trying to incorporate changing the mesh_inset in the NoPro version, so that obviously failed to work. but I went back to a working code for ProUI. I tried mimicking the same code so the NoPro could use it, unfortunately I dont understand why it just didn't. and the ProUI source library is a "secret", basically unless you know how to reverse engineer assembly compiler language its impossible to know without trial and error or just know exactly what to do. way out of my league to get it right, I was close though.

I will be uploading as soon as I tweak it a bit. and ill reply back when I do.

classicrocker883, 2024-04-13:
so I have reuploaded the changes, its in the same release 2.1.3f-4
the mesh inset is working as it should now

yrabbit, 2024-04-13:
Great. Anyway, now it’s too late to start printing for many hours, so I’ll try.

closed 2024-04-13 by classicrocker883