classicrocker883/MRiscoCProUI #127: [BUG] IS extruder steps

wlkmanist, 2024-04-18:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Actual extruder steps is doubled with Input Shaping (probably) enabled. Even with manual E axis move and US frequencies set to 0.

### Printer Model

Voxelab Aquila

### Your Mainboard

Aquila GD32

### ProUI?

ProUI

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

TMC2208

classicrocker883, 2024-04-18:
This may be an issue with Marlin source code and not exclusive to *this* repo. but i will try to understand the problem.
so the extruder is twice speed with **IS**?

setting input shaping to `0` doesn't actually disable it. I do not think there is a way to actually disable it. maybe there is a Gcode command which does that.

but if extruder steps are double what do you mean? in the LCD menu, are they showing double from default value, or value is normal number and just the extruder moving faster?

what if you set extruder steps/mm to half, does this fix the issue?

wlkmanist, 2024-04-18:
> This may be an issue with Marlin source code and not exclusive to _this_ repo. but i will try to understand the problem. so the extruder is twice speed with **IS**?
>
> setting input shaping to `0` [...]

I am testing this right now. Same behavour with my old config (for 2.1.3-f3, no IS, enabled JD). Dispay shows right steps as a gcode command. I measured actual filament length, and it is a half less than should be. Idk what causes problem, but maybe it is my config issue. I will write here if I found a solution or fix my own issue.

wlkmanist, 2024-04-18:
Yep, I doubled E steps and it is ok. Probably my issue.

wlkmanist, 2024-04-18:
I found doubled E microsteps in config, so it is my issue. Sorry for that.

closed 2024-04-18 by wlkmanist