classicrocker883/MRiscoCProUI #128: [BUG] Some icons are missing for DWIN LCD

wlkmanist, 2024-04-18:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Missed icons 249-254, may be more.
Used latest available code, same with March builds.

### Printer Model

Voxelab Aquila

### Your Mainboard

Aquila GD32

### ProUI?

ProUI

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Used latest available "Original Icons" pack.

[image]
[image]
[image]

classicrocker883, 2024-04-19:
https://github.com/classicrocker883/MRiscoCProUI/wiki/How-to-update-the-display
>since the 2023-September release, the DWIN_SET has been updated and includes extra ICON's. So, it is recommended to reflash the LCD display.
>>You may need to update to a compatible display firmware / DWIN_SET. You will find the instructions to download and install the display assets from here: https://github.com/classicrocker883/MRiscoCProUI/tree/HEAD/display%20assets

wlkmanist, 2024-04-19:
Yep, I flashed "Custom" dwin set and icons are ok, but with "Original" icons are missing. Is it expected? How can I build that firmware by myself? I can see that icons (Original one) in Icon Packs folder but I can't see it in my display.

wlkmanist, 2024-04-19:
I found a solution. It seems like total icon count issue, I reduced 106->100, like in "Custom" set.

Fixed 9.ICO for "Original" icon set:
[9.zip](https://github.com/classicrocker883/MRiscoCProUI/files/15045515/9.zip)

classicrocker883, 2024-04-20:
> UPD: I splitted the 9.ICO from "Original" set and that icons are existing. Idk why it's not work then.
> 
> UPD2: I splitted also "Custom" and there is different item count. Namings (and numbers) of these icons are different too. Or it is something wrong with tools, because id's are different when creating then splitting.

that is strange. it should work if they exist.
the reason why the namings are different is because the .py script has been updated and some things renamed.
but only 2 icons have been renamed. were there more than 2 renamed? and how can the numbers be different? how different were they?

classicrocker883, 2024-04-20:
> I found a solution. It seems like a total icon count issue, I reduced it 106->100, like in "Custom" set.

this shouldn't be the issue because if you look at the other Pack's like Voxelab Red, this also has 105 icons and it works fine. the only difference is the Original has this icon the others do not - `92 DegreesC`

I believe this might be the cause of the issue because the scripts do not have this. so that is maybe why your ICON's after 92 did not show

classicrocker883, 2024-04-20:
I updated the display assets, I updated the .py script, so now it works even if an icon is missing, that is why it showed different namings and numberings.

I removed the unused 092_ICON_DegreesC, which none of the other's had. I did leave 082-083-084-086-088, which are the 5 others you had removed from your comment because those shouldn't be an issue.
I posted a new 9.ICO file for Original DWIN firmware set, feel free to flash it in the display to confirm it is working and let me know.

wlkmanist, 2024-04-20:
> I posted a new 9.ICO file for Original DWIN firmware set, feel free to flash it in the display to confirm it is working and let me know.

Okay, I will try it.

wlkmanist, 2024-04-20:
I tested, and no, behavour is the same as before. Missed icons (249-254). Icons from 200 to 205 are working, and worked before (206 I can't find to test it).

classicrocker883, 2024-04-21:
ok thanks, ill post the 9.ICO you uploaded, i checked differences between them and it looks like besides the the 5 deleted 82-83-84..... `047-ICON_SetBedTemp` was different. maybe this was the issue? i dont know.

but i have to ask because I'm trying to get the variable Mesh Inset to work. you're using MM - Manual Mesh bed leveling right? with ProUI extra features like the 5 icon toolbar on main menu?

does the Mesh Inset work for you? like if you changed the minimum and maximum mesh inset, you can see a change? to test this go to the Mesh Inset menu in Mesh Settings, adjust the mesh min/max x/y to something else, like more than default so you can notice. such as Min: 50 , Max: 180

then go to Edit Mesh, select Move to Point checkbox, go to X point and reselect 0, or go to the limit of grid points, can be 5 or 7 ( -1 ), you should see the points on the bed correspond to where you set the Min/Max. in other words you should see the new mesh much more within the bed rather than near the edge.

confirm that the variable Mesh Insets are working. you can go back to Mesh Inset settings and select Maximize Area and this will change it near the edge of the bed.

wlkmanist, 2024-04-22:
> `047-ICON_SetBedTemp` was different. maybe this was the issue? i dont know.

I just builded that .ICO for personal use firstly, as experiment, then I found that it works, after a few tries. And because of that I also changed this 047 icon, because if you look for another icons you can see that it is slightly different, at least its resolution different.

> you're using MM - Manual Mesh bed leveling right? with ProUI extra features like the 5 icon toolbar on main menu?

Yep. But not tried yet this feature.

> does the Mesh Inset work for you?

I will try that later, I need some time to learn and test how it works. And then I test it like you ask.

wlkmanist, 2024-04-22:
btw, may it be memory issue? because I tried to flash bootlogo icon with larger size and it is flashed and worked, but with artefacts like a part of jpeg file missed (probably limit for bootlogo is 32k bytes).
"Original" 9.ICO have the most size, as I can see.

classicrocker883, 2024-04-25:
> btw, may it be memory issue? Because I tried to flash bootlogo icon with larger size and it is flashed and worked, but with artefacts like a part of jpeg file missed (probably limit for bootlogo is 32k bytes). "Original" 9.ICO have the most size, as I can see. May be test that someway? Just add something to it to achive only a part of last icons missed, for example.

This could be the issue, because I have noticed the same thing with several ICON's, when the image would appear, it would look like it went into a shredder, lines chopped up or wavy. I attempted to cleanup/restore 78-CON_BLTouch 79-ICON_TempTooLow 80-ICON_AutoLeveling 81-ICON_TempTooHigh.  
Only these did I leave original because updating them did not work.
I did not pay attention to the size of the file, I would not think it increased in size because I made the background all one color (if you look close, when saving as JPG, it loses quality and causes distortion. so what I did was save as PNG and then find a converter that made PNG => JPG without losing quality and causing it to look with random pixels)

same with ICON 47, originally that was 18x18. and to be similar like all the others I had made it 20x20, but I must have overlooked that one when creating the new 9.ICO.

I would not think that there is a limit in size for the ICO file to be flashed to the LCD, I had thought it can hold much more. 16MB Flash memory

From the documentation: "the size of a single JPG image file should not exceed 252KB". I saw for a similar DWIN display that maximum data is 248KB

https://github.com/ihrapsa/T5UIC1-DWIN-toolset/blob/main/DMT48270C043_04WN_datasheet.pdf

closed 2024-05-16 by classicrocker883