classicrocker883/MRiscoCProUI #134: no permite guardar malla aquila s3 se bloquea [BUG]

Manric0421, 2024-05-12:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

no permite guardar malla aquila s3 se bloquea [BUG] 

### Printer Model

Voxelab Aquila

### Model Type

s3

### Your Mainboard

Aquila N32

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Manric0421, 2024-05-12:
Aquila_N32_IND_UBL-ProUI-EX-PLR-05-12
Aquila_IND_N32_UBL-ProUI-EX-PLR-05-11 en estas falla guardar malla
Aquila_IND_N32_UBL-ProUI-EX-PLR-05-10 en esta si funciona

classicrocker883, 2024-05-13:
I uploaded an updated release. please try the new firmware, it should be working correctly

https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-5

closed 2024-05-14 by Manric0421