classicrocker883/MRiscoCProUI #135: Starting.....please wait [BUG]

Oldskool81, 2024-05-13:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Installed  latest firmware and Voxi red DWIN. Aquila_GD32_MM-ProUI-EX-MPC-05-13 screen is stuck on Starting......Please wait. will not boot.

### Printer Model

Voxelab Aquila

### Model Type

Aquila

### Your Mainboard

Aquila GD32

### ProUI?

ProUI

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-05-16:
im pretty sure I know exactly what the cause of this issue is. I will post a new release soon. thanks for pointing this out

classicrocker883, 2024-05-16:
[Aquila_GD32_MM-ProUI-EX-MPC.zip](https://github.com/classicrocker883/MRiscoCProUI/files/15334424/Aquila_GD32_MM-ProUI-EX-MPC.zip)

I made a quick firmware file you can try flashing and see if that works

classicrocker883, 2024-05-18:
> That worked great thank you so much!!!! Can I ask what it was that was causing the issue? Thanks again for all that you do!!!

I had updated the **settings.cpp** file so that it saved memory so more features could be added to some versions. I tested the firmware using a probe so I figured it should be all good. some variables act as place holders in the EEPROM, I just reverted a couple changes I made.

Marlin just updated with a few more file changes, like this [⚡️ Fix motion smoothness ](https://github.com/MarlinFirmware/Marlin/pull/27013)  
So, I'll be uploading a revised release, lookout for that soon.

closed 2024-05-18 by classicrocker883