classicrocker883/MRiscoCProUI #136: No funciona [-PLR] | Recuperación de pérdida de energía | (reanuda donde se quedó un trabajo de impresión cuando hay un corte de energía )[BUG]

Manric0421, 2024-05-14:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

[-PLR]
| Recuperación de pérdida de energía | (reanuda donde se quedó un trabajo de impresión cuando hay un corte de energía )
al momento de corte de energía y reiniciar la impresora no sale nada no da opción de iniciar trabajo no funciona

### Printer Model

Voxelab Aquila

### Model Type

S3

### Your Mainboard

Aquila N32

### ProUI?

ProUI

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-05-16:
can you explain how this does not work?

¿Puedes explicar cómo esto no funciona?

Manric0421, 2024-05-16:
al momento de imprimir y si se apaga la impresora al encenderla nuevamente sale aviso de iniciar el trabajo donde estaba en aquila x3 no lo hace no recupera la impresion prende e inicia como sin recuperar donde iva el trabajo

classicrocker883, 2024-05-16:
OK, I will look into it. did the previous April release work?
¿Funcionó la versión anterior de abril?
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-4

Manric0421, 2024-05-17:
realice pruebas en Aquila s3 y Aquila x3 al reinstalar en firmware funciono de manera correcta la función [-PLR]
| Recuperación de pérdida de energía | (reanuda donde se quedó un trabajo de impresión cuando hay un corte de energía) gracias

closed 2024-05-17 by Manric0421